jsk-enshu/robot-programming #48: urdfモデルはありますか

kimhc6028, 2015-11-04:
@snozawa 先生、このロボットのurdfモデルはありますか？

cc: @k-okada

k-okada, 2015-11-10:
https://github.com/jsk-enshu/robot-programming/pull/53 に追加しました．
https://github.com/jsk-enshu/robot-programming
みて，使ってみて下さい．
roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch 
で立ち上がります．

モデル，チョット（かなり）手を抜いたので，軸間の距離が正しいかk，
細かいパーツやマスパラメータ等入れてもらえると嬉しいです．

kimhc6028, 2015-11-11:
ありがとうございます。詳しい調整は頑張ってやってみます。
後、https://github.com/dgossow/dynamixel_urdfがrobot_programmingに入ってないようですが…

k-okada, 2015-11-12:
https://github.com/jsk-enshu/robot-programming
のreadmeを見て下さい

```
$ wstool set robot-programming https://github.com/jsk-enshu/robot-programming --git -t src
$ wstool set dynamixel_urdf    https://github.com/jsk-enshu/dynamixel_urdf    --git -t src
```

kimhc6028, 2015-11-16:
ごめんなさい。gripperつけるところに詰まっちゃています…

closed 2018-11-15 by tongtybj