jsk-enshu/robot-programming #49: １回目の講義資料のwstool init srcがわかりづらいです。

kimhc6028, 2015-11-07:
githubの方には
```
$ source /opt/ros/hydro/setup.bash   
$ mkdir -p ~/catkin_ws/src
$ cd ~/catkin_ws   
$ wstool init src
$ wstool set robot-programming https://github.com/jsk-enshu/robot-programming --git -t src -v 1.0.2
$ wstool update robot-programming -t src                                                                 
$ rosdep update                                                                                          
$ rosdep install --from-paths src  -y -r                                                                 
$ catkin_make                                                                                            
$ source ~/catkin_ws/devel/setup.bash ## 毎回ターミナルを開く度にこれを行うこと！！   
```
のようにwstool init srcが全体の流れの中に入っていてわかりやすいですけど、第一回講義資料の4pには
```
 cd ~/catkin_ws
$ wstool set robot-programming https://github.com/jsk-enshu/robot-programming --git -t src -v 1.0.1
$ wstool update robot-programming -t src
$ rosdep update
$ rosdep install --from-paths src  -y -r
$ sudo apt-get install ros-hydro-turtlebot ros-hydro-turtlebot-apps ros-hydro-turtlebot-viz
ros-hydro-dynamixel-motor
```
で、wstool init srcが抜けていて、より下の方にありますが、これはちょっとわかりずらいではないかと思います。

kimhc6028, 2015-11-07:
後。。。演習資料でおかしいと思われるところをどんどん追加していきます。
同じく4pで,sudo apt-get install ... ros-hydro-turtlebot-viz ...　がうまく行きませえん。
```
mech-user@test1-pc:~/catkin_ws$ sudo apt-get install ros-hydro-turtlebot-viz
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-hydro-turtlebot-viz
```
ros-indigo-turtlebot-vizにしても一緒でした。

k-okada, 2015-11-09:
On Sun, Nov 8, 2015 at 5:55 AM, Kim Heecheol <notifications@github.com>
wrote:

> ros-indigo-turtlebot-viz
>

 ros-indigo-turtlebot-rviz-launchers
に名前が変わったかな

--
◉ Kei Okada

kimhc6028, 2015-11-09:
既にありました！ありがとうございます

kimhc6028, 2015-11-24:
3回資料の９ｐの８行目あたりで
/camera/depth_registered/cluster_decomposer/debug_output
でなくて
/camera/depth_registered/debug_output
だと思います。
後、11pの「3.2.3Euslispから結果を利用するサンプル」の下に
hsi_colro_filter.launch
と書いていますが、打ち間違いだと思われます。

yuki-asano, 2016-11-24:
> 3回資料の９ｐの８行目あたりで
> /camera/depth_registered/cluster_decomposer/debug_output
> でなくて
> /camera/depth_registered/debug_output
> だと思います。

/camera/depth_registered/cluster_decomposer/debug_output
がpublishされるように修正しました．
https://github.com/jsk-enshu/robot-programming/commit/153108c7027ef504025d469f34d9cd3c8d19c236

> 後、11pの「3.2.3Euslispから結果を利用するサンプル」の下に
> hsi_colro_filter.launch
> と書いていますが、打ち間違いだと思われます。

こちらも修正しました．

closed 2016-11-24 by yuki-asano