jsk-enshu/robot-programming #72: dxl__armed_turtlebot (jsk_pcl_ros) /hsi_color_filterを使っての色抽出のパラメータの決め方について

kochigami, 2016-03-09:
```
 roslaunch dxl_armed_turtlebot hsi_color_filter.launch DEFAULT_NAMESPACE:=/camera/depth_registered INPUT:=points h_max:=30 h_min:=-10 FILTER_NAME_SUFFIX:=red
```
の中の，抽出する色に関するパラメータ，h_max，h_minの決め方についてご教示ください．

hueの範囲は元々0~360で，
```rqt_reconfigure```で```camera→depth_registered→hsi_filter```を選び```h_limit_max, h_limit_min```のパラメータを確かめると，このプログラムでは-128~127に変換されています．
hue: 変換前の値（0~360）
HUE: 変換後の値（-128~127）とすると，
(hue/360)*256-128=HUE となると思うのですが，

この式のHUEに-30, 10(赤色抽出で使われたh_max, h_minの値)を代入すると，
それぞれhue=約223, hue=約166となり，緑色～青色のhue値が出てしまいます． 
勘違いをしていると思うのですが，何が間違っていますでしょうか．

kochigami, 2016-03-09:
[PCLのドキュメント] (http://docs.pointclouds.org/trunk/conditional__removal_8h_source.html)のcondition_removal.h268行目に，
-128 to 127 corresponds to -pi to pi という記述がありました．失礼しました．

closed 2016-03-09 by kochigami

k-okada, 2016-03-10:
launch fileのコメントの所にかいておくといいですね．
あるいは
 <arg name="h_max" default="127" doc="maximum hue, 128 correspond to pi" />
みたいなかくのもあります．
roslaunch --ros-args hogehoge.launch
すると，このdocが参照できます．