jsk-enshu/robot-programming #76: [dxl_armed_robot] display-bounding-box-array.l 内の(send bx :worldcoords)が必要な理由が分かりません．

kochigami, 2016-04-10:
``` dxl_armed_robot/euslisp/display-bounding-box-array.l ```
の中の，

```
(send bx :newcoords (send cam-coords :transform cam->obj-coords))
(send bx :worldcoords)
```
の一行目では，バウンディングボックスの座標をeuslispのワールド座標系にしていますが，
二行目がどうして必要なのか分かりません．

```dxl_armed_robot/euslisp/display-checkerboard.l``` では

```
 (send *target-object* :newcoords (send cam-coords :transform cam->obj-coords))
```
のみが書かれています．
ドキュメントを読んだけれど，
:newcoords, :worldcoords のメソッドの違いが分かりませんでした．

snozawa, 2016-04-10:
ここで表示しているbodyはcascaded-coordsクラスのサブクラスになっています。
cascaded-coordsは連結座標系で、ForwardKinematics演算を可能にするクラスですが、
pos、rot、coordsのセットの度に毎回forwardKinematicsを子すべてに行うのは大変なので、
値のセットとForwardKinematics演算がよばれるタイミングを分けています。

https://github.com/euslisp/EusLisp/blob/master/lisp/l/coordinates.l#L370-L377
のように、cascaded-coordsクラスのworldcoordsでは
updateといってForwardKinematicsを行う計算がよばれます。
worldcoordsメソッドは絶対座標系coords取得の他、なかでFKをといています。

dxl_armed_robot/euslisp/display-checkerboard.l で
worldcoordsの部分をコメントアウトして実行してみてください。
そうすると、認識結果のboudingboxが原点にあつまって適切でないような表示になると思います。
worldcoordsをいれると、newcoordsでセットされた座標のセットが
各頂点などまでFK演算で伝播されて、結果として正しく表示されるようになります。

kochigami, 2016-04-11:
cascaded-coordsについて勉強不足でした．
理解することができました．

ご教示ありがとうございました．

closed 2016-04-11 by kochigami