jsk-enshu/robot-programming #79: gazebo did not start

yuta-inoue, 2016-10-17:
```
sfully spawned model
Error [Plugin.hh:156] Failed to load plugin libgazebo_ros_kobuki.so: libgazebo_ros_kobuki.so: cannot open shared object file: No such file or directory
Error [Plugin.hh:156] Failed to load plugin libgazebo_ros_control.so: libgazebo_ros_control.so: cannot open shared object file: No such file or directory
[ INFO] [1476667252.772324507, 0.240000000]: Physics dynamic reconfigure ready.
[urdf_spawner-6] process has finished cleanly
log file: /home/mech-user/.ros/log/a9013ee8-9407-11e6-93ef-e8b1fce1a4c4/urdf_spawner-6*.log
```

yuta-inoue, 2016-10-17:
```
sudo apt-get install ros-indigo-gazebo-plugins ros-indigo-kobuki-gazebo-plugins 
```
で解決しそう

yuta-inoue, 2016-10-17:
```
Error [Plugin.hh:156] Failed to load plugin libgazebo_ros_control.so: libgazebo_ros_control.so: cannot open shared object file: No such file or directory
```
というエラーもある。

yuta-inoue, 2016-10-17:
```
[ERROR] [1476668028.097318102, 1.230000000]: Could not load controller 'trajectory_controller' because controller type 'velocity_controllers/JointTrajectoryController' does not exist.
[ERROR] [1476668028.097352650, 1.230000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [WallTime: 1476668029.098392] [2.230000] Failed to load trajectory_controller
```

yuta-inoue, 2016-10-17:
```
 2004  sudo apt-get intsall ros-indigo-ros-control
 2005  sudo apt-get install ros-indigo-ros-control
 2006  sudo apt-get install ros-indigo-gazebo-ros-control
 2007  sudo apt-get install ros-indigo-velocity-controller
 2008  sudo apt-get install ros-indigo-joint-state-controller 
 2009* sudo apt-get install ros-indigo-
 2010  sudo apt-get install ros-indigo-velocity-controller
 2011  sudo apt-get install ros-indigo-velocity-controllers
```
したらうごいていそう。

yuta-inoue, 2016-10-17:
```
sudo apt-get install ros-indigo-joint-trajectory-controller
```
も

yuki-asano, 2016-10-17:
報告ありがとうございます．
私の手元では，以下を実行すると動くようでした．

```
sudo apt-get install ros-indigo-gazebo-ros
sudo apt-get install ros-indigo-gazebo-plugins
sudo apt-get install ros-indigo-gazebo-ros-control
sudo apt-get install ros-indigo-kobuki-gazebo
sudo apt-get install ros-indigo-controller-manager
sudo apt-get install ros-indigo-joint-state-controller
sudo apt-get install ros-indigo-joint-trajectory-controller
```

shihonshugi, 2016-10-24:
僕もうまく動きませんでした。

```
$ roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch
... logging to /home/mech-user/.ros/log/f5c35fe4-997b-11e6-9b46-e8b1fce9934b/roslaunch-test1-pc-4394.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://test1-pc:34928/

SUMMARY
========

PARAMETERS
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: indigo
 * /rosversion: 1.11.16
 * /trajectory_controller/action_monitor_rate: 100
 * /trajectory_controller/constraints/arm_link1_joint/goal: 0.02
 * /trajectory_controller/constraints/arm_link1_joint/trajectory: 0.05
 * /trajectory_controller/constraints/arm_link2_joint/goal: 0.02
 * /trajectory_controller/constraints/arm_link2_joint/trajectory: 0.05
 * /trajectory_controller/constraints/arm_link3_joint/goal: 0.02
 * /trajectory_controller/constraints/arm_link3_joint/trajectory: 0.05
 * /trajectory_controller/constraints/arm_link4_joint/goal: 0.02
 * /trajectory_controller/constraints/arm_link4_joint/trajectory: 0.05
 * /trajectory_controller/constraints/arm_link5_joint/goal: 0.02
 * /trajectory_controller/constraints/arm_link5_joint/trajectory: 0.05
 * /trajectory_controller/constraints/arm_link6_joint/goal: 0.02
 * /trajectory_controller/constraints/arm_link6_joint/trajectory: 0.05
 * /trajectory_controller/constraints/arm_link7_joint/goal: 0.02
 * /trajectory_controller/constraints/arm_link7_joint/trajectory: 0.05
 * /trajectory_controller/constraints/goal_time: 0.5
 * /trajectory_controller/constraints/stopped_velocity_tolerance: 0.02
 * /trajectory_controller/gains/arm_link1_joint/d: 0.0
 * /trajectory_controller/gains/arm_link1_joint/i: 0.0
 * /trajectory_controller/gains/arm_link1_joint/p: 25.0
 * /trajectory_controller/gains/arm_link2_joint/d: 0.0
 * /trajectory_controller/gains/arm_link2_joint/i: 0.0
 * /trajectory_controller/gains/arm_link2_joint/p: 25.0
 * /trajectory_controller/gains/arm_link3_joint/d: 0.0
 * /trajectory_controller/gains/arm_link3_joint/i: 0.0
 * /trajectory_controller/gains/arm_link3_joint/p: 25.0
 * /trajectory_controller/gains/arm_link4_joint/d: 0.0
 * /trajectory_controller/gains/arm_link4_joint/i: 0.0
 * /trajectory_controller/gains/arm_link4_joint/p: 25.0
 * /trajectory_controller/gains/arm_link5_joint/d: 0.0
 * /trajectory_controller/gains/arm_link5_joint/i: 0.0
 * /trajectory_controller/gains/arm_link5_joint/p: 25.0
 * /trajectory_controller/gains/arm_link6_joint/d: 0.0
 * /trajectory_controller/gains/arm_link6_joint/i: 0.0
 * /trajectory_controller/gains/arm_link6_joint/p: 25.0
 * /trajectory_controller/gains/arm_link7_joint/d: 0.0
 * /trajectory_controller/gains/arm_link7_joint/i: 0.0
 * /trajectory_controller/gains/arm_link7_joint/p: 5.0
 * /trajectory_controller/joints: ['arm_link1_joint...
 * /trajectory_controller/state_publish_rate: 100
 * /trajectory_controller/stop_trajectory_duration: 0
 * /trajectory_controller/type: velocity_controll...
 * /use_sim_time: True

NODES
  /
    cmd_vel_renamer (topic_tools/relay)
    controller_spawner (controller_manager/spawner)
    gazebo (gazebo_ros/gzserver)
[... 32 lines cut ...]
Msg Waiting for master
Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 10.10.1.220
[ INFO] [1477266837.689363850]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Msg Waiting for master
Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 10.10.1.220
spawn_model script started
[INFO] [WallTime: 1477266837.923429] [0.000000] Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1477266837.967657894, 0.030000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1477266837.992300570, 0.050000000]: Physics dynamic reconfigure ready.
[INFO] [WallTime: 1477266838.038693] [0.000000] Loading model xml from ros parameter
[INFO] [WallTime: 1477266838.043438] [0.000000] Waiting for service /gazebo/spawn_urdf_model
[INFO] [WallTime: 1477266838.044520] [0.000000] Calling service /gazebo/spawn_urdf_model
Error [Plugin.hh:156] Failed to load plugin libgazebo_ros_openni_kinect.so: libgazebo_ros_openni_kinect.so: cannot open shared object file: No such file or directory
[INFO] [WallTime: 1477266839.069264] [0.770000] Spawn status: SpawnModel: Successfully spawned model
Error [Plugin.hh:156] Failed to load plugin libgazebo_ros_kobuki.so: libgazebo_ros_kobuki.so: cannot open shared object file: No such file or directory
Error [Plugin.hh:156] Failed to load plugin libgazebo_ros_control.so: libgazebo_ros_control.so: cannot open shared object file: No such file or directory
Error [Plugin.hh:156] Failed to load plugin libgazebo_ros_p3d.so: libgazebo_ros_p3d.so: cannot open shared object file: No such file or directory
[urdf_spawner-7] process has finished cleanly
log file: /home/mech-user/.ros/log/f5c35fe4-997b-11e6-9b46-e8b1fce9934b/urdf_spawner-7*.log
[WARN] [WallTime: 1477266868.101201] [29.790000] Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_spawner-8] process has finished cleanly
log file: /home/mech-user/.ros/log/f5c35fe4-997b-11e6-9b46-e8b1fce9934b/controller_spawner-8*.log
```

closed 2016-11-28 by yuki-asano