jsk-enshu/robot-programming #84: viewerが立ち上がりません

iZuno-K, 2016-10-27:
rostopic echo /camshift/track_box　までは動くのですが、その次のものが動きません。

mech-user@test1-pc:~/catkin_ws/src$ rosrun image_view2 image_view2 image:=/camera/rgb/image_raw \ ~image_transport:=compressed
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [ ] is not valid as the first character in Graph Resource Name [ ~image_transport].  Valid characters are a-z, A-Z, / and in some cases ~.
中止 (コアダンプ)
mech-user@test1-pc:~/catkin_ws/src$

ayamiya, 2016-10-27:
```
$ rosrun image_view2 image_view2 image:=/camera/rgb/image_raw \
  ~image_transport:=compressed
```
とかいてある場合、`\`は「次の行もつなげて書いてください」という意味なので、`\`は打ち込まずに、一行で、

`rosrun image_view2 image_view2 image:=/camera/rgb/image_raw ~image_transport:=compressed`

と書く意味のようです。

iZuno-K, 2016-10-27:
起動しました。ありがとうございます。

closed 2016-11-28 by yuki-asano