jsk-enshu/robot-programming #108: camshift.launchを立ち上げて１０秒ぐらいすると画面が黒くなる

mirai05, 2016-10-28:
```
roslaunch --screen opencv_apps camshift.launch image:=/camera/rgb/image_raw
```
として１０秒ぐらいすると、画面が黒くなります。
立ち上げた直後の画面は以下。何もしていないつもりだけど、スクロールバーが下（みんなは上）で、色も青い。

[image]

camshiftのソースコードは

https://github.com/ros-perception/opencv_apps/blob/indigo/src/nodelet/camshift_nodelet.cpp

みてみるとGUIを出さないモードもあるので、まずはそれで動くか調べてみる。

mirai05, 2016-10-28:
追記

[ WARN] [1477639187.934265187, 7875.530000000]: '/camshift' subscribes topics only with child subscribers.

このようなエラーが出たあと、画面が黒くなって動かなくなりました。

k-okada, 2016-10-28:
あ、rostopic echo /camshift/track_box とか rosrun image_view2 ... とかしたら変わるかも

mirai05, 2016-11-29:
解決しました。
問題は、OpenCVのバージョンが違ったからで、元に戻すと動くようになりました。

closed 2017-03-08 by yuki-asano