jsk-enshu/robot-programming #128: git fetch --allが通らない

M-Onitsuka, 2016-10-31:
表題のとおりです
メッセージは以下の通り．
```
mech-user@test1-pc:~/catkin_ws/src/robot-programming$ git fetch --all
Fetching origin
Fetching M-Onitsuka
fatal: repository 'https://github.com/M-Onitsuka/' not found
error: Could not fetch M-Onitsuka
```
ご教示お待ちしております

k-okada, 2016-10-31:
git remote -v としてみてください．
https://github.com/jsk-enshu/robot-programming
https://github.com/M-Onitsuka/
https://github.com/M-Onitsuka/robot-programming
となっているかな．いずれにせよ、
https://github.com/M-Onitsuka/
はリポジトリではないので
git branch -d https://github.com/M-Onitsuka/
として消してから
git fetch --all
としてみてください．

M-Onitsuka, 2016-11-01:
とりあえずgit remote -vの結果は以下のとおりになりました．
```
mech-user@test1-pc:~/catkin_ws/src/robot-programming$ git remote -v
M-Onitsuka	https://github.com/M-Onitsuka (fetch)
M-Onitsuka	https://github.com/M-Onitsuka (push)
origin	https://github.com/jsk-enshu/robot-programming (fetch)
origin	https://github.com/jsk-enshu/robot-programming (push)
mech-user@test1-pc:~/catkin_ws/src/robot-programming$ git branch -d https://github.co/M-Onitsuka
error: branch 'https://github.com/M-Onitsuka' not found.
mech-user@test1-pc:~/catkin_ws/src/robot-programming$ git fetch --all
Fetching origin
Fetching M-Onitsuka
fatal: repository 'https://github.com/M-Onitsuka/' not found
error: Could not fetch M-Onitsuka
```
git branch -d でbranchが見つからない状態になっています．originのfetchは通っているようです．

k-okada, 2016-11-01:
まちがえました．
git branch -d M-Onitsuka
かな．

M-Onitsuka, 2016-11-02:
```
mech-user@test1-pc:~/catkin_ws/src/robot-programming/dxl_armed_turtlebot/scripts$ git remote remove M-Onitsuka
mech-user@test1-pc:~/catkin_ws/src/robot-programming/dxl_armed_turtlebot/scripts$ git remote -v
origin	https://github.com/jsk-enshu/robot-programming (fetch)
origin	https://github.com/jsk-enshu/robot-programming (push)
```
でした．

closed 2018-11-15 by tongtybj