jsk-enshu/robot-programming #136: turtlebot_navigation.launch できません

iZuno-K, 2016-11-07:
シミュレータは実行できるのですがmapに関して何かエラーを出しています。その後のturtlebot_navigation.launchを実行すると消えると行っていた人もいたのですが、それはエラーが出て実行できません。rosdep install --from-paths . --ignore-src -y -r 以降が怪しい気がします。

wstool update
```
mech-user@test1-pc:~/catkin_ws/src$ wstool update
[dynamixel_urdf] Updating /home/mech-user/catkin_ws/src/dynamixel_urdf
[dynamixel_urdf] Done.
[robot-programming] Updating /home/mech-user/catkin_ws/src/robot-programming
[robot-programming] Done.
```

rosdep update
```
mech-user@test1-pc:~/catkin_ws/src$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
updated cache in /home/mech-user/.ros/rosdep/sources.cache
```

rosdep install --from-paths . --ignore-src -y -r
```
mech-user@test1-pc:~/catkin_ws/src$ rosdep install --from-paths . --ignore-src -y -r
executing command [sudo -H apt-get install -y ros-indigo-turtlebot-navigation]
[sudo] password for mech-user: 
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています                
状態情報を読み取っています... 完了
以下のパッケージが自動でインストールされましたが、もう必要とされていません:
  fltk1.3-doc fluid hddtemp libfltk-forms1.3 libfltk-images1.3 libfltk1.1
  libfltk1.1-dev libfltk1.3 python-opengl python-psutil python-qt4-gl
  ros-indigo-bond-core ros-indigo-camera-calibration ros-indigo-common-msgs
  ros-indigo-common-tutorials ros-indigo-compressed-depth-image-transport
  ros-indigo-compressed-image-transport ros-indigo-desktop
  ros-indigo-diagnostic-analysis ros-indigo-diagnostic-common-diagnostics
  ros-indigo-diagnostics ros-indigo-executive-smach ros-indigo-filters
  ros-indigo-gazebo-ros-pkgs ros-indigo-geometry ros-indigo-geometry-tutorials
  ros-indigo-image-common ros-indigo-image-pipeline ros-indigo-image-rotate
  ros-indigo-image-transport-plugins ros-indigo-interactive-marker-tutorials
  ros-indigo-interactive-markers ros-indigo-joint-state-publisher
  ros-indigo-laser-assembler ros-indigo-laser-filters
  ros-indigo-laser-pipeline ros-indigo-librviz-tutorial
  ros-indigo-media-export ros-indigo-mk ros-indigo-nodelet-core
  ros-indigo-nodelet-tutorial-math ros-indigo-pluginlib-tutorials
  ros-indigo-pointcloud-to-laserscan ros-indigo-qt-gui-cpp ros-indigo-robot
  ros-indigo-robot-model ros-indigo-ros ros-indigo-ros-base
  ros-indigo-ros-comm ros-indigo-ros-core ros-indigo-ros-tutorials
  ros-indigo-rosboost-cfg ros-indigo-roscpp-core ros-indigo-roscpp-tutorials
  ros-indigo-roscreate ros-indigo-roslint ros-indigo-roslisp
  ros-indigo-rosmake ros-indigo-rqt-action ros-indigo-rqt-bag
  ros-indigo-rqt-bag-plugins ros-indigo-rqt-common-plugins
  ros-indigo-rqt-console ros-indigo-rqt-dep ros-indigo-rqt-gui-cpp
  ros-indigo-rqt-image-view ros-indigo-rqt-launch ros-indigo-rqt-logger-level
  ros-indigo-rqt-moveit ros-indigo-rqt-msg ros-indigo-rqt-nav-view
  ros-indigo-rqt-pose-view ros-indigo-rqt-publisher ros-indigo-rqt-py-console
  ros-indigo-rqt-reconfigure ros-indigo-rqt-robot-dashboard
  ros-indigo-rqt-robot-monitor ros-indigo-rqt-robot-plugins
  ros-indigo-rqt-robot-steering ros-indigo-rqt-runtime-monitor
  ros-indigo-rqt-rviz ros-indigo-rqt-service-caller ros-indigo-rqt-shell
  ros-indigo-rqt-srv ros-indigo-rqt-tf-tree ros-indigo-rqt-top
  ros-indigo-rqt-topic ros-indigo-rqt-web ros-indigo-rviz
[... 253 lines cut ...]
[build] Runtime: 4.8 seconds total.                        
```

turtlebot_navigation.launch
```
mech-user@test1-pc:~$ roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_navigation.launch
... logging to /home/mech-user/.ros/log/2f340610-a4cc-11e6-98ad-e8b1fce891a3/roslaunch-test1-pc-6597.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Invalid <arg> tag: turtlebot_navigation
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/mech-user/catkin_ws/src/beginner_tutorials
ROS path [2]=/home/mech-user/catkin_ws/src/robot-programming/dynamixel_7dof_arm
ROS path [3]=/home/mech-user/catkin_ws/src/dynamixel_urdf
ROS path [4]=/home/mech-user/catkin_ws/src/robot-programming/turtleboteus
ROS path [5]=/home/mech-user/catkin_ws/src/robot-programming/dxl_armed_turtlebot
ROS path [6]=/home/mech-user/catkin_ws/src/robot-programming/daisya_euslisp_tutorials
ROS path [7]=/opt/ros/indigo/share
ROS path [8]=/opt/ros/indigo/stacks. 

Arg xml is <arg default="$(find turtlebot_navigation)/param/dummy.yaml" name="custom_param_file"/>
The traceback for the exception was written to the log file
```

k-okada, 2016-11-07:
sudo apt-get install python-rosdep
してから、もう一度
rosdep install --from-paths . --ignore-src -y -r
してください．
これでダメなら、
sudo updatedb
locate rosdep
の結果を教えて下さい．

iZuno-K, 2016-11-08:
apt-getとrosdep installを実行してみましたがエラーは消えませんでした。

sudo updatedbとlocate rosdepの結果
```
mech-user@test1-pc:~/catkin_ws/src$ sudo updatedb
mech-user@test1-pc:~/catkin_ws/src$ locate rosdep
/etc/ros/rosdep
/etc/ros/rosdep/sources.list.d
/etc/ros/rosdep/sources.list.d/20-default.list
/home/mech-user/.ros/rosdep
/home/mech-user/.ros/rosdep/sources.cache
/home/mech-user/.ros/rosdep/sources.cache/0ce507007835947045d89dfa382199f6e83de08c.pickle
/home/mech-user/.ros/rosdep/sources.cache/5d76dbb61bf5385acb00a1ffce292b2eb2e000a8.pickle
/home/mech-user/.ros/rosdep/sources.cache/6a3d722dfe2d09a104a761d77901e00b2e7a540c.pickle
/home/mech-user/.ros/rosdep/sources.cache/7cfbeed211a3f9a6507c86cebaeedb3e3c4387f8.pickle
/home/mech-user/.ros/rosdep/sources.cache/825fae71ab9ea366e0b22e8dbe1e2948232cb2e8.pickle
/home/mech-user/.ros/rosdep/sources.cache/a4b3c8956ab7a3e6d293ba33b57fac42799e69e6.pickle
/home/mech-user/.ros/rosdep/sources.cache/a75d3a0d86182afb2eb92f6c5473bf18db157d0e.pickle
/home/mech-user/.ros/rosdep/sources.cache/b4931581ca7c5741f7ee4ac9767a1174838296f7.pickle
/home/mech-user/.ros/rosdep/sources.cache/b67f64890f6d5269c51a68744711f115a8f207ae.pickle
/home/mech-user/.ros/rosdep/sources.cache/f7622dadcf19914a1d2d3d8417ec62c3060c5d3e.pickle
/home/mech-user/.ros/rosdep/sources.cache/index
/home/mech-user/catkin_ws/ws_euslisp/src/jsk_roseus/.travis/rosdep-install.sh
/opt/ros/indigo/lib/python2.7/dist-packages/roswtf/rosdep_db.py
/opt/ros/indigo/lib/python2.7/dist-packages/roswtf/rosdep_db.pyc
/usr/bin/rosdep
/usr/bin/rosdep-source
/usr/lib/python2.7/dist-packages/rosdep-0.11.5.egg-info
/usr/lib/python2.7/dist-packages/rosdep2
/usr/lib/python2.7/dist-packages/rosdep-0.11.5.egg-info/PKG-INFO
/usr/lib/python2.7/dist-packages/rosdep-0.11.5.egg-info/SOURCES.txt
/usr/lib/python2.7/dist-packages/rosdep-0.11.5.egg-info/dependency_links.txt
/usr/lib/python2.7/dist-packages/rosdep-0.11.5.egg-info/requires.txt
/usr/lib/python2.7/dist-packages/rosdep-0.11.5.egg-info/top_level.txt
/usr/lib/python2.7/dist-packages/rosdep2/__init__.py
/usr/lib/python2.7/dist-packages/rosdep2/__init__.pyc
/usr/lib/python2.7/dist-packages/rosdep2/_version.py
/usr/lib/python2.7/dist-packages/rosdep2/_version.pyc
/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.py
/usr/lib/python2.7/dist-packages/rosdep2/catkin_packages.pyc
/usr/lib/python2.7/dist-packages/rosdep2/catkin_support.py
/usr/lib/python2.7/dist-packages/rosdep2/catkin_support.pyc
/usr/lib/python2.7/dist-packages/rosdep2/core.py
/usr/lib/python2.7/dist-packages/rosdep2/core.pyc
/usr/lib/python2.7/dist-packages/rosdep2/dependency_graph.py
/usr/lib/python2.7/dist-packages/rosdep2/dependency_graph.pyc
/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py
/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.pyc
/usr/lib/python2.7/dist-packages/rosdep2/installers.py
/usr/lib/python2.7/dist-packages/rosdep2/installers.pyc
/usr/lib/python2.7/dist-packages/rosdep2/loader.py
/usr/lib/python2.7/dist-packages/rosdep2/loader.pyc
/usr/lib/python2.7/dist-packages/rosdep2/lookup.py
/usr/lib/python2.7/dist-packages/rosdep2/lookup.pyc
/usr/lib/python2.7/dist-packages/rosdep2/main.py
/usr/lib/python2.7/dist-packages/rosdep2/main.pyc
/usr/lib/python2.7/dist-packages/rosdep2/model.py
/usr/lib/python2.7/dist-packages/rosdep2/model.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms
/usr/lib/python2.7/dist-packages/rosdep2/rep3.py
/usr/lib/python2.7/dist-packages/rosdep2/rep3.pyc
/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.py
/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.pyc
/usr/lib/python2.7/dist-packages/rosdep2/rospack.py
/usr/lib/python2.7/dist-packages/rosdep2/rospack.pyc
/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.py
/usr/lib/python2.7/dist-packages/rosdep2/rospkg_loader.pyc
/usr/lib/python2.7/dist-packages/rosdep2/shell_utils.py
/usr/lib/python2.7/dist-packages/rosdep2/shell_utils.pyc
/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py
/usr/lib/python2.7/dist-packages/rosdep2/sources_list.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/__init__.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/__init__.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/arch.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/arch.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/cygwin.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/cygwin.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/freebsd.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/freebsd.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/gem.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/gem.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/gentoo.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/gentoo.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/opensuse.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/opensuse.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/osx.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/osx.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/pip.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/pip.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/redhat.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/redhat.pyc
/usr/lib/python2.7/dist-packages/rosdep2/platforms/source.py
/usr/lib/python2.7/dist-packages/rosdep2/platforms/source.pyc
/usr/share/doc/python-rosdep
/usr/share/doc/python-rosdep/changelog.Debian.gz
/var/cache/apt/archives/python-rosdep_0.11.5-1_all.deb
/var/lib/dpkg/info/python-rosdep.list
/var/lib/dpkg/info/python-rosdep.md5sums
/var/lib/dpkg/info/python-rosdep.postinst
/var/lib/dpkg/info/python-rosdep.prerm
```

k-okada, 2016-11-08:
apt-get update
のあとにもう一度同じことをやるのはどうでしょうか．

iZuno-K, 2016-11-09:
```
sudo apt-get update
sudo apt-get install python-rosdep
rosdep install --from-paths . --ignore-src -y -r
```
で実行できるようになりました。ありがとうございます。

closed 2016-11-28 by yuki-asano