jsk-enshu/robot-programming #147: No such file or directory: u'/opt/ros/indigo/share/turtlebot_navigation/launch/includes/amcl.launch. と表示される

nkg1995, 2016-11-29:
```
$ roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_navigation.launch 
... logging to /home/mech-user/.ros/log/4c441b7a-b609-11e6-89f5-e8b1fce9936e/roslaunch-test1-pc-3821.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

while processing /opt/ros/indigo/share/turtlebot_navigation/launch/includes/amcl.launch.xml:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/opt/ros/indigo/share/turtlebot_navigation/launch/includes/amcl.launch.
```

nkg1995, 2016-11-29:
~/catkin_ws/src/robot-programming で
```
git pull origin master
rosdep install --from-paths . --ignore-src
```
とするとなおりました。

yuki-asano, 2016-12-12:
wikiに転載しました。
https://github.com/jsk-enshu/robot-programming/wiki/FAQ

closed 2016-12-12 by yuki-asano

k-okada, 2016-12-12:
https://github.com/jsk-enshu/robot-programming/pull/134
で治っているはずです．ちゃんと資料通りにwstool update していなかったのかと、