jsk-enshu/robot-programming #149: シミュレータで *ri* の gripper を動かすとエラーする

kwmrmsy, 2016-11-29:
turtlebotの実機に接続せずシミュレータで動かしていると、 *ri* のグリッパに指令を送った際に
```
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :angle-vector in (send :dxl-armed-turtlebot* :angle-vector)
```
というエラーが出ます。
グリッパ以外のアームではエラーは出ず、実機ではグリッパも動いているようなので、サーボモータのコンプライアンス制御まわりでシミュレータではメソッドが見つからないということでしょうか。
```
1.irteusgl$ load "package://dxl_armed_turtlebot/euslisp/dxl-armed-turtlebot-interface.l"
;; (dxl-armed-turtlebot-init) ;; for initialize
t
2.irteusgl$ dxl-armed-turtlebot-init

3.irteusgl$ send *dxl-armed-turtlebot* :gripper :arm :angle-vector #f(50)
#f(50.0)
4.irteusgl$ send *ri* :angle-vector (send :dxl-armed-turtlebot* :angle-vector) 1000 :gripper-controller
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :angle-vector in (send :dxl-armed-turtlebot* :angle-vector)
5.E1-irteusgl$ send *ri* :start-grasp
[ERROR] [1480428280.037438510]: attempted to call service  /arm_j7_controller/set_compliance_slope, but failed 
[ERROR] [1480428280.037981640]: attempted to call service  /arm_j7_controller/set_torque_limit, but failed 
:simulate should be defined in lower class
:simulate should be defined in lower class
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :push-angle-vector-simulation in (send-all (gethash ctype controller-table) :push-angle-vector-simulation av tm prev-av)
```

k-okada, 2016-12-01:
`:dxl-armed-turtlebot*` というオブエクとは無い気がしますが、`*dxl-armed-turtlebot*` ?

verry619, 2016-12-05:
同様のエラーが出てしまうようです。

```
 (send *ri* :angle-vector (send *dxl-armed-turtlebot* :angle-vector) 1000 :gripper-controller)
:simulate should be defined in lower class
:simulate should be defined in lower class
/opt/ros/indigo/share/euslisp/jskeus/eus/Linux64/bin/irteusgl 0 error: cannot find method :push-angle-vector-s\
imulation in (send-all (gethash ctype controller-table) :push-angle-vector-simulation av tm prev-av)
```

[image]