jsk-enshu/robot-programming #162: dynamixel を使ってカスタムアームを作る

k-okada, 2016-12-12:
https://gist.github.com/k-okada/a19f7d10f9221093e70ccb06282d5957 を解読するとできそう

M-Onitsuka, 2016-12-12:
```
wget https://gist.githubusercontent.com/k-okada/a19f7d10f9221093e70ccb06282d5957/raw/e6b37deb1ccd5e4445b23237b07186f86d10c389/gistfile1.txt -O 228.diff
```
としてから，
`catkin ws でpatch -p3 < 228.diff ` とすると，spine_3というパッケージができるので，
```

mv spine_3 oni_robot
cd oni_robot
find . -type f -exec sed -i s/spine_3/oni_robot/g  {} \;
```
とすると良さそう．１軸だと，
`dynamixel_bringup.launch` で，`max_motor` と, 2つのcontroller_spawnerのargsでj1だけにして，
euslisp/spine-3-interfaceで，ロボットモデルを１軸にすれば良さそう．

M-Onitsuka, 2016-12-20:
dynamixel AX-12を３個用いた３自由度のロボットをとりあえず次のようにして動かしてみました．
まずdynamixel_joint3_controllers.yamlなどを作って
```
arm_j1_controller:
    controller:
        package: dynamixel_controllers
        module: joint_position_controller
        type: JointPositionController
    joint_name: arm_joint1
    joint_speed: 2.0
    motor:
        id: 1
        init: 512
        min: 0
        max: 1023

arm_j2_controller:
    controller:
        package: dynamixel_controllers
        module: joint_position_controller
        type: JointPositionController
    joint_name: arm_joint2
    joint_speed: 2.0
    motor:
        id: 2
        init: 512
        min: 0
        max: 1023

arm_j3_controller:
    controller:
        package: dynamixel_controllers
        module: joint_position_controller
        type: JointPositionController
    joint_name: arm_joint3
    joint_speed: 2.0
    motor:
        id: 3
        init: 512
        min: 0
        max: 1023

fullbody_controller:
   controller:
       package: dynamixel_controllers
       module: joint_trajectory_action_controller
       type: JointTrajectoryActionController
   joint_trajectory_action_node:
       min_velocity: 0.25
       constraints:
           goal_time: 0.0

gripper_controller:
   controller:
       package: dynamixel_controllers
       module: joint_trajectory_action_controller
       type: JointTrajectoryActionController
   joint_trajectory_action_node:
       min_velocity: 0.25
       constraints:
           goal_time: 0.0

```
としました．概要としては元からあったyamlファイルを3自由度に変えました．

次にdymixel_new_branch.launchなどを書いて
```
<!-- -*- mode: XML -*- -->

<launch>
    <node name="dynamixel_manager" pkg="dynamixel_controllers" type="controller_manager.py" required="true" output="screen">
        <rosparam>
            namespace: dxl_manager
            serial_ports:
                pan_tilt_port:
                    port_name: "/dev/dynamixel_arm"
                    baud_rate: 1000000
                    min_motor_id: 1
                    max_motor_id: 25
                    update_rate: 20
        </rosparam>
    </node>

    <!-- Load controller configuration to parameter server -->
    <rosparam file="$(find dynamixel_7dof_arm)/config/dynamixel_joint3_controllers.yaml" command="load"/>

    <!-- start specified joint controllers -->
    <node name="dynamixel_controller_spawner" pkg="dynamixel_controllers" type="controller_spawner.py"
          args="--manager=dxl_manager
                --port=pan_tilt_port
                --type=simple
                arm_j1_controller
                arm_j2_controller
                arm_j3_controller"
          output="screen"/>

    <!-- start trajectory controllers for fullbody + gripper -->
    <node name="dynamixel_trajectory_controller_spawner_for_fullbody" pkg="dynamixel_controllers" type="controller_spawner.py"
          args="--manager=dxl_manager
                --port=pan_tilt_port
                --type=meta
                fullbody_controller
                arm_j1_controller
                arm_j2_controller
                arm_j3_controller"
          output="screen"/>

</launch>
```
としてみました．こちらも自由度を変えただけです．
このlaunchファイルを実行をした上でtest-dynamixel-motor.lをroseusで実行したらとりあえず動きました．
今のところの問題は*ri*に:angle-vectorとかしても動かないことです．

また，新しい質問として，なんとなくlaunchファイル内で
```
serial_ports:
                7dof_arm_port:
```
を
```
serial_ports:
                pan_tilt_port:
```
変えてみたのですが，ここがどう効いてるのかよくわからないです．

k-okada, 2016-12-20:
３軸のロボットであれあば、上に書いた方法でspine_3パッケージを作って
以下のサンプルで試して下さい．

```
$ roseus spine-3-interface.l
(setq *ri* (instance spine-3-interface :init))
(send *ri* :state :potentio-vector)
(send *ri* :angle-vector #f(0 10 20) 10000)
(send *ri* :wait-interpolation)

```
>

ayamiya, 2016-12-20:
AX-12 だと以下の変更が必要でした。
```
diff -r ../config/dynamixel_joint_controllers.yaml /tmp/spine_3/config/dynamixel_joint_controllers.yaml
10c10
<         init: 512
---
>         init: 2048
12c12
<         max: 1023
---
>         max: 4096
23c23
<         init: 512
---
>         init: 2048
25c25
<         max: 1023
---
>         max: 4096
36c36
<         init: 512
---
>         init: 2048
38c38
<         max: 1023
---
>         max: 4096
diff -r ../launch/dynamixel_bringup.launch /tmp/spine_3/launch/dynamixel_bringup.launch
10c10
<                     baud_rate: 1000000
---
>                     baud_rate: 57143
```