jsk-enshu/robot-programming #165: euslispで時間計測するためには

ayamiya, 2016-12-22:
euslispで一定時間が経過したら行動を変更するプログラムを書きたいです。
ただ、時間経過中は元の行動を継続していたいのですが、どうすればいいですか。
time-nowとかsleepだと行動が停止してしまうのでうまくいきません。

yuki-asano, 2016-12-22:
多分出来ました。
これで目的の動作はできそうでしょうか。

```
#!/usr/bin/env roseus

(ros::roseus "time-test")

(setq *passed-time* 0)

(do-until-key
 (setq before (ros::time-now))
 (unix::usleep (* 1000 1000))
 (setq after (ros::time-now))
 (setq delta-t (send (ros::time- after before) :to-sec))
 (setq *passed-time* (+ delta-t *passed-time*))
 (cond ((< *passed-time* 5)
        (format t "before 5sec (~A sec)~%" *passed-time*))
       ((> *passed-time* 5)
        (format t "after 5sec (~A sec)~%" *passed-time*)))
 )
```

[image]

参考
http://wiki.ros.org/ja/ROS/Tutorials/WritingServiceClient(euslisp)

k-okada, 2016-12-22:
かぶったけど、せっかくプログラム書いたので．．．
https://gist.github.com/k-okada/af04ff0f55e0251549c254ca893352b7

ayamiya, 2016-12-22:
ありがとうございます。

closed 2017-03-08 by yuki-asano