jsk-enshu/robot-programming #166: opencv_appsの顔認識プログラムが動きません

M-Onitsuka, 2017-01-03:
11/7の資料の8ページ目，terminalに
```
mech-user@test1-pc:~$ rosrun opencv_apps face_detection image:=image_rect _debug_view:=true __name:=face_detection
```
としたところ
```
[ERROR] [1483454524.440904906]: --Error loading /usr/share/opencv/haarcascades/haarcascade_frontalface_alt.xml
[ERROR] [1483454524.440991712]: --Error loading /usr/share/opencv/haarcascades/haarcascade_eye_tree_eyeglasses.xml
[ WARN] [1483454529.378356860]: '/face_detection' subscribes topics only with child subscribers.
```
と出てしまい顔認識がされませんでした．
```
rostopic echo face_detection/face
```
としても
```
---
header: 
  seq: 587
  stamp: 
    secs: 1483454850
    nsecs: 671584875
  frame_id: camera
faces: []

```
となりました．

k-okada, 2017-01-04:
ファイルがあるか調べて下さい．なければ、インストールされていないので、sudo apt-get install opencv-data
とするとよいですが、これがインストールされていない、という状況はopencvがインストールされていない可能性が高いですが、opencvそのものもは元々インストールされているはずなので、apt-get
install ros-indigo-opencv-apps
したら入っているので、間違えて無理やり削除していたり、opencv3を使おうとしたりしている可能性が高いです．心当たりありませんか．
```
k-okada@p40-yoga:~$ ls -al
/usr/share/opencv/haarcascades/haarcascade_frontalface_alt.xml
-rw-r--r-- 1 root root 919871  1月 28  2014
/usr/share/opencv/haarcascades/haarcascade_frontalface_alt.xml
k-okada@p40-yoga:~$ dpkg -S haarcascade_frontalface_alt.xmlopencv-data:
/usr/share/opencv/haarcascades/haarcascade_frontalface_alt.xml
opencv-data: /usr/share/opencv/haarcascades/haarcascade_frontalface_alt.xml
```

M-Onitsuka, 2017-01-04:
ファイル自体はありましたがディレクトリ名前がOpenCVでした
```
mech-user@test1-pc:~$ ls /usr/share/OpenCV/haarcascades |grep frontalface_alt
haarcascade_frontalface_alt.xml
haarcascade_frontalface_alt2.xml
haarcascade_frontalface_alt_tree.xml
```