jsk-enshu/robot-programming #168: camshift.launchが立ち上がりません

Ill0510, 2017-01-24:
11/14のp12にあるcamshift.lauchを実行しようとすると、以下のエラーが出ます。

... logging to /home/mech-user/.ros/log/07f3c3fe-e1fa-11e6-88e4-e8b1fce891c6/roslaunch-test1-pc-16297.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://test1-pc:37569/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /usb_cam/camera_info_url: file:///home/mech...

NODES
  /
    camshift (jsk_perception/camshiftdemo)
    image_view2 (image_view2/image_view2)
    usb_cam (usb_cam/usb_cam_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
ERROR: cannot launch node of type [usb_cam/usb_cam_node]: usb_cam
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/mech-user/catkin_ws/src/beginner_tutorials
ROS path [2]=/home/mech-user/catkin_ws/src/robot-programming/dynamixel_7dof_arm
ROS path [3]=/home/mech-user/catkin_ws/src/dynamixel_urdf
ROS path [4]=/home/mech-user/catkin_ws/src/robot-programming/turtleboteus
ROS path [5]=/home/mech-user/catkin_ws/src/robot-programming/dxl_armed_turtlebot
ROS path [6]=/home/mech-user/catkin_ws/src/robot-programming/daisya_euslisp_tutorials
ROS path [7]=/opt/ros/indigo/share
ROS path [8]=/opt/ros/indigo/stacks
process[camshift-2]: started with pid [16315]
process[image_view2-3]: started with pid [16316]
^C[image_view2-3] killing on exit
[camshift-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

k-okada, 2017-01-24:
usb_camがないと言っているので
sudo apt-get install ros-indigo-usb-can
してみましょう

Ill0510, 2017-01-24:
解決しました
ありがとうございます

closed 2017-03-08 by yuki-asano