jsk-enshu/robot-programming #189: roscd beginner_tutorials ができません

KeitaIto123, 2017-10-03:
```
mech-user@test1-pc:~$ mkdir -p ~/catkin_ws/src
mech-user@test1-pc:~$ cd catkin_ws/src/
mech-user@test1-pc:~/catkin_ws/src$ catkin create pkg beginner_tutorials --catkin-deps std_msgs geometry_msgs rospy roscpp message_generation message_runtime
Creating package "beginner_tutorials" in "/home/mech-user/catkin_ws/src"...
Created file beginner_tutorials/package.xml
Created file beginner_tutorials/CMakeLists.txt
Created folder beginner_tutorials/include/beginner_tutorials
Created folder beginner_tutorials/src
Successfully created package files in /home/mech-user/catkin_ws/src/beginner_tutorials.
mech-user@test1-pc:~/catkin_ws/src$ cd ..
src
mech-user@test1-pc:~/catkin_ws$ catkin build 
--------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/mech-user/catkin_ws/devel/.private/catkin_tools_prebuild:/opt/ros/kinetic
Workspace:                   /home/mech-user/catkin_ws
--------------------------------------------------------------------------------
Source Space:       [exists] /home/mech-user/catkin_ws/src
Log Space:         [missing] /home/mech-user/catkin_ws/logs
Build Space:        [exists] /home/mech-user/catkin_ws/build
Devel Space:        [exists] /home/mech-user/catkin_ws/devel
Install Space:      [unused] /home/mech-user/catkin_ws/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Finished  <<< catkin_tools_prebuild                [ 3.8 seconds ]             
Starting  >>> beginner_tutorials                                               
Finished  <<< beginner_tutorials                   [ 1.8 seconds ]             
[build] Summary: All 2 packages succeeded!                                     
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 5.6 seconds total.                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.
mech-user@test1-pc:~/catkin_ws$ source devel/setup.sh 
mech-user@test1-pc:~/catkin_ws$ roscd beginner_tutorials
roscd: No such package/stack 'beginner_tutorials'
```
2.3節の説明に沿って進めていったのですが、roscd beginner_tutorials のところで、上記のようなエラーが出てしまい困っています。
対処法がわかる方がいらっしゃれば、ご教示願いたいです。

k-okada, 2017-10-03:
~/catkin_ws で `catkin clean -y` ` catkin build -vi` として結果を貼ってみてください．

KeitaIto123, 2017-10-03:
```
mech-user@test1-pc:~/catkin_ws$ catkin clean -y
[clean] Removing develspace: /home/mech-user/catkin_ws/devel
[clean] Removing buildspace: /home/mech-user/catkin_ws/build
[clean] Removing log space: /home/mech-user/catkin_ws/logs
mech-user@test1-pc:~/catkin_ws$ catkin build -vi
--------------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/mech-user/catkin_ws/devel/.private/catkin_tools_prebuild:/opt/ros/kinetic
Workspace:                   /home/mech-user/catkin_ws
--------------------------------------------------------------------------------
Source Space:       [exists] /home/mech-user/catkin_ws/src
Log Space:         [missing] /home/mech-user/catkin_ws/logs
Build Space:        [exists] /home/mech-user/catkin_ws/build
Devel Space:        [exists] /home/mech-user/catkin_ws/devel
Install Space:      [unused] /home/mech-user/catkin_ws/install
DESTDIR:            [unused] None
--------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting  >>> catkin_tools_prebuild                                            
Starting   >> catkin_tools_prebuild:loadenv                                    
[catkin_tools_prebuild:loadenv] Loading environment from: /home/mech-user/catkin_ws/devel/env.sh
Starting   >> catkin_tools_prebuild:mkdir                                      
Starting   >> catkin_tools_prebuild:mkdir                                      
Starting   >> catkin_tools_prebuild:cache-manifest                             
Starting   >> catkin_tools_prebuild:ctr-nuke                                   
Starting   >> catkin_tools_prebuild:cmake                                      
Subprocess  > catkin_tools_prebuild:cmake `cd /home/mech-user/catkin_ws/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/mech-user/catkin_ws/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mech-user/catkin_ws/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/mech-user/catkin_ws/install; cd -`
[catkin_tools_prebuild:cmake] Not searching for unused variables given on the command line.
[catkin_tools_prebuild:cmake] Re-run cmake no build system arguments            
[catkin_tools_prebuild:cmake] -- The C compiler identification is GNU 5.4.0     
[catkin_tools_prebuild:cmake] -- The CXX compiler identification is GNU 5.4.0   
[catkin_tools_prebuild:cmake] -- Check for working C compiler: /usr/bin/cc      
[catkin_tools_prebuild:cmake] -- Check for working C compiler: /usr/bin/cc -- works
[catkin_tools_prebuild:cmake] -- Detecting C compiler ABI info                  
[catkin_tools_prebuild:cmake] -- Detecting C compiler ABI info - done           
[catkin_tools_prebuild:cmake] -- Detecting C compile features                   
[catkin_tools_prebuild:cmake] -- Detecting C compile features - done            
[catkin_tools_prebuild:cmake] -- Check for working CXX compiler: /usr/bin/c++   
[catkin_tools_prebuild:cmake] -- Check for working CXX compiler: /usr/bin/c++ -- works
[catkin_tools_prebuild:cmake] -- Detecting CXX compiler ABI info                
[catkin_tools_prebuild:cmake] -- Detecting CXX compiler ABI info - done         
[catkin_tools_prebuild:cmake] -- Detecting CXX compile features                 
[catkin_tools_prebuild:cmake] -- Detecting CXX compile features - done          
[catkin_tools_prebuild:cmake] -- Using CATKIN_DEVEL_PREFIX: /home/mech-user/catkin_ws/devel/.private/catkin_tools_prebuild
[catkin_tools_prebuild:cmake] -- Using CMAKE_PREFIX_PATH: /home/mech-user/catkin_ws/devel/.private/catkin_tools_prebuild;/opt/ros/kinetic
[catkin_tools_prebuild:cmake] -- This workspace overlays: /home/mech-user/catkin_ws/devel/.private/catkin_tools_prebuild;/opt/ros/kinetic
[catkin_tools_prebuild:cmake] -- Found PythonInterp: /usr/bin/python (found version "2.7.12")
[catkin_tools_prebuild:cmake] -- Using PYTHON_EXECUTABLE: /usr/bin/python       
[catkin_tools_prebuild:cmake] -- Using Debian Python package layout             
[catkin_tools_prebuild:cmake] -- Using empy: /usr/bin/empy                      
[catkin_tools_prebuild:cmake] -- Using CATKIN_ENABLE_TESTING: ON                
[catkin_tools_prebuild:cmake] -- Call enable_testing()                          
[... 88 lines cut ...]
[beginner_tutorials:make] /usr/bin/cmake -H/home/mech-user/catkin_ws/src/beginner_tutorials -B/home/mech-user/catkin_ws/build/beginner_tutorials --check-build-system CMakeFiles/Makefile.cmake 0
[... cut ...]
[build]   Abandoned: None.                                                     
[build]   Failed:    None.                                                     
[build] Runtime: 7.3 seconds total.                                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.
mech-user@test1-pc:~/catkin_ws$ source devel/setup.sh 
mech-user@test1-pc:~/catkin_ws$ roscd beginner_tutorials
roscd: No such package/stack 'beginner_tutorials'

```
同じエラーが出てしまいました。

k-okada, 2017-10-03:
一回別のターミナルを開いて source /opt/ros/kinetic/setup.bash としてから同じことをするとどうなるでしょうか．

closed 2018-11-15 by tongtybj