jsk-enshu/robot-programming #199: talker2.pyを実行するとエラーが出る

k-sekiya, 2017-10-07:
```
mech-user@test1-pc:~/catkin_ws/src/beginner_tutorials/scripts$ python ./talker2.py 
Traceback (most recent call last):
  File "./talker2.py", line 4, in <module>
    from beginner_tutorials import Hello
ImportError: No module named beginner_tutorials
```
9/25の資料の2.3.3でtalker2.pyを実行しようとするとエラーが出てしまいます。msgディレクトリ以下にHello.msgを作成し、catkin buildもうまく行ったのですがtalker2.py実行がうまく行きません。

k-okada, 2017-10-07:
source ...setup.bash はしましたか，そうでない場合は，コンパイルがうまく行っていないので，
```
catkin clean -y
catkin build -vi
source devel/setup.bash
cd rc/beginner_tutorials/scripts
python ./talker2.py
```
の結果をすべて送ってみてください．

k-sekiya, 2017-10-07:
新しくターミナルを開いた時にsource...setup.bashをしていなかったことが原因でした。
ありがとうございました。

closed 2017-10-07 by k-sekiya