jsk-enshu/robot-programming #221: 10/23資料の１１ページの操作が実行できない

morifuji551, 2017-11-07:
10/23資料p11でMoveIt!を動かそうとして、まず、次のとおりにdenso-launchをインストールしました。
~~~~
sudo apt-get install ros-kinetic-denso-launch
~~~~
そのうえで、プリントにあるように次の文を打ちました。
~~~~
roslaunch denso_launch denso_vs060_moveit_demo_simulation.launch
~~~~
しかし、次のような出力となり、ロボットモデルは表示されませんでした。
~~~~
... logging to /home/mech-user/.ros/log/1b2f66fc-c3a0-11e7-8e28-e8b1fce99346/roslaunch-test1-pc-19539.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 307, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 746, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 718, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 625, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 652, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 625, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 682, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 625, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 678, in _recurse_load
    self._rosparam_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 228, in _rosparam_tag
    cmd, ns, file, param, subst_value = self.opt_attrs(tag, context, (XmlLoader.ROSPARAM_OPT_ATTRS))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 202, in opt_attrs
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: denso_controller
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
~~~~
うまく行かないので、プリントに載っている以下のリンクを確認しました。
~~~~
https://github.com/start-jsk/denso/pull/88
~~~~
しかし、私の理解力不足でそこからどうすればいいのかわからず、一時間以上手詰まりの状態です。
解決策をご存知の方はご教示下さい。

k-okada, 2017-11-08:
https://github.com/start-jsk/denso/pull/88
をみるとmergedとなるので，最新のソースツリーに反映されています．
ということで，catkinワークスペースで
cd src
git clone http://github.com/start-jsk/denso
cd ..
catkin build
とすると最新のプログラムをダウンロードしてコンパイルされるので，
source devel/setup.bash
としてから，roslaunchしてみてください．

morifuji551, 2017-11-08:
お忙しい中、どうもありがとうございます。早速試してみます。

morifuji551, 2017-11-08:
度々申し訳ございません。先ほど教えてくださった方法を試してみました。おかげ様でMoveIt自体は開くことができるようになりました。しかし、ロボットの表示が一部うまくいっていません。
roslaunchをする前までが以下のとおりです。
~~~~
mech-user@test1-pc:~$ cd ~/catkin_ws/src
mech-user@test1-pc:~/catkin_ws/src$ git clone http://github.com/start-jsk/denso
Cloning into 'denso'...
remote: Counting objects: 1951, done.
remote: Compressing objects: 100% (102/102), done.
remote: Total 1951 (delta 96), reused 127 (delta 70), pack-reused 1777
Receiving objects: 100% (1951/1951), 2.39 MiB | 324.00 KiB/s, done.
Resolving deltas: 100% (1328/1328), done.
Checking connectivity... done.
mech-user@test1-pc:~/catkin_ws/src$ cd ..
mech-user@test1-pc:~/catkin_ws$ catkin build
--------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/kinetic
Workspace:                   /home/mech-user/catkin_ws
--------------------------------------------------------------
Source Space:       [exists] /home/mech-user/catkin_ws/src
Log Space:          [exists] /home/mech-user/catkin_ws/logs
Build Space:        [exists] /home/mech-user/catkin_ws/build
Devel Space:        [exists] /home/mech-user/catkin_ws/devel
Install Space:      [unused] /home/mech-user/catkin_ws/install
DESTDIR:            [unused] None
--------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------
[build] Found '12' packages in 0.0 seconds.                                  
[build] Updating package table.                                              
Starting  >>> beginner_tutorials                                             
Starting  >>> denso_ros_control                                              
Starting  >>> dynamixel_7dof_arm                                             
Starting  >>> dynamixel_urdf                                                 
Starting  >>> turtleboteus                                                   
Starting  >>> vs060                                                          
Finished  <<< dynamixel_urdf                          [ 0.2 seconds ]        
Finished  <<< turtleboteus                            [ 0.2 seconds ]        
Finished  <<< beginner_tutorials                      [ 0.5 seconds ]        
Finished  <<< dynamixel_7dof_arm                      [ 0.2 seconds ]        
Starting  >>> dxl_armed_turtlebot                                            
Finished  <<< dxl_armed_turtlebot                     [ 0.2 seconds ]        
Starting  >>> daisya_euslisp_tutorials                                       
Finished  <<< daisya_euslisp_tutorials                [ 0.1 seconds ]        
_____________________________________________________________________________
Warnings   << denso_ros_control:make /home/mech-user/catkin_ws/logs/denso_ros_control/build.make.000.log
cc1: warning: command line option ‘-std=c++11’ is valid for C++/ObjC++ but not for C
In file included from /opt/ros/kinetic/include/hardware_interface/internal/resource_manager.h:40:0,
                 from /opt/ros/kinetic/include/hardware_interface/internal/hardware_resource_manager.h:36,
                 from /opt/ros/kinetic/include/hardware_interface/joint_command_interface.h:35,
                 from /home/mech-user/catkin_ws/src/denso/denso_ros_control/include/denso_ros_control/denso_hardware_interface.h:1,
                 from /home/mech-user/catkin_ws/src/denso/denso_ros_control/src/main.cpp:1:
/home/mech-user/catkin_ws/src/denso/denso_ros_control/include/denso_ros_control/denso_robot.h: In member function ‘BCAP_HRESULT DensoRobot::bCapClearError()’:
/opt/ros/kinetic/include/ros/console.h:346:176: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 9 has type ‘long int’ [-Wformat=]
     ::ros::console::print(filter, __rosconsole_define_location__loc.logger_, __rosconsole_define_location__loc.level_, __FILE__, __LINE__, __ROSCONSOLE_FUNCTION__, __VA_ARGS__)
                                                                                                                                                                                ^
/opt/ros/kinetic/include/ros/console.h:349:5: note: in expansion of macro ‘ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER’
     ROSCONSOLE_PRINT_AT_LOCATION_WITH_FILTER(0, __VA_ARGS__)
[... 298 lines cut ...]
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1510133443.039871227]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1510133443.039888535]: MoveGroup context initialization complete

You can start planning now!

[INFO] [1510133443.733769]: Stopping all controllers...
[INFO] [1510133443.741320]: Unloading all loaded controllers...
[INFO] [1510133443.741573]: Trying to unload position_trajectory_controller
[INFO] [1510133443.755328]: Succeeded in unloading position_trajectory_controller
[INFO] [1510133443.755577]: Trying to unload joint_state_controller
[INFO] [1510133443.758569]: Succeeded in unloading joint_state_controller

~~~~

[image]

k-okada, 2017-11-08:
dpkg -l ros-kinetic-moveit*
としたらなにがでるでしょうか？また，
sudo apt-get update
sudo apt-get install ros-kinetic-moveit*
としてインストールするとどうなるでしょうか？

morifuji551, 2017-11-08:
返信有り難うございます。ひとつ目は、次のような出力となりました。
~~~
mech-user@test1-pc:~/catkin_ws$ dpkg -l ros-kinetic-moveit*
要望=(U)不明/(I)インストール/(R)削除/(P)完全削除/(H)保持
| 状態=(N)無/(I)インストール済/(C)設定/(U)展開/(F)設定失敗/(H)半インストール/(W)トリガ待ち/(T)トリガ保留
|/ エラー?=(空欄)無/(R)要再インストール (状態,エラーの大文字=異常)
||/ 名前           バージョン   アーキテクチ 説明
+++-==============-============-============-=================================
ii  ros-kinetic-mo 0.9.9-0xenia amd64        Python interfaces to MoveIt
ii  ros-kinetic-mo 0.9.4-0xenia amd64        Core libraries used by MoveIt!
ii  ros-kinetic-mo 0.9.4-0xenia amd64        A fake controller manager plugin 
ii  ros-kinetic-mo 0.9.4-0xenia amd64        Package for all inverse kinematic
ii  ros-kinetic-mo 0.9.1-0xenia amd64        Messages, services and actions us
ii  ros-kinetic-mo 0.9.4-0xenia amd64        MoveIt interface to OMPL
ii  ros-kinetic-mo 0.9.9-0xenia amd64        Components of MoveIt used for man
ii  ros-kinetic-mo 0.9.4-0xenia amd64        The move_group node for MoveIt
ii  ros-kinetic-mo 0.9.4-0xenia amd64        Components of MoveIt connecting t
ii  ros-kinetic-mo 0.9.4-0xenia amd64        Planning components of MoveIt tha
ii  ros-kinetic-mo 0.9.9-0xenia amd64        Components of MoveIt that offer s
ii  ros-kinetic-mo 0.9.9-0xenia amd64        Components of MoveIt that offer i
ii  ros-kinetic-mo 0.9.9-0xenia amd64        Components of MoveIt that offer v
ii  ros-kinetic-mo 0.9.9-0xenia amd64        Components of MoveIt connecting t
ii  ros-kinetic-mo 0.9.4-0xenia amd64        A generic, simple controller mana
~~~
一方、
~~~
sudo apt-get update 
sudo apt-get install ros-kinetic-moveit*
~~~
とした結果は以下のようになりました。roslaunchの結果は以前の質問のままです。
~~~
mech-user@test1-pc:~/catkin_ws$ sudo apt-get update
ヒット:1 http://jp.archive.ubuntu.com/ubuntu xenial InRelease
ヒット:2 http://jp.archive.ubuntu.com/ubuntu xenial-backports InRelease        
ヒット:3 https://repo.skype.com/deb stable InRelease                           
ヒット:4 https://repo.skype.com/deb unstable InRelease                         
無視:5 http://dl.google.com/linux/chrome/deb stable InRelease                  
ヒット:6 http://archive.canonical.com/ubuntu xenial InRelease                  
ヒット:7 http://security.ubuntu.com/ubuntu xenial-security InRelease           
ヒット:8 http://dl.google.com/linux/chrome/deb stable Release                  
ヒット:10 http://packages.ros.org/ros/ubuntu saucy InRelease                   
ヒット:11 http://archive.ubuntu.com/ubuntu xenial-updates InRelease
ヒット:12 http://archive.ubuntu.com/ubuntu xenial InRelease     
パッケージリストを読み込んでいます... 完了
mech-user@test1-pc:~/catkin_ws$ sudo apt-get install ros-kinetic-moveit*
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています                
状態情報を読み取っています... 完了
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-fake-controller-manager' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-msgs' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-ros-robot-interaction' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-ros-planning-interface' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-commander' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-kinematics' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-ros-move-group' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-ros-perception' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-simple-controller-manager' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-ros-warehouse' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-ros-manipulation' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-ros-planning' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-planners-ompl' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-ros-visualization' を選択します
注意、glob 'ros-kinetic-moveit*' のために 'ros-kinetic-moveit-core' を選択します
ros-kinetic-moveit-simple-controller-manager はすでに最新バージョン (0.9.4-0xenial-20170227-163306-0800) です。
ros-kinetic-moveit-kinematics はすでに最新バージョン (0.9.4-0xenial-20170227-165944-0800) です。
ros-kinetic-moveit-ros-robot-interaction はすでに最新バージョン (0.9.9-0xenial-20171025-135737-0800) です。
ros-kinetic-moveit-planners-ompl はすでに最新バージョン (0.9.4-0xenial-20170227-165852-0800) です。
ros-kinetic-moveit-ros-planning はすでに最新バージョン (0.9.4-0xenial-20170227-164209-0800) です。
ros-kinetic-moveit-ros-warehouse はすでに最新バージョン (0.9.9-0xenial-20171025-140436-0800) です。
ros-kinetic-moveit-commander はすでに最新バージョン (0.9.9-0xenial-20171025-142859-0800) です。
[... 34 lines cut ...]
  ros-kinetic-perception ros-kinetic-perception-pcl ros-kinetic-polled-camera
  ros-kinetic-position-controllers ros-kinetic-ros-control
  ros-kinetic-ros-controllers ros-kinetic-rospy-message-converter
  ros-kinetic-rqt-joint-trajectory-controller ros-kinetic-stdr-gui
  ros-kinetic-stdr-msgs ros-kinetic-stdr-parser ros-kinetic-stdr-resources
  ros-kinetic-stdr-robot ros-kinetic-stdr-server ros-kinetic-stereo-image-proc
  ros-kinetic-tf2-geometry-msgs ros-kinetic-theora-image-transport
  ros-kinetic-turtlebot-stage ros-kinetic-turtlebot-stdr
  ros-kinetic-velocity-controllers ros-kinetic-vision-opencv
  ros-kinetic-yocs-math-toolkit ros-kinetic-yocs-msgs
  ros-kinetic-yocs-virtual-sensor texlive-bibtex-extra texlive-fonts-extra
  texlive-fonts-extra-doc texlive-formats-extra texlive-games
  texlive-generic-extra texlive-htmlxml texlive-humanities
  texlive-humanities-doc texlive-lang-african texlive-lang-arabic
  texlive-lang-cyrillic texlive-lang-czechslovak texlive-lang-english
  texlive-lang-european texlive-lang-german texlive-lang-greek
  texlive-lang-indic texlive-lang-italian texlive-lang-polish
  texlive-lang-portuguese texlive-lang-spanish texlive-luatex
  texlive-math-extra texlive-music texlive-omega texlive-plain-extra
  texlive-publishers texlive-publishers-doc texlive-science
  texlive-science-doc ttf-adf-accanthis ttf-adf-gillius ttf-adf-universalis
これを削除するには 'sudo apt autoremove' を利用してください。
アップグレード: 0 個、新規インストール: 0 個、削除: 0 個、保留: 493 個。

~~~

morifuji551, 2017-11-13:
今日の授業の貴重なお時間を使って解決していただきありがとうございました。
おかげさまで全て動くようになりました。
今後もよろしくお願いいたします。

closed 2017-11-13 by morifuji551