jsk-enshu/robot-programming #222: AnacondaとROSが闘っている？

u-0N0, 2017-11-09:
お忙しいところ失礼します。

10月23日分資料のp.10の
```
$ ~/vision_ws/src
$ catkin b
```
を実行したところエラーが次のように表示されました。
```
mech-user@test1-pc:~/3A/robosys/vision_ws$ catkin b
==> Expanding alias 'b' from 'catkin b' to 'catkin build'
-----------------------------------------------------------------------------
Profile:                     default
Extending:             [env] /home/mech-user/catkin_ws/devel:/opt/ros/kinetic
Workspace:                   /home/mech-user/3A/robosys/vision_ws
-----------------------------------------------------------------------------
Source Space:       [exists] /home/mech-user/3A/robosys/vision_ws/src
Log Space:         [missing] /home/mech-user/3A/robosys/vision_ws/logs
Build Space:        [exists] /home/mech-user/3A/robosys/vision_ws/build
Devel Space:        [exists] /home/mech-user/3A/robosys/vision_ws/devel
Install Space:      [unused] /home/mech-user/3A/robosys/vision_ws/install
DESTDIR:            [unused] None
-----------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
-----------------------------------------------------------------------------
[build] Found '1' packages in 0.0 seconds.                                         
[build] Updating package table.                                                    
Starting  >>> catkin_tools_prebuild                                                
___________________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /home/mech-user/3A/robosys/vision_ws/logs/catkin_tools_prebuild/build.cmake.000.log
ImportError: "from catkin_pkg.package import parse_package" failed: No module named catkin_pkg.package
Make sure that you have installed "catkin_pkg", it is up to date and on the PYTHONPATH.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/home/mech-user/anaconda3/envs/py2/bin/python
  "/opt/ros/kinetic/share/catkin/cmake/parse_package_xml.py"
  "/opt/ros/kinetic/share/catkin/cmake/../package.xml"
  "/home/mech-user/3A/robosys/vision_ws/build/catkin_tools_prebuild/catkin/catkin_generated/version/package.cmake")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package_xml.cmake:63 (safe_execute_process)
  /opt/ros/kinetic/share/catkin/cmake/all.cmake:151 (_catkin_package_xml)
  /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:4 (find_package)

cd /home/mech-user/3A/robosys/vision_ws/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/mech-user/3A/robosys/vision_ws/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mech-user/3A/robosys/vision_ws/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/mech-user/3A/robosys/vision_ws/install; cd -
...................................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]          
Failed    <<< catkin_tools_prebuild                [ 5.1 seconds ]                 
Abandoned <<< my_cv_pkg                            [ Unrelated job failed ]        
[build] Summary: 0 of 2 packages succeeded.                                        
[build]   Ignored:   None.                                                         
[build]   Warnings:  None.                                                         
[build]   Abandoned: 1 packages were abandoned.                                    
[build]   Failed:    1 packages failed.                                            
[build] Runtime: 5.2 seconds total.                                                
```
先日の授業後にも謎のトラブルを先生に診断していただきまして、Anacondaを入れたせいだということがわかったのでそれ以来は必要に応じて.bashrcの設定でAnacondaのPATHを通したり外したりしていました。

Errorメッセージを確認する限りおそらく今回もそうだろうと思ってちゃんと.bashrcを確認したのですがちゃんとPATHは外してありました。

念の為
```
$ source ~/.bashrc
```
としてみたのですが、何も変わりませんでした。解決する方法ありましたらぜひお教えください。

u-0N0, 2017-11-11:
解決しました。

http://ossyaritoori.hatenablog.com/entry/2017/09/14/ROS_%E3%82%A2%E3%83%B3%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB%E5%BE%8C%EF%BC%8C%E5%86%8D%E3%82%A4%E3%83%B3%E3%82%B9%E3%83%88%E3%83%BC%E3%83%AB%EF%BC%88Anaconda%E3%81%A8%E3%81%AE

こちらのサイトを参考にしつつ、PATHからanacondaをまるっきり消去し、
```
$ rm -rf ~/catkin_ws/devel ~/catkin_ws/build
```
をおこない、同様にして
```
$ rm -rf vision_ws/devel vision_ws/build vision_ws/log
```
を実行したところ、エラーなくビルドすることができました。

お騒がせしました。

closed 2017-11-11 by u-0N0

k-okada, 2017-11-11:
ごめんなさい．見逃していました．そうです．AnacondaはROSというか，標準のubuntu のpythonに異存しているシステムと共存できないです．なので普段は以下のようにコメントアウトして，必要なときだけanacondaのpathをterminalでセットして使ってください．
http://wiki.ros.org/IDEs#Anaconda