jsk-enshu/robot-programming #223: 2017/10/23資料5ページのrosrun image_proc image_proc実行時エラーについて

future731, 2017-11-12:
10/23資料5ページのrosrun image_proc image_procを実行する際に  
~ $ rosrun image_proc image_proc
[ WARN] [1510492900.700944342]: Started in the global namespace! This is probably wrong. Start image_proc in the camera namespace.
Example command-line usage:
        $ ROS_NAMESPACE=my_camera rosrun image_proc image_proc
terminate called after throwing an instance of 'pluginlib::InvalidXMLException'
  what():  XML Document has no Root Element. This likely means the XML is malformed or missing.
中止 (コアダンプ)

と出てしまって強制終了してしまいます。このエラーはcamera.launchファイルを作成して、

<launch>
    <node pkg="uvc_camera" type="uvc_camera_node" name="camera_driver" />
    <node pkg="image_proc" type="image_proc" name="image_proc" >
        <remap from="image" to="image_raw" />
    </node>
    <node pkg="image_view" type="image_view" name="image_view" >
        <remap from="image" to="image_rect_color" />
    </node>
</launch>

としてstdoutログを確認しても出ます。
その結果、rosrun image_view image_viewにおいてimage:=image_rect_colorとしても、そのようなノードが存在しないため、ウィンドウが固まってままになります。

  
$ rostopic list
の実行結果は
^C~ $ rostopic list
/camera_info
/image_raw
/image_raw/compressed
/image_raw/compressed/parameter_descriptions
/image_raw/compressed/parameter_updates
/image_raw/compressedDepth
/image_raw/compressedDepth/parameter_descriptions
/image_raw/compressedDepth/parameter_updates
/image_raw/theora
/image_raw/theora/parameter_descriptions
/image_raw/theora/parameter_updates
/rosout
/rosout_agg
です。

google検索して調べたところ、
https://answers.ros.org/question/204751/how-to-use-image_proc-node/
がヒットして、これによると/camera_info及び/image_rawがあるのでうまく行くのではないかと思いましたが、うまくいきません。

どういった原因が考えられるでしょうか。

ちなみに、環境変数については以下のようになっていました。
catkin_ws[master] $ export | grep ROS
declare -x ROSLISP_PACKAGE_DIRECTORIES="/home/future731/private/lecture2017/catkin_ws/devel/share/common-lisp"
declare -x ROS_DISTRO="kinetic"
declare -x ROS_ETC_DIR="/opt/ros/kinetic/etc/ros"
declare -x ROS_MASTER_URI="http://localhost:11311"
declare -x ROS_PACKAGE_PATH="/home/future731/private/lecture2017/catkin_ws/src/beginner_tutorials:/home/future731/private/lecture2017/catkin_ws/src/robot-programming/dynamixel_7dof_arm:/home/future731/private/lecture2017/catkin_ws/src/dynamixel_urdf:/home/future731/private/lecture2017/catkin_ws/src/rtmros_tutorials/hironx_tutorial:/home/future731/private/lecture2017/catkin_ws/src/rtmros_common/hrpsys_tools:/home/future731/private/lecture2017/catkin_ws/src/2017-semi/kondo_driver:/home/future731/private/lecture2017/catkin_ws/src/2017-semi/kondo_models:/home/future731/private/lecture2017/catkin_ws/src/2017-semi/kondo_ros_bridge:/home/future731/private/lecture2017/catkin_ws/src/rtmros_tutorials/openhrp3_tutorials:/home/future731/private/lecture2017/catkin_ws/src/rtmros_common/openrtm_tools:/home/future731/private/lecture2017/catkin_ws/src/rtmros_tutorials/hrpsys_tutorials:/home/future731/private/lecture2017/catkin_ws/src/rtmros_common/rosnode_rtc:/home/future731/private/lecture2017/catkin_ws/src/rtmros_common/rtmbuild:/home/future731/private/lecture2017/catkin_ws/src/rtmros_common/hrpsys_ros_bridge:/home/future731/private/lecture2017/catkin_ws/src/2017-semi/hrpsys_kondo_iob:/home/future731/private/lecture2017/catkin_ws/src/rtmros_tutorials/hrpsys_ros_bridge_tutorials:/home/future731/private/lecture2017/catkin_ws/src/rtmros_tutorials/hrpsys_gazebo_tutorials:/home/future731/private/lecture2017/catkin_ws/src/rtmros_common/openrtm_ros_bridge:/home/future731/private/lecture2017/catkin_ws/src/rtmros_common/rtmros_common:/home/future731/private/lecture2017/catkin_ws/src/robot-programming/turtleboteus:/home/future731/private/lecture2017/catkin_ws/src/robot-programming/dxl_armed_turtlebot:/home/future731/private/lecture2017/catkin_ws/src/robot-programming/daisya_euslisp_tutorials:/opt/ros/kinetic/share"
declare -x ROS_ROOT="/opt/ros/kinetic/share/ros"

k-okada, 2017-11-12:
camera.launch のファイルの中身とそれを`roslaunch --screen camera.launch`として実行したときの出力を教えてください．

future731, 2017-11-13:
${HOME}/private/lecture2017/catkin_ws/src/beginner_tutorials/launch/camera.launchの中身は

<launch>
    <node pkg="uvc_camera" type="uvc_camera_node" name="camera_driver" />
    <node pkg="image_proc" type="image_proc" name="image_proc" >
        <remap from="image" to="image_raw" />
    </node>
    <node pkg="image_view" type="image_view" name="image_view" >
        <remap from="image" to="image_rect_color" />
    </node>
</launch>

です。当該のファイルがあるディレクトリで
$ roslaunch camera.launch
したところ

launch[master] $ roslaunch camera.launch
... logging to /home/future731/.ros/log/56d2e0da-c805-11e7-b16e-e8b1fce891d5/roslaunch-future731-20395.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://future731:39992/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.11

NODES
  /
    camera_driver (uvc_camera/uvc_camera_node)
    image_proc (image_proc/image_proc)
    image_view (image_view/image_view)

auto-starting new master
process[master]: started with pid [20408]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 56d2e0da-c805-11e7-b16e-e8b1fce891d5
process[rosout-1]: started with pid [20421]
started core service [/rosout]
process[camera_driver-2]: started with pid [20425]
process[image_proc-3]: started with pid [20439]
process[image_view-4]: started with pid [20440]
terminate called after throwing an instance of 'pluginlib::InvalidXMLException'
  what():  XML Document has no Root Element. This likely means the XML is malformed or missing.
[image_proc-3] process has died [pid 20439, exit code -6, cmd /opt/ros/kinetic/lib/image_proc/image_proc image:=image_raw __name:=image_proc __log:=/home/future731/.ros/log/56d2e0da-c805-11e7-b16e-e8b1fce891d5/image_proc-3.log].
log file: /home/future731/.ros/log/56d2e0da-c805-11e7-b16e-e8b1fce891d5/image_proc-3*.log

となりました。
その状態で別のウィンドウを開いて

launch[master] $ roslaunch --screen camera.launch
... logging to /home/future731/.ros/log/7fd96670-c805-11e7-b16e-e8b1fce891d5/roslaunch-future731-21027.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://future731:35944/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.11

NODES
  /
    camera_driver (uvc_camera/uvc_camera_node)
    image_proc (image_proc/image_proc)
    image_view (image_view/image_view)

auto-starting new master
process[master]: started with pid [21040]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7fd96670-c805-11e7-b16e-e8b1fce891d5
process[rosout-1]: started with pid [21053]
started core service [/rosout]
process[camera_driver-2]: started with pid [21057]
process[image_proc-3]: started with pid [21071]
process[image_view-4]: started with pid [21072]
[ WARN] [1510531160.927513663]: Started in the global namespace! This is probably wrong. Start image_proc in the camera namespace.
Example command-line usage:
        $ ROS_NAMESPACE=my_camera rosrun image_proc image_proc
[ INFO] [1510531160.997627786]: using default calibration URL
[ INFO] [1510531160.997690466]: camera calibration URL: file:///home/future731/.ros/camera_info/camera.yaml
[ WARN] [1510531160.998276475]: [camera] does not match name 08144361026320a0 in file /home/future731/.ros/camera_info/camera.yaml
terminate called after throwing an instance of 'pluginlib::InvalidXMLException'
  what():  XML Document has no Root Element. This likely means the XML is malformed or missing.
opening /dev/video0
[ INFO] [1510531161.061724012]: Using transport "raw"
pixfmt 0 = 'YUYV' desc = 'YUYV 4:2:2'
  discrete: 640x480:   1/30 1/15
  discrete: 320x180:   1/30 1/15
  discrete: 320x240:   1/30 1/15
  discrete: 352x288:   1/30 1/15
  discrete: 424x240:   1/30 1/15
  discrete: 640x360:   1/30 1/15
  discrete: 848x480:   1/20
  discrete: 960x540:   1/15
  discrete: 1280x720:   1/10
pixfmt 1 = 'MJPG' desc = 'Motion-JPEG'
  discrete: 640x480:   1/30 1/15
  discrete: 320x180:   1/30 1/15
  discrete: 320x240:   1/30 1/15
  discrete: 352x288:   1/30 1/15
  discrete: 424x240:   1/30 1/15
  discrete: 640x360:   1/30 1/15
  discrete: 848x480:   1/30 1/15
  discrete: 960x540:   1/30 1/15
  discrete: 1280x720:   1/30 1/15
  int (Brightness, 0, id = 980900): 0 to 255 (1)
  int (Contrast, 0, id = 980901): 0 to 255 (1)
  int (Saturation, 0, id = 980902): 0 to 100 (1)
  int (Hue, 0, id = 980903): -180 to 180 (1)
  bool (White Balance Temperature, Auto, 0, id = 98090c): 0 to 1 (1)
  int (Gamma, 0, id = 980910): 90 to 150 (1)
  menu (Power Line Frequency, 0, id = 980918): 0 to 2 (1)
    0: Disabled
    1: 50 Hz
    2: 60 Hz
  int (White Balance Temperature, 16, id = 98091a): 2800 to 6500 (1)
  int (Sharpness, 0, id = 98091b): 0 to 7 (1)
  int (Backlight Compensation, 0, id = 98091c): 0 to 2 (1)
  menu (Exposure, Auto, 0, id = 9a0901): 0 to 3 (1)
  int (Exposure (Absolute), 16, id = 9a0902): 4 to 1250 (1)
  bool (Exposure, Auto Priority, 0, id = 9a0903): 0 to 1 (1)
[image_proc-3] process has died [pid 21071, exit code -6, cmd /opt/ros/kinetic/lib/image_proc/image_proc image:=image_raw __name:=image_proc __log:=/home/future731/.ros/log/7fd96670-c805-11e7-b16e-e8b1fce891d5/image_proc-3.log].
log file: /home/future731/.ros/log/7fd96670-c805-11e7-b16e-e8b1fce891d5/image_proc-3*.log

となりました。

future731, 2017-11-13:
/etc/apt/sources.list.d/ros-latest.listに
deb http://packages.ros.org/ros-shadow-fixed/ubuntu xenial main
を追加していたため、最新版でXMLファイルがないとプログラムが落ちる仕様になっていたのが原因でした。
当面/opt/ros/kinetic/shared/micros_swarm_framework/package.xmlの62,63行目にあった
 61   <export>
 62     
 63     
 64   </export>
とこの2行をコメントアウトすることで対応することになりました。

closed 2017-11-13 by future731