jsk-enshu/robot-programming #231: ロボットシステム9/25資料の１０ページの操作が実行できない

morifuji551, 2017-12-19:
ロボットシステム9/25の資料を復習中、演習用ロボットのシミュレーターを起動しようとしました。
まず、
~~~~
source ~/catkin_ws/devel/setup.bash
~~~~
としたうえで、
~~~~
roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch 
~~~~
と打ち込みましたが、次のようなエラーが出ました。
~~~
... logging to /home/mech-user/.ros/log/a529947e-e4b4-11e7-85af-e8b1fce99346/roslaunch-test1-pc-3257.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/share/xacro/xacro.py", line 55, in <module>
    import xacro
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 35, in <module>
    import glob
  File "/home/mech-user/.pyenv/versions/3.6.0/lib/python3.6/glob.py", line 4, in <module>
    import re
ModuleNotFoundError: No module named 're'
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/mech-user/catkin_ws/src/robot-programming/dxl_armed_turtlebot/urdf/robot.urdf.xacro] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py $(arg model)" name="robot_description"/>
The traceback for the exception was written to the log file
~~~
Googleで調べてみると、pythonのバージョンが関係しているように思えたので、以下のようにpythonのバージョンを切り替えました。
~~~~
$ python -V
Python 3.6.0
$ pyenv versions
  system
  2.7.13
* 3.6.0 (set by /home/mech-user/.pyenv/version)
$ pyenv global 2.7.13
$ python -V
Python 2.7.13
~~~~

この上でもう一度試すと次のエラーが出てきました。（エラーの内容が若干変わりました。）
~~~~
... logging to /home/mech-user/.ros/log/f9e0928c-e4b5-11e7-85af-e8b1fce99346/roslaunch-test1-pc-16351.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/share/xacro/xacro.py", line 55, in <module>
    import xacro
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 42, in <module>
    from roslaunch import substitution_args
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 48, in <module>
    import rospkg
ImportError: No module named rospkg
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/mech-user/catkin_ws/src/robot-programming/dxl_armed_turtlebot/urdf/robot.urdf.xacro] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py $(arg model)" name="robot_description"/>
The traceback for the exception was written to the log file
~~~~
今までうまくいっていたのですが、今日試すとエラーが出るようになり、独力では原因がつかめない状況です。
解決策をご存知の方はご教示下さい。

k-okada, 2017-12-19:
Pyenvを入れてしまいましたね
Systemのpythonでないと動かないので
Pyenvを外すことはできますか？
できない場合はbashrcをpasteしてみてください

morifuji551, 2017-12-19:
ご返信ありがとうございます。
pyenvの外し方をしらべてみましたが、まだ情報が見つかっていないので、まず.bashrcを確認したところ、次のようになっていました。
~~~~
# ~/.bashrc: executed by bash(1) for non-login shells.
# see /usr/share/doc/bash/examples/startup-files (in the package bash-doc)
# for examples

# If not running interactively, don't do anything
case $- in
    *i*) ;;
      *) return;;
esac

# don't put duplicate lines or lines starting with space in the history.
# See bash(1) for more options
HISTCONTROL=ignoreboth

# append to the history file, don't overwrite it
shopt -s histappend

# for setting history length see HISTSIZE and HISTFILESIZE in bash(1)
HISTSIZE=1000
HISTFILESIZE=2000

# check the window size after each command and, if necessary,
# update the values of LINES and COLUMNS.
shopt -s checkwinsize

# If set, the pattern "**" used in a pathname expansion context will
# match all files and zero or more directories and subdirectories.
#shopt -s globstar

# make less more friendly for non-text input files, see lesspipe(1)
[ -x /usr/bin/lesspipe ] && eval "$(SHELL=/bin/sh lesspipe)"

# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, unless we know we "want" color)
case "$TERM" in
    xterm-color) color_prompt=yes;;
esac

# uncomment for a colored prompt, if the terminal has the capability; turned
# off by default to not distract the user: the focus in a terminal window
# should be on the output of commands, not on the prompt
#force_color_prompt=yes

if [ -n "$force_color_prompt" ]; then
    if [ -x /usr/bin/tput ] && tput setaf 1 >&/dev/null; then
	# We have color support; assume it's compliant with Ecma-48
	# (ISO/IEC-6429). (Lack of such support is extremely rare, and such
	# a case would tend to support setf rather than setaf.)
	color_prompt=yes
    else
	color_prompt=
    fi
fi

if [ "$color_prompt" = yes ]; then
    PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
else
    PS1='${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '
fi
unset color_prompt force_color_prompt

# If this is an xterm set the title to user@host:dir
case "$TERM" in
xterm*|rxvt*)
    PS1="\[\e]0;${debian_chroot:+($debian_chroot)}\u@\h: \w\a\]$PS1"
    ;;
*)
    ;;
esac

# enable color support of ls and also add handy aliases
if [ -x /usr/bin/dircolors ]; then
    test -r ~/.dircolors && eval "$(dircolors -b ~/.dircolors)" || eval "$(dircolors -b)"
    alias ls='ls --color=auto'
    #alias dir='dir --color=auto'
    #alias vdir='vdir --color=auto'

    alias grep='grep --color=auto'
    alias fgrep='fgrep --color=auto'
    alias egrep='egrep --color=auto'
fi

# some more ls aliases
alias ll='ls -alF'
alias la='ls -A'
alias l='ls -CF'

# Add an "alert" alias for long running commands.  Use like so:
#   sleep 10; alert
alias alert='notify-send --urgency=low -i "$([ $? = 0 ] && echo terminal || echo error)" "$(history|tail -n1|sed -e '\''s/^\s*[0-9]\+\s*//;s/[;&|]\s*alert$//'\'')"'

# Alias definitions.
# You may want to put all your additions into a separate file like
# ~/.bash_aliases, instead of adding them here directly.
# See /usr/share/doc/bash-doc/examples in the bash-doc package.

if [ -f ~/.bash_aliases ]; then
    . ~/.bash_aliases
fi

# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi
source /opt/ros/kinetic/setup.bash
export PATH="$HOME/.rbenv/bin:$PATH"
eval "$(rbenv init -)"
export PATH="$HOME/.rbenv/plugins/ruby-build/bin:$PATH"
alias matlab='export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/1stday/realworldrecog/mexopencv/+cv/private/; LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6:/usr/lib/x86_64-linux-gnu/libgfortran.so.3 matlab'
unset GTK_IM_MODULE

source ~/android_core /devel/setup.bash
source rospack find jsk_tools/src/bashrc.ros
rossetmaster 192.168.10.5
rossetip 192.168.10.4
export TURTLEBOT_3D_SENSOR=kinect
export TURTLEBOT_3D_SENSOR=kinect
export TURTLEBOT_3D_SENSOR=kinect
export PYENV_ROOT="$HOME/.pyenv"
export PATH="$PYENV_ROOT/bin:$PATH"
eval "$(pyenv init -)"
~~~~
よろしくお願いいたします。

k-okada, 2017-12-19:
```
export PYENV_ROOT="$HOME/.pyenv"
export PATH="$PYENV_ROOT/bin:$PATH"
eval "$(pyenv init -)"
```
の三行をコメントアウトしましょう．

morifuji551, 2017-12-19:
ご返信有り難うございます。該当箇所をコメントアウトし、
~~~~
$ roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch 
~~~~
としましたが、反応はなく、何も出てきませんでした。

そこで、Ctrl+Cとすると、以前と同じエラーが出ました。
~~~~
[dxl_armed_turtlebot_gazebo.launch] is neither a launch file in package [dxl_armed_turtlebot] nor is [dxl_armed_turtlebot] a launch file name
The traceback for the exception was written to the log file
[http://192.168.10.5:11311][192.168.10.4] mech-user@test1-pc:~$ source ~/catkin_ws/devel/setup.bash
[http://192.168.10.5:11311][192.168.10.4] mech-user@test1-pc:~$ roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch 
^C... logging to /home/mech-user/.ros/log/1f1dd7c6-e4ca-11e7-85af-e8b1fce99346/roslaunch-test1-pc-12845.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/share/xacro/xacro.py", line 55, in <module>
    import xacro
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 42, in <module>
    from roslaunch import substitution_args
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 48, in <module>
    import rospkg
ImportError: No module named rospkg
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/mech-user/catkin_ws/src/robot-programming/dxl_armed_turtlebot/urdf/robot.urdf.xacro] returned with code [1]. 

Param xml is <param command="$(find xacro)/xacro.py $(arg model)" name="robot_description"/>
The traceback for the exception was written to the log file
~~~~
度々ご迷惑をお掛けして申し訳ありません。よろしくお願いいたします。

k-okada, 2017-12-19:
エラーの内容はちがっていますよね．

/opt/ros/kinetic/share/xacro/xacro.py
/home/mech-user/catkin_ws/src/robot-programming/dxl_armed_turtlebot/urdf/robot.urdf.xacro

とあるので， robot.urdf.xacro, 編集しちゃっていないか確認しながら，このコマンドをうってみましょう．

morifuji551, 2017-12-19:
ありがとうございます。
早速試してみます。夜晩遅くまで申し訳ありませんでした。

morifuji551, 2017-12-20:
返信が遅れまして申し訳ありません。
昨日のコマンドを打ってみると、次のようになりました。
~~~~
$ /opt/ros/kinetic/share/xacro/xacro.py
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/xacro/xacro.py", line 55, in <module>
    import xacro
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/xacro/__init__.py", line 42, in <module>
    from roslaunch import substitution_args
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 48, in <module>
    import rospkg
ImportError: No module named rospkg
~~~~
2つのうち、１つ目の方でrospkgがないというエラーが出ました。
~~~~
$ /home/mech-user/catkin_ws/src/robot-programming/dxl_armed_turtlebot/urdf/robot.urdf.xacro 
bash: /home/mech-user/catkin_ws/src/robot-programming/dxl_armed_turtlebot/urdf/robot.urdf.xacro: 許可がありません
~~~~
２つ目の方で、許可がありませんという内容のエラーが出たので、権限を確認しました。しかし、rootとなっているものはありませんでした。
~~~~
$ ll
合計 24
drwxrwxr-x  2 mech-user mech-user  4096  9月 28 09:00 ./
drwxrwxr-x 10 mech-user mech-user  4096 11月 28 13:20 ../
-rw-rw-r--  1 mech-user mech-user 11416  9月 28 09:00 dynamixel_7dof_arm.urdf.xacro
-rw-rw-r--  1 mech-user mech-user   752  9月 28 09:00 robot.urdf.xacro
~~~~
念のため、それぞれのコードを確認したところ、次のようになりました。

[xacro.py]
~~~~
#! /usr/bin/env python
# Copyright (c) 2015, Open Source Robotics Foundation, Inc.
# Copyright (c) 2013, Willow Garage, Inc.
# All rights reserved.
#
# Redistribution and use in source and binary forms, with or without
# modification, are permitted provided that the following conditions are met:
#
#     * Redistributions of source code must retain the above copyright
#       notice, this list of conditions and the following disclaimer.
#     * Redistributions in binary form must reproduce the above copyright
#       notice, this list of conditions and the following disclaimer in the
#       documentation and/or other materials provided with the distribution.
#     * Neither the name of the Open Source Robotics Foundation, Inc. nor the
#       names of its contributors may be used to endorse or promote products
#       derived from this software without specific prior written permission.
#
# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
# AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
# IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE
# ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT OWNER OR CONTRIBUTORS BE
# LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR
# CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF
# SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS
# INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN
# CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE)
# ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
# POSSIBILITY OF SUCH DAMAGE.

# Authors: Stuart Glaser, William Woodall, Robert Haschke
# Maintainer: Morgan Quigley <morgan@osrfoundation.org>

#### READ THIS! THIS FILE ONLY FOR BACKWARDS COMPATIBILITY!!!!!!
#
# this script is just for backwards compatibility. the roslaunch $(find ...)
# expects the script to be in the package directory, and in catkin
# the package direcory is the share directory. going forward, we should
# transition to scripts/xacro

from __future__ import print_function

# Guard against self import
import os
import sys
backup_path = list(sys.path)
this_dir = os.path.dirname(__file__)
# Use os.getcwd() to avoid weird symbolic link problems
cur_dir = os.getcwd()
os.chdir(this_dir)
this_dir_cwd = os.getcwd()
os.chdir(cur_dir)
# Remove this dir from path
sys.path = filter(lambda a: a not in [this_dir, this_dir_cwd], sys.path)

import xacro
from xacro.color import warning

# Restore the path
sys.path = backup_path

xacro.main()

# issue deprecation warning
warning('xacro.py is deprecated; please use xacro instead')
~~~~
[robot.urdf.xacro]
~~~~
<robot name="dxl_armed_turtlebot">
<xacro:include filename="$(find kobuki_description)/urdf/kobuki.urdf.xacro"/>
<xacro:include filename="$(find turtlebot_description)/urdf/stacks/hexagons.urdf.xacro"/>
<xacro:include filename="$(find turtlebot_description)/urdf/sensors/kinect.urdf.xacro"/>
<xacro:include filename="$(find dxl_armed_turtlebot)/urdf/dynamixel_7dof_arm.urdf.xacro"/>
<kobuki/>
<stack_hexagons parent="base_link"/>
<sensor_kinect parent="base_link"/>
<dynamixel_7dof_arm/>
<joint name="joint0" type="fixed">
<parent link="plate_top_link"/>
<child link="arm_base_link"/>
<origin xyz="0.15 0 0.025" rpy="0 0 0"/>
</joint>
</robot>
~~~~
Pythonをデフォルトのバージョンに戻すことも並行して試していますが、成功しておらず、まだ解決に至っていません。ご迷惑をお掛けして申し訳ありません。

k-okada, 2017-12-20:
いまのpythonは何がつかわれていますか? python -v で確認してみてください．

morifuji551, 2017-12-20:
ご返信有り難うございます。
~~~~
$ python -V
Python 2.7.6
~~~~
python2.7.6でした。

k-okada, 2017-12-20:
小文字のvです

morifuji551, 2017-12-20:
失礼しました。次のような出力となりました。
~~~~
$ python -v
# installing zipimport hook
import zipimport # builtin
# installed zipimport hook
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/site.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/site.py
import site # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/site.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/os.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/os.py
import os # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/os.pyc
import errno # builtin
import posix # builtin
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/posixpath.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/posixpath.py
import posixpath # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/posixpath.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/stat.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/stat.py
import stat # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/stat.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/genericpath.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/genericpath.py
import genericpath # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/genericpath.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/warnings.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/warnings.py
import warnings # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/warnings.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/linecache.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/linecache.py
import linecache # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/linecache.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/types.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/types.py
import types # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/types.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/UserDict.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/UserDict.py
import UserDict # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/UserDict.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_abcoll.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_abcoll.py
import _abcoll # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_abcoll.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/abc.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/abc.py
import abc # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/abc.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_weakrefset.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_weakrefset.py
import _weakrefset # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_weakrefset.pyc
import _weakref # builtin
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/copy_reg.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/copy_reg.py
import copy_reg # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/copy_reg.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/traceback.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/traceback.py
import traceback # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/traceback.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sysconfig.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sysconfig.py
import sysconfig # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sysconfig.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/re.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/re.py
import re # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/re.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_compile.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_compile.py
import sre_compile # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_compile.pyc
import _sre # builtin
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_parse.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_parse.py
import sre_parse # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_parse.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_constants.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_constants.py
import sre_constants # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/sre_constants.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_sysconfigdata.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_sysconfigdata.py
import _sysconfigdata # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/_sysconfigdata.pyc
import encodings # directory /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/encodings
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/encodings/__init__.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/encodings/__init__.py
import encodings # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/encodings/__init__.pyc
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/codecs.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/codecs.py
[... cut ...]
# /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/encodings/utf_8.pyc matches /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/encodings/utf_8.py
import encodings.utf_8 # precompiled from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/encodings/utf_8.pyc
Python 2.7.6 (default, Dec 19 2017, 23:47:34) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
dlopen("/home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/lib-dynload/readline.so", 2);
import readline # dynamically loaded from /home/mech-user/.pyenv/versions/2.7.6/lib/python2.7/lib-dynload/readline.so
>>> 
~~~~

morifuji551, 2017-12-20:
申し訳ありません。友達に教えてもらったところ、単純に
~~~~
pyenv global system
~~~~
とすることで、pythonがデフォルトのバージョンになり、全て解決しました。
不勉強でこの対処法を知りませんでした。ご迷惑をお掛けして申し訳ありませんでした。

k-okada, 2017-12-20:
まだpyenvの設定が残っているのでbashrc を確認して見て下さい

morifuji551, 2017-12-20:
ご返信有り難うございます。
確認したところ、pyenv関連の３行はコメントアウト済でした。
~~~
# ~/.bashrc: executed by bash(1) for non-login shells.
# see 
# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, unless we know we "want" color)
case "$TERM" in
    xterm-color) color_prompt=yes;;
esac

# uncomment for a colored prompt, if the terminal has the capability; turned
# off by default to not distract the user: the focus in a terminal window
# should be on the output of commands, not on the prompt
#force_color_prompt=yes

if [ -n "$force_color_prompt" ]; then
    if [ -x /usr/bin/tput ] && tput setaf 1 >&/dev/null; then
	# We have color support; assume it's compliant with Ecma-48
	# (ISO/IEC-6429). (Lack of such support is extremely rare, and such
	# a case would tend to support setf rather than setaf.)
	color_prompt=yes
    else
	color_prompt=
    fi
fi

if [ "$color_prompt" = yes ]; then
    PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
else
    PS1='${debian_chroot:+($debian_chroot)}\u@\h:\w\$ '
fi
unset color_prompt force_color_prompt

# If this is an xterm set the title to user@host:dir
case "$TERM" in
xterm*|rxvt*)
    PS1="\[\e]0;${debian_chroot:+($debian_chroot)}\u@\h: \w\a\]$PS1"
    ;;
*)
    ;;
esac

# enable color support of ls and also add handy aliases
if [ -x /usr/bin/dircolors ]; then
    test -r ~/.dircolors && eval "$(dircolors -b ~/.dircolors)" || eval "$(dircolors -b)"
    alias ls='ls --color=auto'
    #alias dir='dir --color=auto'
    #alias vdir='vdir --color=auto'

    alias grep='grep --color=auto'
    alias fgrep='fgrep --color=auto'
    alias egrep='egrep --color=auto'
fi

# some more ls aliases
alias ll='ls -alF'
alias la='ls -A'
alias l='ls -CF'

# Add an "alert" alias for long running commands.  Use like so:
#   sleep 10; alert
alias alert='notify-send --urgency=low -i "$([ $? = 0 ] && echo terminal || echo error)" "$(history|tail -n1|sed -e '\''s/^\s*[0-9]\+\s*//;s/[;&|]\s*alert$//'\'')"'

# Alias definitions.
# You may want to put all your additions into a separate file like
# ~/.bash_aliases, instead of adding them here directly.
# See /usr/share/doc/bash-doc/examples in the bash-doc package.

if [ -f ~/.bash_aliases ]; then
    . ~/.bash_aliases
fi

# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi
source /opt/ros/kinetic/setup.bash
export PATH="$HOME/.rbenv/bin:$PATH"
eval "$(rbenv init -)"
export PATH="$HOME/.rbenv/plugins/ruby-build/bin:$PATH"
alias matlab='export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:~/1stday/realworldrecog/mexopencv/+cv/private/; LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libstdc++.so.6:/usr/lib/x86_64-linux-gnu/libgfortran.so.3 matlab'
unset GTK_IM_MODULE

source ~/android_core /devel/setup.bash
source rospack find jsk_tools/src/bashrc.ros
#rossetmaster 192.168.10.5
#rossetmaster 192.168.122.1
#rossetip 192.168.10.4
export TURTLEBOT_3D_SENSOR=kinect
export TURTLEBOT_3D_SENSOR=kinect
export TURTLEBOT_3D_SENSOR=kinect
#export PYTHONPATH=$PYTHONPATH:/usr/lib/python2.7/dist-package
#export PYENV_ROOT="$HOME/.pyenv"
#export PATH="$PYENV_ROOT/bin:$PATH"
#eval "$(#pyenv init -)"
~~~

k-okada, 2017-12-20:
見逃していました

pyenv global system

で良かったんですね

動いていればそれで正解です

morifuji551, 2017-12-20:
お騒がせして申し訳ありませんでした。どうもありがとうございました。

closed 2017-12-20 by morifuji551