jsk-enshu/robot-programming #233: ros-kinetic-turtlebot-xxxxがros-shadow-fixedのxenialリストから消えている

tongtybj, 2018-05-30:
@k-okada 

ros-kinetic-turtlebot-bringupとか、　ros-kinetic-turtlebot-teleopがros-shadow-fixed(dist: xenial)から外されているようです。
- ros-shadow-fixed: http://packages.ros.org/ros-shadow-fixed/ubuntu/dists/xenial/main/binary-amd64/Packages
- ros (default): http://packages.ros.org/ros/ubuntu/dists/xenial/main/binary-amd64/Packages

また、[travis.sh](https://github.com/jsk-enshu/robot-programming/blob/master/.travis.yml#L7)を見るとros-shadow-fixedでもbuildを回していますが、[travisのビルド結果](https://travis-ci.org/jsk-enshu/robot-programming/jobs/384300260)みても、
```
E: Unable to locate package ros-kinetic-turtlebot-teleop
ERROR: the following rosdeps failed to install
  apt: command [apt-get install -y ros-kinetic-turtlebot-teleop] failed
```
ただ、travis全体のビルドは通っているのは、dockerだからここ部分のエラーはtravisに返されていないという理解でいいでしょうか。

となれば、travisのenv matrixからros-shadow-fixedを外してもいいと思っていますが、どうでしょうか。

k-okada, 2018-06-05:
おそくなりました．
sync がかかるタイミングで一瞬消えることがあるので，
https://discourse.ros.org/t/preparing-for-kinetic-sync-2018-05-14/4768
みたらいいのは
http://repositories.ros.org/status_page/ros_kinetic_default.html?q=turtlebot_bringup
になります．ここのX64の一番右がreleaes reositoryで真ん中がshadow fixedです．左はbuilding．

sourceの左を見て
http://build.ros.org/view/Ksrc_uX/job/Ksrc_uX__turtlebot_bringup__ubuntu_xenial__source/
その下流のbinaryをみて
http://build.ros.org/view/Ksrc_uX/job/Kbin_uX64__turtlebot_bringup__ubuntu_xenial_amd64__binary/
upstreamのrealsenseがblockerなのがわかるので，
http://build.ros.org/view/Ksrc_uX/job/Kbin_uX64__realsense_camera__ubuntu_xenial_amd64__binary/92/console
ログをみると
```
23:26:48   4856 'Setting up ros-kinetic-librealsense (1.12.1-0xenial-20180222-192514-0800) ...'
23:26:48   4857 '/var/lib/dpkg/info/ros-kinetic-librealsense.postinst: 40: /var/lib/dpkg/info/ros-kinetic-librealsense.postinst: udevadm: not found'
23:26:48   4858 'libkmod: ERROR ../libkmod/libkmod.c:586 kmod_search_moddep: could not open moddep file '/lib/modules/4.15.0-20-generic/modules.dep.bin''
23:26:48   4859 'modinfo: ERROR: Module alias uvcvideo not found.'
23:26:48   4860 'dpkg: error processing package ros-kinetic-librealsense (--configure):'
23:26:48   4861 ' subprocess installed post-installation script returned error exit status 1'
23:26:48   4862 'Setting up ros-kinetic-roslint (0.11.0-0xenial-20180222-193217-0800) ...'
```
なので，https://github.com/IntelRealSense/librealsense がおかしいことがわかる．
ということで，
https://github.com/IntelRealSense/librealsense/pull/1829
https://github.com/ros/rosdistro/pull/18111
と２PR稼ぐことができる．

k-okada, 2018-06-05:
show-fixed がないと，もしregressingがあるとsyncの瞬間に全員のturtlebotが動かなくなるので，予め何が起こりそうか把握するためにもあったほうがよいとおもいます．

tongtybj, 2018-06-05:
@k-okada 

一連の流れを勉強させて頂きました。
shadow-fixed の重要さも理解しました。
ありがとうございます。