jsk-enshu/robot-programming #242: dxl_armed_turtlebotのrun_dependにdynamixel_motor(metapackage)が入っている。

708yamaguchi, 2018-09-12:
@tongtybj

metapackageである`dynamixel_motor`がROSのworkspaceにソースで入っていると、`dxl_armed_turtlebot`のbuildが失敗します(aptで入っている場合は成功します)。以下に、その失敗コードを貼ります。
```bash
$ catkin build
----------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /home/leus/ros/indigo
----------------------------------------------------------
Source Space:       [exists] /home/leus/ros/indigo/src
Log Space:          [exists] /home/leus/ros/indigo/logs
Build Space:        [exists] /home/leus/ros/indigo/build
Devel Space:        [exists] /home/leus/ros/indigo/devel
Install Space:      [unused] /home/leus/ros/indigo/install
DESTDIR:            [unused] None
----------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
----------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
----------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------
[build] Found '204' packages in 0.0 seconds.                                                                                                                           
[build] Package table is up to date.                                                                                                                                   
Abandoned <<< dxl_armed_turtlebot                         [ Depends on unknown jobs: dynamixel_motor ]                                                                 
Abandoned <<< daisya_euslisp_tutorials                    [ Depends on unknown jobs: dynamixel_motor ]                                                                 
```

調べてみると、metapackageがpackage.xmlのrun_dependに入っていることが悪さをしているようで、package.xmlのdependencyにはmetapackageではなく依存しているpackageそのものを書かなければならないらしいです。
問題コード
https://github.com/jsk-enshu/robot-programming/blob/master/dxl_armed_turtlebot/package.xml#L23
参考
https://github.com/ros/catkin/issues/824
http://www.ros.org/reps/rep-0127.html#metapackage

ここからが本題なのですが、`dxl_armed_turtlebot`のpackage.xmlのrun_dependから`dynamixel_motor`を抜いた場合、他に依存関係を記述しなければならないpackageをご存知でしょうか。
ちなみに、run_dependから`dynamixel_motor`を抜いた状態でcatkin_buildすると、ソースでdynamixel_motorが入っていてもbuildに成功します。
お手数をお掛けしますが、お時間のあるときにでもご回答のほどよろしくお願いいたします。

tongtybj, 2018-09-12:
調べてくれてありがとう。

[dynamixel_motor](https://github.com/arebgun/dynamixel_motor)及びその下のrospackageに関しては、
ビルド時に必要というよりかは、rosdep install時にちゃんと見つかるように明記しているというのが自分の理解です。

また、演習ではaptから入れることを推奨している以上、rosdep installは必須で、metapackageを記入することが最適解かなと思っています。

YutoUchimi, 2018-09-12:
@tongtybj さん

@708yamaguchi の報告と同じ問題を確認しました。`dynamixel_motor`が明示的に依存している4つのパッケージの中で特に`dynamixel_driver`は、`dynamixel_7dof_arm`パッケージに必要なようです。

なので、少なくとも`dynamixex_7dof_arm`のpackage.xmlには`dynamixel_driver`を記述する必要があるのかなと思います。

それ以外にも`dynamixel_motor`の依存パッケージが使われているのでしょうか？

tongtybj, 2018-09-12:
敢えて、dynaxmielをソースから入れる学生がいると仮定するなら、それでいいかもしれないが、
 rosdep installから入れる学生が49/50ぐらいなので、metapackageに含まれるros packageを全部記入しないとaptからinstallされません(dynamixel_tutorialsは要らない)。

k-okada, 2018-09-12:
いや、そもそもdependsにメタパッケージを書いてはいけない、というのがREPに書いてあるよ、とtully
が教えてくれました、という話です。なので今動いているのはマグレなので直してみて下さい、という話でした

rep、tully ってなんなの？という時は調べてみると勉強になると思います。

tongtybj, 2018-09-12:
勉強になりました。

よく考えてみたら、dxl_armed_turtlebotはdynamixel_7dof_armを依存しているのであれば、package.xmlにdynamixel_motorを再度依存する必要はなかったのかな。

ということで、 @YutoUchimi が指摘したdynamixel_driverをdynamixel_7dof_armに追記しました。

708yamaguchi, 2018-09-13:
ありがとうございました。
解決出来ました。

closed 2018-09-13 by 708yamaguchi