jsk-enshu/robot-programming #255: Gazeboで腕を動かすと振動する

ykawamura96, 2018-11-11:
irteus上で、`(send *ri* : honyahonya`等を実行した時に、kimatics vewierではなく、gazeboで動かしてみたいと思い、[このページ](https://github.com/tongtybj/robot-programming/tree/gazebo/dxl_armed_turtlebot#1-grasping-test-in-gazebo-via-roseus)を参考にgazeboでもturtlebotを動かせるようにしてみました。
`armarm-move-sample.l`を実行して、アームを動かしてみたところ、以下のように動作しました。

kinematics viewerで動かした時は、このようにアームや台車が振動することはなかったのですが、なぜgazeboだと振動するのでしょうか？（kinematics viewerとgzeboの違いは何なのでしょうか？)

また、授業で配れれたプリントの図には、アーム駆動ノードから、USBシリアルでDXHUB基を板介して、各サーボモーターに指令が送られると描いてありました。ROS topc通信なのは、アーム駆動ノードまでなので、アームのサーボモーターのゲインなどはPCから操作できないということなのでしょうか？

[image]

ykawamura96, 2018-11-12:
@tongtybj 
とても分かりやすい説明ありがとうございます。

どのような指令が出ているのか見てみようと思い`$ rostopic echo /fullbody_controller/state `としてみました。みてみると、たしかにeffortの指令はないことが確かめられました。
```
header: 
  seq: 192789
  stamp: 
    secs: 1929
    nsecs: 110000000
  frame_id: ''
joint_names: [arm_joint1, arm_joint2, arm_joint3, arm_joint4, arm_joint5, arm_joint6]
desired: 
  positions: [0.0, 0.0, -0.17176284156590915, 0.0, 0.17176284156590915, 0.0]
  velocities: [0.0, 0.0, 0.3856786115014115, 0.0, -0.3856786115014115, 0.0]
  accelerations: [0.0, 0.0, -0.3486745929939201, 0.0, 0.3486745929939201, 0.0]
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
actual: 
  positions: [0.1976109894030138, 0.050513765743611216, -0.2677518737328253, -0.2150776855313019, 0.20997056469816666, 0.0030499490862645473]
  velocities: [-1.320940840836216, 0.991366288767145, 0.18936872757663, -4.897584821751522, -0.32592852780130277, 0.026617040493625677]
  accelerations: []
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0
error: 
  positions: [-0.1976109894030138, -0.050513765743611216, 0.09598903216691568, 0.2150776855313019, -0.038207723132257954, -0.0030499490862645473]
  velocities: [1.320940840836216, -0.991366288767145, 0.19630988392478146, 4.897584821751522, -0.0597500837001087, -0.026617040493625677]
  accelerations: []
  effort: []
  time_from_start: 
    secs: 0
    nsecs:         0

```
ところで、[ここ](https://github.com/tongtybj/robot-programming/blob/f5881faaf2476bc46924f26a5f028efda35668c1/dxl_armed_turtlebot/config/dxl_armed_turtlebot_controllers.yaml#L16)にある、
 ```
constraints:
    goal_time: 0.5                   # Override default
    stopped_velocity_tolerance: 0.02 # Override default
    arm_joint1:
      #trajectory: 0.1               # Not enforced if unspecified
      goal: 0.02                     # Not enforced if unspecified
```
はgoa_time, stopped_velocity_tolerance, goalなどの数字はどのような意味があるのでしょうか？
また、シミュレーションについてですが、アームが揺れてる時に、台車がここまで振動することは直感的にないような気がするのですが、これもなにか原因があるのでしょうか？（実際に台車が揺れるということでしたら的はずれな質問で申し訳ありません。）

最後に、もうひとつ質問があるのですが、`/fullbody_controller/gains/arm_joint1/parameter_descriptions`と`/fullbody_controller/gains/arm_joint1/parameter_updates`の違いは何でしょうか？`rqt_graph`としてみてもどちらもグラフ上に見当たらないので、何をしているトピックなのか分かりませんでした。（rqt_graph以外で、あるトピックがどのような役割を持っているか調べるいい方法があれば教えていただけると幸いです。）

ykawamura96, 2018-11-12:
これは完全にでっち上げですが, gazebo上でarm_link{num}_linkのgravityをゼロにすると、アームから台車へ働く慣性力?が消え、台車の振動はなくなりました。(gazeboでやる意味が失われている気しかしない)
パラメーターですが`arm_joint1: {p: 1.0, i: 0.0, d: 0.5}`として、pゲインを小さくすると、振動がだいぶ減るようです。arm_joint1以外のパラメーターは変更していません。
 どちらにせよ、その場凌ぎの対応でしかないので、あまり意味がないかもです。
物理世界をうまくシミュレーションするのは難しいなと思いました。

tongtybj, 2018-11-15:
> goa_time, stopped_velocity_tolerance, goalなどの数字はどのような意味があるのでしょうか？

これは軌道追跡制御のいろんな制約(しきい値)を表しています。
http://wiki.ros.org/joint_trajectory_controller の3.15に詳細が書かれています。
まず、自分で理解してみてください。

> シミュレーションについて、アームが揺れてる時に、台車がここまで振動することは直感的にないような気がするのですが、これもなにか原因があるのでしょうか？

　一般的に言うと、シミュレーションと実機での揺れの原因は違う場合がほとんどで、最近の物理シミュレーションは進化しているとはいえ、多リンク系の動力学を計算しているなかで、少なからず計算誤差が生じたり、関節部の内力がおかしかったりと現実世界とは違う挙動をします。
　また、今回のケースに関しては、トルクベースの力学モデル(多リンク系の力学モデル)を無理やり速度ベースで制御しているので、いくら頑張ってもいい感じの収束ゲインがないというのが自分の見解です。　
　なので、シミュレーション上の挙動を良くするのはいいことですが、頑張り過ぎるのもあまり意味はなく、本演習のシミュレーションの役目は認識に基づくロボット動作生成計画法の評価という意味では、アームが多少揺れても物体をちゃんと検出し、接近して把持できればいいと思います。

> 最後に、もうひとつ質問があるのですが、/fullbody_controller/gains/arm_joint1/parameter_descriptionsと/fullbody_controller/gains/arm_joint1/parameter_updatesの違いは何でしょうか

arm jointのゲイン調整は[dynamic reconfigure](http://wiki.ros.org/dynamic_reconfigure)という動的調整を可能にするros内の特殊サーバで行っています。この２つのトピックはここからpublishされています。
```
$  rosrun rqt_reconfigure rqt_reconfigure 
```
を立ち上げると、以下のようにゲインを調整できます。
[image]
この状態で、上記のトピックをechoすると、値が返ってきます。まず試してみてください。

>  rqt_graph以外で、あるトピックがどのような役割を持っているか調べるいい方法があれば教えていただけると幸いです。

```rostopic  info```がおすすめです、どこからpublishされて、どこにsubscriveされているのがわかるので

tongtybj, 2018-11-15:
> これは完全にでっち上げですが, gazebo上でarm_link{num}_linkのgravityをゼロにすると、アームから台車へ働く慣性力?が消え、台車の振動はなくなりました。(gazeboでやる意味が失われている気しかしない)

とても良い試みです！　これで試しに物体を掴んでみてください。
```
$ roscd dxl_armed_turtlebot/euslisp
$ roseus gazebo-grasp-with-recognition.l
```
の実行結果を教えてみてください。