jsk-enshu/robot-programming #262: 2018/11/12 演習課題１（７班）

ToiTakahashi, 2018-11-15:
armのend-coordsの座標軸の、実世界との対応付けがわからなかったです。
あと、end-coordsの原点の場所と、長さの単位がわかりませんでした。
原点の場所は起動時にあったend-coordsの場所ということでしょうか。

k-okada, 2018-11-20:
end-coordsの単位は`[mm]`です．原点は `(send *world-coords* :draw-on :flush t :size 100)` で表示されます．

Cc: @jsk-enshu/ta

tongtybj, 2018-11-20:
```
irteusgl$ (send *dxl-armed-turtlebot* :arm :end-coords)
#<cascaded-coords #X6a6b420  366.725 9.525 441.5 / 0.245 -1.571 2.897>
```
でわかることは，kinematics simulatorの原点からみるend-coordsのxyzの位置が
`[366.725mm 9.525mm 441.5mm]`で，
yaw,pitch,rollの順で表した姿勢が`[0.245rad -1.571rad 2.897rad]`です。

第2回の演習資料の付録A.5に追記しました。
[image]