jsk-enshu/robot-programming #264: 2018/11/12 演習課題１（７班）

ToiTakahashi, 2018-11-15:
screenの利用について質問です。
screenとは一つのPC内に複数のPCからwifiを用いてアクセスし直接操作するという認識なのですが、その場合、今までのような遠隔PC内で操作して通信するという場合とは何が異なるのでしょうか。直接その端末内のものを開いている分速度などが上がるのでしょうか。

TakadaTakumi, 2018-11-19:
キーボード操縦に関する質問です。roslaunch turtlebot_teleop keyboard_teleop.launch を実行すると
process[turtlebot_teleop_keyboard-1]: started with pid [7906]
  File "/opt/ros/kinetic/lib/turtlebot_teleop/turtlebot_teleop_key", line 106
    print msg
            ^
というエラーがでます。これはどうやら、スクリプトがpython 3.7に対応していないのが原因のようなので、新しいpython のバージョンでリリースして欲しいです。

TakadaTakumi, 2018-11-19:
逆運動学に関して、スクリプトで  (send *ri* :inverse-kinematics (make-coords :pos (float-vector 400 0 200)) :rotaion-axis nil)
のように書くと動きません。実機のほうにいきなり逆運動学を適用するのではなく、シミュレーター上
のマシンに適用させてから同じ値を実機に適用するという二度手間なので、一度で実機に逆運動学をてきちょうできるようにしてほしいです。

TakadaTakumi, 2018-11-19:
アームに関して、ネジがすぐ落ちる状態です。その結果、アームの固定がゆるく、規定範囲外の領域に曲がりwarningが出る状態でした。組み立ての段階でナットが挟まれていないのが原因のようです。

TakadaTakumi, 2018-11-19:
ジョイスティックをつなげてタートルロボを操作しようとすると、process has diedと出てプロセスがすぐ死んでしまいます。よく起こることのようですが、これを防ぐにはどうすればよいでしょうか？

itohdak, 2018-11-20:
@TakadaTakumi 

> キーボード操縦に関する質問です。roslaunch turtlebot_teleop keyboard_teleop.launch を実行すると
> ```
> process[turtlebot_teleop_keyboard-1]: started with pid [7906]
> File [...]

こちらは，Python 3.7というよりPython 3全体に共通するもので，print関数の構文がPython2から変わったことが原因です．
具体的には，`print "hoge"`みたいな表現はエラーになり，`print("hoge")`というような括弧付きの書き方をしなければいけません．

sourceの方では[こちらのコミット](https://github.com/turtlebot/turtlebot/commit/a6e93c8bedc72ee11769d8c5ed700560ad1c841c)で解決していますが，まだリリースはされていないようです．
以下のように，sourceから落としてくると反映させることができると思いますが，演習の環境を変に壊してしまう可能性もあるので，試すときは（覚悟を持って）慎重にやってください．
```
$ cd ~/catkin_ws/src
$ wstool set turtlebot https://github.com/turtlebot/turtlebot.git --git
$ wstool up turtlebot
$ catkin build turtlebot
$ source ~/catkin_ws/devel/setup.bash
$ roslaunch turtlebot_teleop keyboard_teleop.launch
```

k-okada, 2018-11-20:
> スクリプトがpython 3.7に対応していないのが原因

pipかanacondaが入っている可能性があります．ROSはUbuntuのシステムの上に構築されているので，Ubuntuが提供するpython2を前提にしています．ROSを使うときはこれらのカスタムのpythonを切ってください．
その上で，python3へのパッチは以下であたっています．
 https://github.com/turtlebot/turtlebot/commit/a6e93c8bedc72ee11769d8c5ed700560ad1c841c
リリースはまだなので，こちらのリポジトリで新しいバージョンのリリースをリクエストするとy位でしょう．

> 実機のほうにいきなり逆運動学を適用するのではなく、シミュレーター上のマシンに適用させてから同じ値を実機に適用するという二度手間なので、

`*ri*` は実機のモデルではなく，実機へのインターフェースなのでIKを解く，という概念はありません．
一方で，MoveItという軌道計画プログラムをturtlebotでもサポートすれば，
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/8d078ec08db42f533c8786b446c656e429fcb2fe/pr2eus_moveit/euslisp/robot-moveit.l#L616
で手先位置姿勢で指令を出すことができます．

> process has diedと出てプロセスがすぐ死んでしまいます。よく起こることのようですが、これを防ぐにはどうすればよいでしょうか？

以下のように，エラーの内容，あるいはログを詳細に貼り付けてもらいたいとおもいますが，例えば同じ名前のROSノードは２つ存在できないので，２つ目のノードを立ち上げた時に，元のノードが落ちるようになります．
https://github.com/jsk-enshu/robot-programming/issues/269#issuecomment-440112596

cc: @jsk-enshu/ta