jsk-enshu/robot-programming #283: 演習用ロボットのシミュレーションができない

ss0902, 2019-10-03:
git fetch --all
source ~/catkin_ws/devel/setup.bash
を行った後, 
roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch
を実行すると
... logging to /home/mech-user/.ros/log/546946ac-e4ca-11e9-bcf2-2079185ba004/roslaunch-KI00107-4798.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro.py is deprecated; please use xacro instead
started roslaunch server http://KI00107:34663/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/mech-user/c...
 * /depthimage_to_laserscan/output_frame_id: /camera_depth_frame
 * /depthimage_to_laserscan/range_min: 0.45
 * /depthimage_to_laserscan/scan_height: 10
 * /fullbody_controller/action_monitor_rate: 100
 * /fullbody_controller/constraints/arm_joint1/goal: 0.02
 * /fullbody_controller/constraints/arm_joint2/goal: 0.02
 * /fullbody_controller/constraints/arm_joint3/goal: 0.02
 * /fullbody_controller/constraints/arm_joint4/goal: 0.02
 * /fullbody_controller/constraints/arm_joint5/goal: 0.02
 * /fullbody_controller/constraints/arm_joint6/goal: 0.02
 * /fullbody_controller/constraints/goal_time: 0.5
 * /fullbody_controller/constraints/stopped_velocity_tolerance: 0.02
 * /fullbody_controller/gains/arm_joint1/d: 0.0
 * /fullbody_controller/gains/arm_joint1/i: 0.0
 * /fullbody_controller/gains/arm_joint1/p: 25.0
 * /fullbody_controller/gains/arm_joint2/d: 0.0
 * /fullbody_controller/gains/arm_joint2/i: 0.0
 * /fullbody_controller/gains/arm_joint2/p: 25.0
 * /fullbody_controller/gains/arm_joint3/d: 0.0
 * /fullbody_controller/gains/arm_joint3/i: 0.0
 * /fullbody_controller/gains/arm_joint3/p: 25.0
 * /fullbody_controller/gains/arm_joint4/d: 0.0
 * /fullbody_controller/gains/arm_joint4/i: 0.0
 * /fullbody_controller/gains/arm_joint4/p: 25.0
 * /fullbody_controller/gains/arm_joint5/d: 0.0
 * /fullbody_controller/gains/arm_joint5/i: 0.0
 * /fullbody_controller/gains/arm_joint5/p: 25.0
 * /fullbody_controller/gains/arm_joint6/d: 0.0
 * /fullbody_controller/gains/arm_joint6/i: 0.0
 * /fullbody_controller/gains/arm_joint6/p: 25.0
 * /fullbody_controller/joints: ['arm_joint1', 'a...
 * /fullbody_controller/state_publish_rate: 100
 * /fullbody_controller/stop_trajectory_duration: 0
 * /fullbody_controller/type: velocity_controll...
 * /gazebo/enable_ros_network: True
 * /gripper_controller/action_monitor_rate: 100
 * /gripper_controller/constraints/goal_time: 0.5
 * /gripper_controller/constraints/gripper_joint/goal: 0.02
 * /gripper_controller/constraints/stopped_velocity_tolerance: 0.02
 * /gripper_controller/gains/gripper_joint/d: 0.0
 * /gripper_controller/gains/gripper_joint/i: 0.0
 * /gripper_controller/gains/gripper_joint/p: 10.0
 * /gripper_controller/joints: ['gripper_joint']
 * /gripper_controller/state_publish_rate: 100
 * /gripper_controller/stop_trajectory_duration: 0
 * /gripper_controller/type: velocity_controll...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: melodic
 * /rosversion: 1.14.3
[... 67 lines cut ...]
[ERROR] [1570065775.857000958, 0.100000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint5
[ERROR] [1570065775.857571901, 0.100000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint6
[ERROR] [1570065775.858134730, 0.100000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/gripper_joint
[ INFO] [1570065775.863694081, 0.100000000]: Loaded gazebo_ros_control.
[urdf_spawner-6] process has finished cleanly
log file: /home/mech-user/.ros/log/546946ac-e4ca-11e9-bcf2-2079185ba004/urdf_spawner-6*.log
[INFO] [1570065776.057468, 0.290000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1570065776.059588, 0.290000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1570065776.061498, 0.290000]: Loading controller: joint_state_controller
[INFO] [1570065776.076650, 0.310000]: Loading controller: fullbody_controller
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[INFO] [1570065776.256967, 0.490000]: Loading controller: gripper_controller
[INFO] [1570065776.307266, 0.540000]: Controller Spawner: Loaded controllers: joint_state_controller, fullbody_controller, gripper_controller
[INFO] [1570065776.317247, 0.550000]: Started controllers: joint_state_controller, fullbody_controller, gripper_controller
[tuck_arm-11] process has finished cleanly
log file: /home/mech-user/.ros/log/546946ac-e4ca-11e9-bcf2-2079185ba004/tuck_arm-11*.log
とエラーが出ながらrvizとgazeboが起動しました。
その後, 別のターミナルで
roscd dxl_armed_turtlebot/launch
とすると
roscd: No such package/stack 'dxl_armed_turtlebot/launch'
と表示されてしまいました。

k-okada, 2019-10-03:
```
[ERROR] [1570065775.857571901, 0.100000000]: No p gain specified for pid. Namespace: /gazebo_ros_control/pid_gains/arm_joint6
[ERROR] [1570065775.858134730, 0.100000000]: No p gain specified for pid. Namespace: /gazebo_ros_control/pid_gains/gripper_joint
```
は無視して良いエラーです。

その後　別ターミナルで、、、の部分で
```
roscd dxl_armed_turtlebot/launch
```
の部分は` dxl_armed_tutrtlebot `パッケージ以下の`launch`ディレクトリに移動するコマンドです。
`roscd dxl_armed_turtlebot` `cd launch`とするのと同じですので、これができているか確認してください
Gazeboの立ち上げで`dxl_armed_turtlebot`パッケージには移動できているはずですが、
`source ~/catkin_ws/devel/setup.bash ` を忘れているでしょうか？いずれにせよ、ターミナルを立ち上げてから何をどうして最後に`No such packages/stack`になるか知りたいのでログを全て貼り付けるか、ターミナルの画像をキャプチャして貼り付けてください。

Gihubの issueではコードの部分は` ``` ` で書こうときれいになります。

ss0902, 2019-10-03:
おっしゃるとおり, 新しく立ち上げたターミナルで
```source ~/catkin_ws/devel/setup.bash```を忘れていました。
以下はターミナルを起動してから
```roslaunch dxl__armed_turtlebot dxl_armed_turtlebot_gazebo.launch```
を行うまでの一つのターミナルのキャプチャです。
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

k-okada, 2019-10-03:
`roscd dxl_armed_turtlebot/launch` 実行できたでしょうか？

ss0902, 2019-10-03:
それは実行できました。
ですが, キャプチャの最後の画像にある通りgripper_jointなどについてエラーが出ており, 課題をクリアした友人に動きを見せたところやはりジョイントの動きが少しおかしいようでした。

k-okada, 2019-10-03:
どうおかしいかビデオにとるとか、animation gif などにしてみてください

以下のエラーは無視していいものです

[ERROR] [1570065775.858134730, 0.100000000]: No p gain specified for
pid. Namespace: /gazebo_ros_control/pid_gains/gripper_joint

ss0902, 2019-10-04:
返信が遅れ申し訳ありません。
一連の流れを動画にしましたがサイズが大きすぎてgithubには載せられませんでした。
お忙しいところ申し訳ありませんが、ご確認いただいておかしいところがあればご指摘いただければ幸いです。

よろしくお願いします。

ss0902, 2019-10-04:
大変申し訳ありません。
動画の添付を忘れていました。
サイズが大きいのでgoogleドライブのリンクを貼らせていただきます。
よろしくお願いします。
 robot-programming-2019-10-04_11.40.43.mp4
<https://drive.google.com/file/d/10OmoxSpJGb8RY0NCNYapA2LNX6HAzORj/view?usp=drive_web>

2019年10月4日(金) 12:02 清水省吾 <shogo.gyomu@gmail.com>: