jsk-enshu/robot-programming #285: catkin buildが通らなくなりました

YKaiho11, 2019-10-11:
演習用ロボットプログラムの環境の構築をワークスペースを作るところからやり直したところ、catkin buildでエラーが起こりました。そのままシミュレーションを行おうとしましたが、
$ roslaunch opencv_apps camshift.launch image:=/camera/rgb/image_raw
としたときに、
RLException: [camshift.launch] is neither a launch file in package [opencv_apps] nor is [opencv_apps] a launch file name
The traceback for the exception was written to the log file
と表示されます。
$ sudo apt-get install ros-melodic-opencv-apps
$  rosdep install --from-paths . --ignore-src -y -r
としても、変わりませんでした。

catkin buildのエラーメッセージ
```
mech-user@KI00097:~/catkin_ws$ catkin build
--------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/melodic
Workspace:                   /home/mech-user/catkin_ws
--------------------------------------------------------------
Build Space:        [exists] /home/mech-user/catkin_ws/build
Devel Space:        [exists] /home/mech-user/catkin_ws/devel
Install Space:      [unused] /home/mech-user/catkin_ws/install
Log Space:         [missing] /home/mech-user/catkin_ws/logs
Source Space:       [exists] /home/mech-user/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.

NOTE: Forcing CMake to run for each package.
--------------------------------------------------------------
[build] Found '39' packages in 0.0 seconds.                                                                                                                                                                
[build] Updating package table.                                                                                                                                                                            
Starting  >>> catkin_tools_prebuild                                                                                                                                                                        
Finished  <<< catkin_tools_prebuild                     [ 1.8 seconds ]                                                                                                                                    
Starting  >>> dynamixel_7dof_arm                                                                                                                                                                           
Starting  >>> dynamixel_urdf                                                                                                                                                                               
Starting  >>> kobuki_auto_docking                                                                                                                                                                          
Starting  >>> kobuki_bumper2pc                                                                                                                                                                             
Starting  >>> kobuki_controller_tutorial                                                                                                                                                                   
Starting  >>> kobuki_dashboard                                                                                                                                                                             
Starting  >>> kobuki_description                                                                                                                                                                           
Starting  >>> kobuki_gazebo_plugins                                                                                                                                                                        
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << kobuki_dashboard:cmake /home/mech-user/catkin_ws/logs/kobuki_dashboard/build.cmake.000.log                                                                                                   
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "kobuki_msgs" with
  any of the following names:

    kobuki_msgsConfig.cmake
    kobuki_msgs-config.cmake

  Add the installation prefix of "kobuki_msgs" to CMAKE_PREFIX_PATH or set
  "kobuki_msgs_DIR" to a directory containing one of the above files.  If
  "kobuki_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:4 (find_package)

cd /home/mech-user/catkin_ws/build/kobuki_dashboard; catkin build --get-env kobuki_dashboard | catkin env -si  /usr/bin/cmake /home/mech-user/catkin_ws/src/kobuki_desktop/kobuki_dashboard --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mech-user/catkin_ws/devel/.private/kobuki_dashboard -DCMAKE_INSTALL_PREFIX=/home/mech-user/catkin_ws/install; cd -
[... 111 lines cut ...]
  Could not find a package configuration file provided by "kobuki_msgs" with
  any of the following names:

    kobuki_msgsConfig.cmake
    kobuki_msgs-config.cmake

  Add the installation prefix of "kobuki_msgs" to CMAKE_PREFIX_PATH or set
  "kobuki_msgs_DIR" to a directory containing one of the above files.  If
  "kobuki_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:7 (find_package)

cd /home/mech-user/catkin_ws/build/kobuki_gazebo_plugins; catkin build --get-env kobuki_gazebo_plugins | catkin env -si  /usr/bin/cmake /home/mech-user/catkin_ws/src/kobuki_desktop/kobuki_gazebo_plugins --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mech-user/catkin_ws/devel/.private/kobuki_gazebo_plugins -DCMAKE_INSTALL_PREFIX=/home/mech-user/catkin_ws/install; cd -
...........................................................................................................................................................................................................
Failed     << kobuki_gazebo_plugins:cmake               [ Exited with code 1 ]                                                                                                                             
Failed    <<< kobuki_gazebo_plugins                     [ 4.8 seconds ]                                                                                                                                    
[build] Summary: 4 of 35 packages succeeded.                                                                                                                                                               
[build]   Ignored:   5 packages were skipped or are blacklisted.                                                                                                                                           
[build]   Warnings:  None.                                                                                                                                                                                 
[build]   Abandoned: 26 packages were abandoned.                                                                                                                                                           
[build]   Failed:    5 packages failed.                                                                                                                                                                    
[build] Runtime: 6.9 seconds total.                                                                                                                                                                        
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

k-okada, 2019-10-11:
```
$ sudo apt-get install ros-melodic-opencv-apps
```
でエラー無くインストールできていれば必ず
```
$ roslaunch opencv_apps camshift.launch image:=/camera/rgb/image_raw
```
出来るはずです．`roscd opencv_apps`としてパッケージがインストールされているか確認してください．

後者の問題は，
```
$ rosdep install --from-paths . --ignore-src -y -r
```
でエラーがエラーがでいませんか？問題を報告するときは実行したコマンドとその出力を貼り付けると良いです．大量のメッセージの場合は [http://gist.github.com ](http://gist.github.com)が便利です．
そこでエラーがでている場合は
```
sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
``
としてください．
この後，
```
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-melodic-pr2eus] failed
  apt: command [sudo -H apt-get install -y ros-melodic-roseus-tutorials] failed
．．．
```
については，１，２週間で解消する予定なので今は無視して大丈夫です．

YKaiho11, 2019-10-11:
ご対応いただきありがとうございます。
理由はよくわかりませんが
```sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654```
を実行したところ、opencv_appsを立ち上げられるようになりました。

closed 2019-10-12 by YKaiho11