jsk-enshu/robot-programming #287: ROS Melodicで依存パッケージをインストール

mmurooka, 2019-10-16:
現時点でのメモです。

以下のように、rosdep installをすると、ros-melodic-pr2eusとros-melodic-roseus-tutorialsがないと言われる。

```bash
murooka@murooka-ThinkPad-P51:~/catkin_ws/src$ wstool info
workspace: /home/murooka/catkin_ws/src

 Localname           S SCM Version (Spec) UID  (Spec)  URI  (Spec) [http(s)://...]
 ---------           - --- -------------- -----------  ---------------------------
 turtlebot_simulator   git indigo  (-)    c5806a922d6b github.com/turtlebot/turtlebot_simulator
 turtlebot_msgs        git indigo  (-)    834c448fb313 github.com/turtlebot/turtlebot_msgs
 turtlebot_apps        git indigo  (-)    4bbafc7bfdfc github.com/turtlebot/turtlebot_apps
 turtlebot             git kinetic  (-)   a66b6eafe008 github.com/turtlebot/turtlebot
 kobuki_desktop        git devel  (-)     58459065087c github.com/yujinrobot/kobuki_desktop
 kobuki                git devel  (-)     161f69310ef9 github.com/yujinrobot/kobuki
 dynamixel_urdf        git master  (-)    048116cedf6c github.com/jsk-enshu/dynamixel_urdf
 dynamixel_motor       git master  (-)    fb20203b9426 github.com/arebgun/dynamixel_motor

Also detected these repositories in the workspace, add using 'wstool scrape' or 'wstool set':

 Localname         SCM   URI  (Spec) [http(s)://...]
 ---------         ---   ---------------------------
 robot-programming --git https://github.com/jsk-enshu/robot-programming
murooka@murooka-ThinkPad-P51:~/catkin_ws/src$ rosdep install --from-paths . --ignore-src -y -r > /dev/null
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
turtlebot_stdr: Cannot locate rosdep definition for [stdr_resources]
turtlebot_rapps: Cannot locate rosdep definition for [world_canvas_server]
kobuki_capabilities: Cannot locate rosdep definition for [rocon_apps]
turtleboteus: Cannot locate rosdep definition for [linux_peripheral_interfaces]
dxl_armed_turtlebot: Cannot locate rosdep definition for [turtlebot_interactive_markers]
turtlebot_description: Cannot locate rosdep definition for [create_description]
turtlebot_bringup: Cannot locate rosdep definition for [rocon_app_manager]
E: Unable to locate package ros-melodic-pr2eus
E: Unable to locate package ros-melodic-roseus-tutorials
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-melodic-pr2eus] failed
  apt: command [sudo -H apt-get install -y ros-melodic-roseus-tutorials] failed
  apt: Failed to detect successful installation of [ros-melodic-pr2eus]
  apt: Failed to detect successful installation of [ros-melodic-roseus-tutorials]
```

http://wiki.ros.org/action/show/TestingRepository?action=show&redirect=ShadowRepository
に従って、testing repositoryを使うと、以下のようにインストールできる。
```bash
murooka@murooka-ThinkPad-P51:~/catkin_ws/src$ rosdep install --from-paths . --ignore-src -y -r > /dev/null
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
turtlebot_stdr: Cannot locate rosdep definition for [stdr_resources]
turtlebot_rapps: Cannot locate rosdep definition for [world_canvas_server]
kobuki_capabilities: Cannot locate rosdep definition for [rocon_apps]
turtleboteus: Cannot locate rosdep definition for [linux_peripheral_interfaces]
dxl_armed_turtlebot: Cannot locate rosdep definition for [turtlebot_interactive_markers]
turtlebot_description: Cannot locate rosdep definition for [create_description]
turtlebot_bringup: Cannot locate rosdep definition for [rocon_app_manager]
```

k-okada, 2019-10-16:
pr2eusと同じで次のタイミングでインストール可能に成るはずです．
http://repositories.ros.org/status_page/ros_melodic_default.html?q=roseus_tutorials

mmurooka, 2019-11-08:
こちらリリースされたようです．
http://repositories.ros.org/status_page/ros_melodic_default.html?q=roseus_tutorials

closed 2019-11-08 by mmurooka