jsk-enshu/robot-programming #292: グリッパのDynamixelのトピック名が変わっている

mmurooka, 2019-10-18:
2018年度 演習資料 第2回 15ページ右上で、
```
roscd dynamixel_7dof_arm/euslisp/old_examples
roseus test-dynamixel-motor.l
(test-read-state)
(test-send-angle-vector-once)
```
というところがありますが（チェックポイント4で必要）、
これがエラーになり実行できません。

原因は、
https://github.com/jsk-enshu/robot-programming/blob/master/dynamixel_7dof_arm/euslisp/old_examples/test-dynamixel-motor.l#L30-L33
で、個々のDynamixelに関するsubscribeするtopicを指定していますが、

https://github.com/jsk-enshu/robot-programming/blame/master/dynamixel_7dof_arm/launch/dynamixel_7dof_arm_bringup.launch#L31
の変更で、グリッパのDynamixelについて、
topicの名前が
`/arm_j7_controller/command`から`/gripper_joint_controller/command`に変わってしまったためです。（`/arm_j7_controller/state`も同様。）

この変更は、どういう経緯だったでしょうか。（gazeboのため？）
もとに戻すか、別の修正が必要です、どうするのが良さそうでしょうか > @tongtybj

そもそも上の講義資料の引用部分を、なくしてしまうのもありでしょうか。
これは、robot-interfaceクラスを導入する直前に、個々のDynamixelをtopicで見たり動かしたりできますね、ということを体験してもらうところだと思います。

tongtybj, 2019-10-19:
返信遅れました.

`/arm_j7_controller` -> `/gripper_joint_controller`の変更は
https://github.com/jsk-enshu/robot-programming/pull/244　によるもので、
さらに辿ると、
https://github.com/jsk-enshu/robot-programming/pull/161 にある @k-okada の以下のコメントに従った変更となります。
> そもそも実機で*ri*も*robot*も:anlge-vectorしてgripperが含まれないようにしておくのが標準的な使い方だと思うので、モデルの方から治していく必要があるかな．

tongtybj, 2019-10-20:
https://github.com/jsk-enshu/robot-programming/blob/master/dynamixel_7dof_arm/euslisp/old_examples/test-dynamixel-motor.l　に関しては、面倒だけど、`*arm-dof* 6` + gripper でsubscriber とpublisherの初期化の部分を直した方がいいね。

というのは引用部分を残す前提の話ですが、
robot-interfaceクラスを使った方がいいので、
自分はこの部分を削除してもいいと思います。

ただ、こういうコードはtestに使うといい気がするので、
rostestに回すべきかと思います。

mmurooka, 2019-10-21:
経緯としては、gazebo対応をする中で
https://github.com/jsk-enshu/robot-programming/blob/master/dynamixel_7dof_arm/urdf/dynamixel_7dof_arm.urdf.xacro#L279
にあるようにシミュレーションでのグリッパjoint名が、"gripper_joint"になっていて、実機のグリッパjoint名は、"arm_joint7"だったので、後者を修正して前者に合わせた、ということですね。（もしかしたら、前者を後者に合わせたほうが良かったほうが、問題は起きなかったかもしれませんね。https://github.com/jsk-enshu/robot-programming/issues/293#issuecomment-544208917 みたいに、いろいろな所でtopic/service名を修正する作業が必要になる気がするので。）
ちなみに、controllerの名前(actionlibのtopic名）さえ一致していれば、joint名はどんな名前であろうと、gazeboと実機でrobot-interfaceを共通して使うというのとは関係ない気もします。

とりあえず
> https://github.com/jsk-enshu/robot-programming/blob/master/dynamixel_7dof_arm/euslisp/old_examples/test-dynamixel-motor.l　に関しては、面倒だけど、`*arm-dof* 6` + gripper でsubscriber とpublisherの初期化の部分を直した方がいいね。

のように修正しておこうと思います。

closed 2019-10-25 by mmurooka