jsk-enshu/robot-programming #293: robot-interfaceで:gripper-controllerが見つからない

mmurooka, 2019-10-18:
2018年度 演習資料 第2回 18ページ上半分で、
```
(send *ri* :angle-vector (send *dxl-armed-turtlebot* :reset-pose) 1000 :gripper-controller)
```
をさせていますが、
```
;; controller-type: :gripper-controller not foundnil
```
とエラーが出て、グリッパが動きません。

原因は、おそらく
https://github.com/jsk-enshu/robot-programming/blame/master/dynamixel_7dof_arm/euslisp/dxl-7dof-arm-interface-common.l#L149
で、`:gripper-controller`の定義がコメントアウトされているようですが、これは何のためだったでしょうか。 > @tongtybj

:start-grasp, :stop-graspは動くようなので、上の行を資料で飛ばせばよいでしょうか、
と書こうとしてよく見ると、正しく動いていないようです。
actionlibのサーバーは正しく上がっているので、:start-grasp, :stop-graspで関節角は変わりますが、
馴染みながら端をするようにゲインを変えているところ、つまり
https://github.com/jsk-enshu/robot-programming/blob/master/dynamixel_7dof_arm/euslisp/dxl-7dof-arm-interface-common.l#L279
から呼ばれる
https://github.com/jsk-enshu/robot-programming/blob/master/dynamixel_7dof_arm/euslisp/dxl-7dof-arm-interface-common.l#L173-L182
で、
https://github.com/jsk-enshu/robot-programming/issues/292 の問題で、グリッパDynamixelのservice名が変わってしまっているので、
ちゃんと呼べていない気がします。

tongtybj, 2019-10-20:
> で、:gripper-controllerの定義がコメントアウトされているようですが、これは何のためだったでしょうか。 > @tongtybj

こちらはpr2-interface.lを参考にして修正した結果だとおもう.
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L49-L52
https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L37

tongtybj, 2019-10-20:
> #292 の問題で、グリッパDynamixelのservice名が変わってしまっているので、

こちらは、service名を直すだけの作業？

mmurooka, 2019-10-21:
> こちらはpr2-interface.lを参考にして修正した結果だとおもう.

なるほど、それでは、演習資料では
```
(send *ri* :angle-vector (send *dxl-armed-turtlebot* :reset-pose) 1000 :gripper-controller)
```
の代わりに、
```
(send *ri* :move-gripper 100)
```
とかを書くようにしようと思います。

>> #292 の問題で、グリッパDynamixelのservice名が変わってしまっているので、
>
> こちらは、service名を直すだけの作業

ごめん、ちゃんとみたら https://github.com/jsk-enshu/robot-programming/blob/master/dynamixel_7dof_arm/euslisp/dxl-7dof-arm-interface-common.l#L177-L179
のif文で問題ないようになっていましたね。

mmurooka, 2019-10-25:
>> (send *ri* :move-gripper 100)
とかを書くようにしようと思います。

のように演習資料を修正しました．

closed 2019-10-25 by mmurooka