jsk-enshu/robot-programming #294: [Melodic] libfreenectがない

mmurooka, 2019-10-18:
一番厄介そうなのでもっと早く気づくべきでしたが、
http://wiki.ros.org/freenect_launch がMelodicでリリースされていません。

これはturtlebot_bringupで呼ばれる以下で使われています。
https://github.com/turtlebot/turtlebot/blob/99bf4e6c6a8e17e0840ec373f0016c957480acee/turtlebot_bringup/launch/includes/3dsensor/kinect.launch.xml#L23

これからちゃんと調べますが、
ソースから入れようとしても、rosdep install でエラーになって、当然catkin buildも通らないです。
```
murooka@murooka-ThinkPad-P51:~/ws_freenect$ rosdep install --from-paths src --ignore-src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
freenect_camera: Cannot locate rosdep definition for [libfreenect]
```

mmurooka, 2019-10-18:
libfreenectをソースから入れて動くことは確認できました。（演習でやるのは避けたいですが。。）

```bash
# install libfreenect
cd /tmp
mkdir src install
cd src
git clone https://github.com/OpenKinect/libfreenect.git
cd libfreenect
mkdir build
cd build
cmake -L .. -DCMAKE_INSTALL_PREFIX=/tmp/install
make
make install

# build freenect stack
export PKG_CONFIG_PATH=$PKG_CONFIG_PATH:/tmp/install/lib/pkgconfig
cd /tmp
mkdir -p ws_freenect/src
cd ws_freenect/src
wstool init
wstool set freenect_stack https://github.com/ros-drivers/freenect_stack --git -y
wstool update
cd freenect_stack/freenect_camera/include
find ./ -name "*.hpp" | xargs sed -i s@\<libfreenect/@\<@g # I don't know but this was necessary
cd /tmp/ws_freenect
catkin b

# test
source devel/setup.bash
export TURTLEBOT_3D_SENSOR=kinect
roslaunch turtlebot_bringup minimal.launch
roslaunch turtlebot_bringup 3dsensor.launch
roslaunch turtlebot_rviz_launchers view_robot.launch
```

mmurooka, 2019-10-18:
turtlebotはlibfreenectを使っていますが、openni_launchならMelodicでもリリースされていて、ソースからのインストールなく、タートルボットに付いているKinectが起動して、registered pointsが正しく出ていました。
```bash
sudo apt install ros-melodic-openni-launch # 何かの依存で既に入っていた
roslaunch openni_launch openni.launch
```

選択肢としては、
1. libfreenectの代わりにopenni_launchを使う。
2. https://github.com/jsk-enshu/robot-programming/issues/294#issuecomment-543773867 の手順でソースから入れてもらう
3. https://github.com/jsk-enshu/robot-programming/issues/294#issuecomment-543773867 の手順でビルドしてできたdebファイルを、wgetで落としてきてもらってdpkgでインストールしてもらう。

1.の場合は、
```
roslaunch turtlebot_bringup 3dsensor.launch
```
に相当するlaunchファイルを、自前で作って、その中で、タートルボットとKinectがtfで繋がらないのでstatic_transform_publisher して、くらいで問題ないはずでしょうか。

tongtybj, 2019-10-20:
1 に一票

mmurooka, 2019-10-21:
一票をありがとう、先生とも相談して、1.opennni にしようと思います。

closed 2019-10-21 by mmurooka