jsk-enshu/robot-programming #303: catkin build におけるエラーについて

poyotamu000, 2019-10-26:
以前 catkin build できていたのですが、なぜか通らなくなってしまいました。
おそらくjskのゼミでhsrに関して学科PCがmelodicなのに対し、のkineticに関するものをインストールしようとして中断したことによる気がしているのですが、まず catkin build をすると、以下のメッセージがでます。
```
mech-user@KI00125:~/catkin_ws$ catkin build
--------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/mech-user/catkin_ws
--------------------------------------------------------------
Build Space:        [exists] /home/mech-user/catkin_ws/build
Devel Space:        [exists] /home/mech-user/catkin_ws/devel
Install Space:      [unused] /home/mech-user/catkin_ws/install
Log Space:          [exists] /home/mech-user/catkin_ws/logs
Source Space:       [exists] /home/mech-user/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
--------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------
[build] Found '40' packages in 0.0 seconds.                                                                                                                                                                
[build] Package table is up to date.                                                                                                                                                                       
Starting  >>> beginner_tutorials                                                                                                                                                                           
Starting  >>> dynamixel_7dof_arm                                                                                                                                                                           
Starting  >>> dynamixel_urdf                                                                                                                                                                               
Starting  >>> kobuki_auto_docking                                                                                                                                                                          
Starting  >>> kobuki_bumper2pc                                                                                                                                                                             
Starting  >>> kobuki_controller_tutorial                                                                                                                                                                   
Starting  >>> kobuki_dashboard                                                                                                                                                                             
Starting  >>> kobuki_description                                                                                                                                                                           
Finished  <<< kobuki_dashboard                          [ 0.2 seconds ]                                                                                                                                    
Finished  <<< dynamixel_7dof_arm                        [ 0.2 seconds ]                                                                                                                                    
Finished  <<< kobuki_description                        [ 0.2 seconds ]                                                                                                                                    
Finished  <<< dynamixel_urdf                            [ 0.2 seconds ]                                                                                                                                    
Finished  <<< kobuki_controller_tutorial                [ 1.0 seconds ]                                                                                                                                    
Starting  >>> kobuki_gazebo_plugins                                                                                                                                                                        
Finished  <<< beginner_tutorials                        [ 1.0 seconds ]                                                                                                                                    
Starting  >>> kobuki_keyop                                                                                                                                                                                 
Starting  >>> kobuki_random_walker                                                                                                                                                                         
Starting  >>> kobuki_rviz_launchers                                                                                                                                                                        
Finished  <<< kobuki_bumper2pc                          [ 1.2 seconds ]                                                                                                                                    
Starting  >>> kobuki_safety_controller                                                                                                                                                                     
Starting  >>> turtlebot_capabilities                                                                                                                                                                       
Starting  >>> turtlebot_msgs                                                                                                                                                                               
Finished  <<< kobuki_auto_docking                       [ 1.3 seconds ]                                                                                                                                    
Finished  <<< kobuki_rviz_launchers                     [ 0.2 seconds ]                                                                                                                                    
Finished  <<< turtlebot_capabilities                    [ 0.2 seconds ]                                                                                                                                    
Finished  <<< turtlebot_msgs                            [ 0.7 seconds ]                                                                                                                                    
Finished  <<< kobuki_random_walker                      [ 1.0 seconds ]                                                                                                                                    
Finished  <<< kobuki_keyop                              [ 1.0 seconds ]                                                                                                                                    
Starting  >>> turtleboteus                                                                                                                                                                                 
Finished  <<< kobuki_safety_controller                  [ 1.0 seconds ]                                                                                                                                    
Finished  <<< turtleboteus                              [ 0.2 seconds ]                                                                                                                                    
Starting  >>> turtlebot_description                                                                                                                                                                        
[... cut ...]
rosdep install --from-paths src --ignore-src -r -n -y

cd ~/catkin_ws/hsr
catkin build jsk_hsr
```

どのようにしたらよいでしょうか？
rosdep install 時のエラーの
```
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-melodic-pr2eus] failed
  apt: command [sudo -H apt-get install -y ros-melodic-roseus-tutorials] failed
  apt: Failed to detect successful installation of [ros-melodic-pr2eus]
  apt: Failed to detect successful installation of [ros-melodic-roseus-tutorials]
```
のsudoコマンドを行っても大丈夫でしょうか...?

k-okada, 2019-10-26:
pr2eus/roseus-tutorials のエラーは現時点では無視して良いです．
`/etc/apt/sources.list.d/tmc.list`や
`/etc/apt/sources.list.d/gazebo-stable.list` の中身をみて，おそらく自分で追加した情報しかないはずですので，ファイルを消して`apt-get update`してください．
そのときに，`ubuntu.com` か `packages.ros.org` だけになっているはずです．

`hsr.io`や`osrfoundation.org` が入っていたら危険です．もう一度，`sources.list.d` 以下を見て，これらの情報が残っていないか調べてください．

あとは，
```
$ ls -al /usr/include/gazebo-9/gazebo/common/Battery.hh
-rw-r--r-- 1 root root 5121  2月 16  2018 /usr/include/gazebo-9/gazebo/common/Battery.hh
$ apt-file search sdf.hh
libsdformat6-dev: /usr/include/sdformat-6.0/sdf/sdf.hh
```
となっているので`apt-get install libsdformat6-dev`で治りそうですが，これでもダメなら．`osrfoundation.org`で同じ名前(gazebo)で中身が違うものがインストールされている可能性も無くはないので，`apt-get remove gazebo*` か何かしてみてください．

poyotamu000, 2019-10-26:
```/etc/apt/sources.list.d/tmc.list```や```/etc/apt/sources.list.d/gazebo-stable.list```には
確かに```hsr.io```や```osrfoundation.org```が入っていました。
これらのファイルを中身を削除ではなく、ファイルごと削除したのですがそれで大丈夫ですか？

また、```sudo apt-get install libsdformat6-dev```をしたのですが、
うまくいかなかったので、
```sudo apt-get remove gazebo*```　を実行しました。

その後に環境構築の一連の流れを実行すると以下のようになりました。
sudo apt-get コマンドでgazeboを入れる必要があるんでしょうか？
```
mech-user@KI00125:~$ sudo apt-get update
Hit:1 https://cloud.r-project.org/bin/linux/ubuntu bionic-cran35/ InRelease
Ign:2 http://dl.google.com/linux/chrome/deb stable InRelease                                        
Hit:3 http://archive.ubuntulinux.jp/ubuntu bionic InRelease                                         
Hit:4 http://jp.archive.ubuntu.com/ubuntu bionic InRelease                                          
Hit:5 http://jp.archive.ubuntu.com/ubuntu bionic-updates InRelease                                  
Hit:6 http://jp.archive.ubuntu.com/ubuntu bionic-backports InRelease                                
Hit:7 http://archive.ubuntulinux.jp/ubuntu-ja-non-free bionic InRelease                             
Hit:8 http://dl.google.com/linux/chrome/deb stable Release                                          
Hit:9 http://packages.microsoft.com/repos/vscode stable InRelease                                   
Hit:10 http://security.ubuntu.com/ubuntu bionic-security InRelease                                  
Hit:11 http://ppa.launchpad.net/team-gcc-arm-embedded/ppa/ubuntu bionic InRelease                   
Hit:12 http://packages.ros.org/ros/ubuntu bionic InRelease                                          
Hit:13 https://packagecloud.io/slacktechnologies/slack/debian jessie InRelease                      
Reading package lists... Done 
mech-user@KI00125:~$ cd catkin_ws/src/
mech-user@KI00125:~/catkin_ws/src$ ls
beginner_tutorials  kobuki          robot-programming  turtlebot_apps  turtlebot_simulator
dynamixel_urdf      kobuki_desktop  turtlebot          turtlebot_msgs
mech-user@KI00125:~/catkin_ws/src$ wstool init .
Error: There already is a workspace config file .rosinstall at ".". Use wstool install/modify.
mech-user@KI00125:~/catkin_ws/src$ wstool merge robot-programming/.rosinstall.melodic 
Merge caused no change, no new elements found
mech-user@KI00125:~/catkin_ws/src$ wstool update 
[dynamixel_urdf] Updating /home/mech-user/catkin_ws/src/dynamixel_urdf
[dynamixel_urdf] Done.
[kobuki] Updating /home/mech-user/catkin_ws/src/kobuki
[kobuki] Done.
[kobuki_desktop] Updating /home/mech-user/catkin_ws/src/kobuki_desktop
[kobuki_desktop] Done.
[turtlebot] Updating /home/mech-user/catkin_ws/src/turtlebot
[turtlebot] Done.
[turtlebot_apps] Updating /home/mech-user/catkin_ws/src/turtlebot_apps

[turtlebot_apps] Done.
[turtlebot_msgs] Updating /home/mech-user/catkin_ws/src/turtlebot_msgs
[turtlebot_msgs] Done.
[turtlebot_simulator] Updating /home/mech-user/catkin_ws/src/turtlebot_simulator
[turtlebot_simulator] Done.
mech-user@KI00125:~/catkin_ws/src$ 
mech-user@KI00125:~/catkin_ws/src$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Add distro "crystal"
Add distro "dashing"
Add distro "eloquent"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
[... 153 lines cut ...]
Abandoned <<< kobuki_node                               [ Unrelated job failed ]                    
Abandoned <<< kobuki_testsuite                          [ Unrelated job failed ]                    
Abandoned <<< kobuki_qtestsuite                         [ Unrelated job failed ]                    
Abandoned <<< turtlebot_actions                         [ Unrelated job failed ]                    
Abandoned <<< turtlebot_bringup                         [ Unrelated job failed ]                    
Abandoned <<< turtlebot_calibration                     [ Unrelated job failed ]                    
Abandoned <<< turtlebot_gazebo                          [ Unrelated job failed ]                    
Abandoned <<< turtlebot_follower                        [ Unrelated job failed ]                    
Abandoned <<< turtlebot_navigation                      [ Unrelated job failed ]                    
Abandoned <<< turtlebot_rapps                           [ Unrelated job failed ]                    
Abandoned <<< turtlebot_stage                           [ Unrelated job failed ]                    
Abandoned <<< turtlebot_stdr                            [ Unrelated job failed ]                    
Abandoned <<< turtlebot_teleop                          [ Unrelated job failed ]                    
Abandoned <<< dxl_armed_turtlebot                       [ Unrelated job failed ]                    
Abandoned <<< daisya_euslisp_tutorials                  [ Unrelated job failed ]                    
[... cut ...]
[build] Summary: 8 of 35 packages succeeded.                                                        
[build]   Ignored:   5 packages were skipped or are blacklisted.                                    
[build]   Warnings:  None.                                                                          
[build]   Abandoned: 26 packages were abandoned.                                                    
[build]   Failed:    1 packages failed.                                                             
[build] Runtime: 1.5 seconds total.                                        
```

poyotamu000, 2019-10-26:
補足しておきますと```apt-get install libsdformat6-dev```をすると以下のようになり、
```catkin build```したときのエラーは特に変化はありませんでした。
```
mech-user@KI00125:~$ sudo apt-get install libsdformat6-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libsdformat6-dev is already the newest version (6.2.0-1~bionic).
libsdformat6-dev set to manually installed.
The following packages were automatically installed and are no longer required:
  fonts-lato libgraphviz-dev libgvc6-plugins-gtk libisl15 libllvm7 libopenal-dev
  libqtpropertybrowser4 libruby2.5 libxdot4 linux-headers-4.15.0-58 linux-headers-4.15.0-58-generic
  linux-image-4.15.0-58-generic linux-modules-4.15.0-58-generic
  linux-modules-extra-4.15.0-58-generic rake ros-melodic-moveit-experimental
  ros-melodic-rosserial-client ruby ruby-did-you-mean ruby-minitest ruby-net-telnet
  ruby-power-assert ruby-test-unit ruby2.5 rubygems-integration
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 27 not upgraded.
```  

また、```apt-get remove gazebo*```したときの状態は以下のようになっていました。
```
mech-user@KI00125:~$ sudo apt-get remove gazebo*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'gazebo6-common-prerelease' for glob 'gazebo*'
Note, selecting 'gazebo7-common-prerelease' for glob 'gazebo*'
Note, selecting 'gazebo8-common' for glob 'gazebo*'
Note, selecting 'gazebo5-plugins-base' for glob 'gazebo*'
Note, selecting 'gazebo2' for glob 'gazebo*'
Note, selecting 'gazebo3' for glob 'gazebo*'
Note, selecting 'gazebo5' for glob 'gazebo*'
Note, selecting 'gazebo6' for glob 'gazebo*'
Note, selecting 'gazebo7' for glob 'gazebo*'
Note, selecting 'gazebo8' for glob 'gazebo*'
Note, selecting 'gazebo9' for glob 'gazebo*'
Note, selecting 'gazebo8-plugins-base' for glob 'gazebo*'
Note, selecting 'gazebo9-doc' for glob 'gazebo*'
Note, selecting 'gazebo9-plugin-base' for glob 'gazebo*'
Note, selecting 'gazebo7-doc' for glob 'gazebo*'
Note, selecting 'gazebo6-common' for glob 'gazebo*'
Note, selecting 'gazebo9-common' for glob 'gazebo*'
Note, selecting 'gazebo6-plugins-base' for glob 'gazebo*'
Note, selecting 'gazebo7-plugin-base' for glob 'gazebo*'
Note, selecting 'gazebo7-common' for glob 'gazebo*'
Note, selecting 'gazebo7-plugins-base' for glob 'gazebo*'
Package 'gazebo7' is not installed, so not removed
Package 'gazebo7-common' is not installed, so not removed
Package 'gazebo7-doc' is not installed, so not removed
Package 'gazebo7-plugin-base' is not installed, so not removed
Package 'gazebo6-common' is not installed, so not removed
Package 'gazebo6-common-prerelease' is not installed, so not removed
Package 'gazebo7-common-prerelease' is not installed, so not removed
Package 'gazebo8-common' is not installed, so not removed
Package 'gazebo5-plugins-base' is not installed, so not removed
Package 'gazebo6-plugins-base' is not installed, so not removed
Package 'gazebo7-plugins-base' is not installed, so not removed
Package 'gazebo8-plugins-base' is not installed, so not removed
Package 'gazebo2' is not installed, so not removed
Package 'gazebo3' is not installed, so not removed
Package 'gazebo5' is not installed, so not removed
Package 'gazebo6' is not installed, so not removed
Package 'gazebo8' is not installed, so not removed
Package 'gazebo9-doc' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  fltk1.3-doc fluid fonts-lato hddtemp libavdevice-dev libavdevice57 libavfilter-dev
  libfltk-cairo1.3 libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3 libfltk1.3-dev
  libfreeimage-dev libgazebo9 libgraphviz-dev libgts-dev libgvc6-plugins-gtk libignition-common
  libignition-common-dev libignition-fuel-tools1-1 libignition-fuel-tools1-dev libignition-msgs
[... 47 lines cut ...]
0 upgraded, 0 newly installed, 15 to remove and 27 not upgraded.
After this operation, 216 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 360725 files and directories currently installed.)
Removing ros-melodic-desktop-full (1.4.1-0bionic.20190920.223803) ...
Removing ros-melodic-simulators (1.4.1-0bionic.20190920.223600) ...
Removing ros-melodic-gazebo-ros-pkgs (2.8.4-0bionic.20190808.220706) ...
Removing ros-melodic-gazebo-plugins (2.8.4-0bionic.20190808.200816) ...
Removing ros-melodic-urdf-sim-tutorial (0.4.0-0bionic.20190920.222258) ...
Removing ros-melodic-turtlebot3-simulations (1.2.0-0bionic.20190827.222925) ...
Removing ros-melodic-turtlebot3-gazebo (1.2.0-0bionic.20190808.210815) ...
Removing ros-melodic-open-manipulator-gazebo (1.1.0-1bionic.20190911.202550) ...
Removing ros-melodic-gazebo-ros-control (2.8.4-0bionic.20190911.200117) ...
Removing ros-melodic-gazebo-ros (2.8.4-0bionic.20190808.204821) ...
Removing ros-melodic-gazebo-dev (2.8.4-0bionic.20190601.003151) ...
Removing libgazebo9-dev:amd64 (9.11.0-1~bionic) ...
Removing gazebo9-plugin-base (9.11.0-1~bionic) ...
Removing gazebo9 (9.11.0-1~bionic) ...
Removing gazebo9-common (9.11.0-1~bionic) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.2) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.60ubuntu1) ...
```

k-okada, 2019-10-27:
`rosdep install ...` が最初のpr2eus/roseus-tutorials のエラーになるだけの状況に戻るまで，それ以外（主にgazeboに成るかとお思いますが）インストールしてみてください．
```
sudo -H apt-get install -y ros-melodic-gazebo-ros
sudo -H apt-get install -y ros-melodic-gazebo-plugins
```

poyotamu000, 2019-10-27:
はい。上記の２つのコマンドを実行すると
```
mech-user@KI00125:~/catkin_ws/src/robot-programming$ sudo -H apt-get install -y ros-melodic-gazebo-rosReading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-melodic-gazebo-ros : Depends: ros-melodic-gazebo-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
```
mech-user@KI00125:~/catkin_ws/src/robot-programming$ sudo -H apt-get install -y ros-melodic-gazebo-plugins
[sudo] password for mech-user: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-melodic-gazebo-plugins : Depends: ros-melodic-gazebo-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
となりました。

依存パッケージがないということなので、それをいれようとおもって以下のコマンドを実行しました。
```
mech-user@KI00125:~/catkin_ws/src/robot-programming$ sudo -H apt-get install -y ros-melodic-gazebo-devReading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ros-melodic-gazebo-dev : Depends: gazebo9
                          Depends: libgazebo9-dev but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```
```
mech-user@KI00125:~/catkin_ws/src/robot-programming$ sudo -H apt-get install -y gazebo9
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9 : Depends: libgazebo9 (= 9.0.0+dfsg5-3ubuntu1+ppa2) but 9.11.0-1~bionic is to be installed
           Recommends: gazebo9-plugin-base
E: Unable to correct problems, you have held broken packages.
```

このようにかなり遡らなければならないようです。
```apt install ros-melodic-desktop-full```
のようにすればまとめてはいってくるといったことはありませんか...？

poyotamu000, 2019-10-27:
ひとまず、順に遡りながらインストールしていこうとしているのですが、
```
mech-user@KI00125:~/catkin_ws/src/robot-programming$ sudo -H apt-get install -y libgazebo9-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgazebo9-dev : Depends: libgazebo9 (= 9.0.0+dfsg5-3ubuntu1+ppa2) but 9.11.0-1~bionic is to be installed
E: Unable to correct problems, you have held broken packages.
```
となったので、依存パッケージのlibgazebo9をいれようとすると以下のようになってしまいました。
```
mech-user@KI00125:~/catkin_ws/src/robot-programming$ sudo -H apt-get install -y libgazebo9
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgazebo9 is already the newest version (9.11.0-1~bionic).
The following packages were automatically installed and are no longer required:
  fltk1.3-doc fluid fonts-lato hddtemp libavdevice-dev libavfilter-dev libfltk-cairo1.3
  libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3 libfltk1.3-dev libfreeimage-dev
  libgraphviz-dev libgts-dev libgvc6-plugins-gtk libignition-common-dev libignition-fuel-tools1-dev
  libignition-msgs-dev libignition-transport4-dev libisl15 libllvm7 libopenal-dev libpostproc-dev
  libprotobuf-dev libprotobuf-lite10 libprotoc-dev libprotoc10 libqtpropertybrowser4 libqwt-headers
  libqwt-qt5-dev libruby2.5 libsimbody-dev libtar-dev libxdot4 libxmu-dev libxmu-headers libyaml-dev
  libzip-dev libzmq3-dev linux-headers-4.15.0-58 linux-headers-4.15.0-58-generic
  linux-image-4.15.0-58-generic linux-modules-4.15.0-58-generic
  linux-modules-extra-4.15.0-58-generic protobuf-compiler python-cairo python-opengl python-psutil
  python-pyqt5.qtopengl python-pyqt5.qtwebkit rake ros-melodic-bond-core
  ros-melodic-camera-calibration ros-melodic-common-msgs ros-melodic-common-tutorials
  ros-melodic-controller-manager ros-melodic-desktop ros-melodic-diagnostic-analysis
  ros-melodic-diagnostic-common-diagnostics ros-melodic-diagnostics
  ros-melodic-diff-drive-controller ros-melodic-executive-smach
  ros-melodic-forward-command-controller ros-melodic-gazebo-msgs ros-melodic-geometry
  ros-melodic-geometry-tutorials ros-melodic-gl-dependency ros-melodic-image-common
  ros-melodic-image-pipeline ros-melodic-image-rotate ros-melodic-image-transport-plugins
  ros-melodic-interactive-marker-tutorials ros-melodic-joint-limits-interface
  ros-melodic-joint-state-controller ros-melodic-kdl-parser-py ros-melodic-laser-pipeline
  ros-melodic-librviz-tutorial ros-melodic-moveit-experimental ros-melodic-nodelet-core
  ros-melodic-nodelet-tutorial-math ros-melodic-perception ros-melodic-perception-pcl
  ros-melodic-pluginlib-tutorials ros-melodic-polled-camera ros-melodic-position-controllers
  ros-melodic-qt-dotgraph ros-melodic-qt-gui-cpp ros-melodic-qt-gui-py-common
  ros-melodic-qwt-dependency ros-melodic-robot ros-melodic-ros-base ros-melodic-ros-comm
  ros-melodic-ros-core ros-melodic-ros-tutorials ros-melodic-roscpp-core ros-melodic-roslint
  ros-melodic-roslisp ros-melodic-rosserial-client ros-melodic-rqt-action
  ros-melodic-rqt-bag-plugins ros-melodic-rqt-common-plugins ros-melodic-rqt-dep
  ros-melodic-rqt-graph ros-melodic-rqt-gui-cpp ros-melodic-rqt-image-view ros-melodic-rqt-launch
  ros-melodic-rqt-moveit ros-melodic-rqt-msg ros-melodic-rqt-nav-view ros-melodic-rqt-plot
  ros-melodic-rqt-pose-view ros-melodic-rqt-publisher ros-melodic-rqt-py-console
  ros-melodic-rqt-robot-dashboard ros-melodic-rqt-robot-monitor ros-melodic-rqt-robot-plugins
  ros-melodic-rqt-robot-steering ros-melodic-rqt-runtime-monitor ros-melodic-rqt-rviz
  ros-melodic-rqt-service-caller ros-melodic-rqt-shell ros-melodic-rqt-srv ros-melodic-rqt-tf-tree
  ros-melodic-rqt-top ros-melodic-rqt-topic ros-melodic-rqt-web ros-melodic-rviz-plugin-tutorials
  ros-melodic-rviz-python-tutorial ros-melodic-self-test ros-melodic-smach ros-melodic-smach-msgs
  ros-melodic-smach-ros ros-melodic-stage ros-melodic-stage-ros ros-melodic-theora-image-transport
  ros-melodic-transmission-interface ros-melodic-turtle-actionlib ros-melodic-turtle-tf
  ros-melodic-turtle-tf2 ros-melodic-turtlebot3-fake ros-melodic-turtlesim
  ros-melodic-urdf-parser-plugin ros-melodic-urdf-tutorial ros-melodic-vision-opencv
  ros-melodic-visualization-marker-tutorials ros-melodic-visualization-tutorials ros-melodic-viz
  ros-melodic-webkit-dependency ruby ruby-did-you-mean ruby-minitest ruby-net-telnet
  ruby-power-assert ruby-test-unit ruby2.5 rubygems-integration sbcl ttf-dejavu-core
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 27 not upgraded.
```
バージョンがあってないということでしょうか？正直よくわかっていません...
教えてもらえると助かります。

k-okada, 2019-10-27:
osrfoundation.org が gazebo 9.11を入れてしまっているので，package.ubuntu.com
に戻らなくなっている，という状況かとおもいます．
一回 sudo apt-get remove libgazebo9 libgazebo9-dev
としてみましょう．
--
◉ Kei Okada

poyotamu000, 2019-10-27:
なんとか環境の再建ができました！ありがとうございます！
もうdockerで構築しようと思っていたところなので、助かりました！

catkin build の際に、以下(catkin build から抜粋)の warning がでたのですが 大丈夫でしょうか？一応gazebo は起動することができ、rqt による指令をうけとることも確認しました。

また、rosdep install の際の pr2eus/roseus-tutorials のエラーについてもよくわかっていないので、簡単にでも説明していただけるとありがたいです。

```
_____________________________________________________________________________________________________
Warnings   << kobuki_gazebo_plugins:check /home/mech-user/catkin_ws/logs/kobuki_gazebo_plugins/build.check.038.log
CMake Warning (dev) at /usr/share/cmake-3.10/Modules/FindBoost.cmake:911 (if):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "chrono" will no longer be dereferenced when the
  policy is set to NEW.  Since the policy is not set the OLD behavior will be
  used.
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:1558 (_Boost_MISSING_DEPENDENCIES)
  /usr/share/OGRE/cmake/modules/FindOGRE.cmake:318 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:175 (find_package)
  CMakeLists.txt:4 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /home/mech-user/catkin_ws/build/kobuki_gazebo_plugins; catkin build --get-env kobuki_gazebo_plugins | catkin env -si  /usr/bin/make cmake_check_build_system; cd -
.....................................................................................................
Finished  <<< turtlebot_follower                        [ 1.2 seconds ]                              
Starting  >>> turtlebot_rapps                                                                        
Finished  <<< turtlebot_rapps                           [ 0.2 seconds ]                              
Finished  <<< turtlebot_actions                         [ 2.0 seconds ]                              
Finished  <<< kobuki_gazebo_plugins                     [ 14.8 seconds ]                             
Starting  >>> kobuki_gazebo                                                                          
Starting  >>> turtlebot_gazebo                                                                       
Finished  <<< kobuki_gazebo                             [ 0.2 seconds ]                              
Starting  >>> dxl_armed_turtlebot                                                                    
Finished  <<< turtlebot_gazebo                          [ 0.1 seconds ]                              
Finished  <<< dxl_armed_turtlebot                       [ 0.1 seconds ]                              
Starting  >>> daisya_euslisp_tutorials                                                               
Finished  <<< daisya_euslisp_tutorials                  [ 0.1 seconds ]                              
[build] Summary: All 35 packages succeeded!                                                          
[build]   Ignored:   5 packages were skipped or are blacklisted.                                     
[build]   Warnings:  1 packages succeeded with warnings.                                             
[build]   Abandoned: None.                                                                           
[build]   Failed:    None.                                                                           
[build] Runtime: 16.7 seconds total.                                                                 
```

k-okada, 2019-10-28:
warning について他の人も出ているものなのでこれでOKとしましょう．

pr2eus/roseus-tutorialsについては，http://packages.ros.org からインストールするための準備が終わっていない段階で，https://discourse.ros.org/t/preparing-for-melodic-sync-2019-09-24/10790/6 で依頼中です．

closed 2019-11-01 by poyotamu000