jsk-enshu/robot-programming #304: 課題4-キー入力イベントを挿入する位置について

knt-fukushima, 2019-10-26:
今週の課題4（自律移動ロボットプログラムの send_goal_actionlib.py ,または演習ロボットのsend_joint_position_actionlib.py を変更し,キーボード入力等のイベントがあった場合に,移動行動がキャンセルされるようにせよ.）の実装方法についての質問です。

コードのどの位置にキー入力イベント判定を入れればよいかがわかっていません。
while文で常にキー入力を監視する方法を考え、これをsend_goalとwait_for_resultの間に挟むことによって行動開始時からゴール到着時の間にキー入力イベントを検知できると考えました。

僕が様々に試して一番答えに近そうな方法では、「キー入力が途中でされた場合行動のキャンセルができる」という動きはできるのですが、
「キー入力がなされない場合while文から抜けられず、wait_for_result()が実行されないためプログラムが終了しない」状態になります。

どのように考えれば答えに近づけそうでしょうか？
ご回答、よろしくお願いいたします。

以下は現時点でのコードです（send_goal_actionlib.py）

`#! /usr/bin/env python
# -*- coding: utf-8 -*-

import rospy
import actionlib
from move_base_msgs.msg import *
import sys
import select
import termios
import tty

def getKey():
    if select.select([sys.stdin], [], [], 0) == ([sys.stdin], [], []):
        key = sys.stdin.read(1)
        return key
    return None

if __name__ == '__main__':
    settings = termios.tcgetattr(sys.stdin)
    tty.setcbreak(sys.stdin.fileno())
    try:
        rospy.init_node('send_goal', anonymous=True)
        client = actionlib.SimpleActionClient('move_base', MoveBaseAction)
        client.wait_for_server()
        goal = MoveBaseGoal()
        goal.target_pose.header.stamp = rospy.Time.now()
        goal.target_pose.header.frame_id = 'map'
        goal.target_pose.pose.position.x = 34
        goal.target_pose.pose.position.y = 32
        goal.target_pose.pose.orientation.w = 1
        rospy.loginfo("send goal")
        rospy.loginfo(goal)
        client.send_goal(goal)
        rospy.loginfo("wait for goal ...")

#主な変更部分---
        while (1):
            key = getKey()
            if (key == 'c'):
                client.cancel_goal()
                rospy.loginfo("goal canceled")
                break
#主な変更部分---

        ret = client.wait_for_result()

        rospy.loginfo("done")
    except rospy.ROSInternalException:
        pass

    finally:
        termios.tcsetattr(sys.stdin, termios.TCSADRAIN, settings)
`

k-okada, 2019-10-27:
while文で毎回 ActionLib（サーバ）の状況を調べることが出来るので，例えば，`client.get_state()` などを使って，ロボットがゴールに到達したか，あるいは，その途中なのか，調べるとよいかとおもいます．
```
            rospy.loginfo('{}: PENDING/{}, ACTIVE/{}'.format(client.get_state(), actionlib.GoalStatus.PENDING, actionlib.GoalStatus.ACTIVE))
            rospy.loginfo(client.get_goal_status_text())
```
などのコードを入れてみるとよいでしょう

knt-fukushima, 2019-10-27:
ご回答ありがとうございます。

教えていただいた`client.get_state()`について詳細を調べ、
現在の状況に対応する整数が返される仕様になっていることを理解しました。
それをもとに、while文の中身を以下のように書き換えることで
期待通りの動きをするようになりました。

ありがとうございました。

```
while (1):
            
            key = getKey()
            if (client.get_state() == 3):
                rospy.loginfo("done")
                break
            else:
                if (key == 'c'):
                    client.cancel_goal()
                    rospy.loginfo("goal canceled")
                    break
        
```