jsk-enshu/robot-programming #312: roslaunch dynamixel_7dof_arm dynamixel_7dof_am_bringup.launchが時々落ちる

mqcmd196, 2019-11-14:
3.3.3のdotimesのコードを動かすと時々
```
[INFO] [1573713115.030327]: Trajectory start time is 1573713115.036, end time is 1573713117.907, total duration is 3.000
shutdown request: new node registered with same name
Exception in thread Thread-59:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/mech-user/catkin_ws/src/dynamixel_motor/dynamixel_controllers/src/dynamixel_controllers/joint_trajectory_action_controller.py", line 359, in update_state
    self.state_pub.publish(self.msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 1041, in publish
    raise ROSException("publish() to a closed topic")
ROSException: publish() to a closed topic

================================================================================REQUIRED process [dynamixel_manager-1] has died!
process has finished cleanly
log file: /home/mech-user/.ros/log/86cc9b3e-06a4-11ea-b6d4-207918595aec/dynamixel_manager-1*.log
Initiating shutdown!
================================================================================
[dynamixel_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

というエラーが出て，マスターのプログラムが落ちます

k-okada, 2019-11-15:
```
shutdown request: new node registered with same name
```
で同じ物を誰かも動かしているきがします．

mqcmd196, 2019-11-19:
解消しました．ありがとうございました

closed 2019-11-19 by mqcmd196