jsk-enshu/robot-programming #318: メソッドが見つからないというエラーについて

knt-fukushima, 2019-11-24:
今週提出の宿題2番について質問させていただきます。
（「 3.1.3 ロボットの目標位置指令プログラム (ActionLib 版 ) 」を参考に ActionLib を用いてmove_base にゴールを送信するプログラムを作成せよ.）

以下のコードを書きましたが、実行すると
ERROR th=0 cannot find method :target_pose:target_pose in (send goal :target_pose)E: 
というエラーが返ってきます。間違いが何か教えていただけますと幸いです。

（いくつかの資料を読み比べながら、書き方を模索しつつコードを書いています。
そのため、間違い箇所の同定が難しく、質問させていただきます。）

以下コード
`
#!/usr/bin/env` roseus

(ros::roseus-add-msgs "move_base_msgs")

(ros::roseus "send_goal" :anonymous t)
(setq *c* (instance ros::simple-action-client :init
		    "move_base" move_base_msgs::MoveBaseAction))
(ros::ros-info "wait-for-server")
(send *c* :wait-for-server)
(setq goal (instance move_base_msgs::MoveBaseAction :init))
(send (send (send goal :target_pose) :header) :frame_id 'map')
(send (send (send (send goal :target_pose) :pose) :position) :x 34)
(send (send (send (send goal :target_pose) :pose) :position) :y 32)
(send (send (send (send goal :target_pose) :pose) :orientation) :w 1)      

(ros::ros-info "send_goal")
(send *c* :send-goal goal)
(ros::ros-info "wait for goal ...")
(send *c* wait-for-result)

(send (send *c* :get-result) :sequence)
(ros::ros-info `"done")
`

k-okada, 2019-11-24:
メッセージの内容を見ると以下のようになるので、
`(setq goal move_base_msgs::MoveBaseActionGoal :init)`とするか、`(send (send goal :action_goal) :target_pose)`でしょうか。
```
$ rosmsg show move_base_msgs/MoveBaseAction
move_base_msgs/MoveBaseActionGoal action_goal
  std_msgs/Header header
    uint32 seq
    time stamp
    string frame_id
  actionlib_msgs/GoalID goal_id
    time stamp
    string id
  move_base_msgs/MoveBaseGoal goal
    geometry_msgs/PoseStamped target_pose
```

knt-fukushima, 2019-11-24:
ご返信ありがとうございます。
`(setq goal move_base_msgs::MoveBaseActionGoal :init)`
とすることによって、メソッドとの対応が取れ、エラーは消えました。

ただ、先程までのコードではタイムスタンプが指定されていないためか、
プログラムは走ってもロボットが移動を開始しません。
そこで、タイムスタンプを指定しようと考えたのですが、
以下のコード
`(send (send (send goal :target_pose) :header) :stamp (send (ros::time-now) :sec-nsec))`
では
`ERROR th=0 cannot find method :sec:sec in (send ros::_stamp :sec)E:`
とエラーが返ってきます。（上記のコードは資料roseus 2.1「 シンプルなパブリッシャを書く」に習ったものです。）
pythonで言うところの`rospy.Time.now()`はどのように取得し、stampに反映させればよいでしょうか。

現状のコード状況
```
#!/usr/bin/env roseus

(ros::roseus-add-msgs "move_base_msgs")

(ros::roseus "send_goal" :anonymous t)
(setq *c* (instance ros::simple-action-client :init
		    "move_base" move_base_msgs::MoveBaseAction))

(ros::ros-info "wait-for-server")
(send *c* :wait-for-server)
(setq goal (instance move_base_msgs::MoveBaseGoal :init))
**(send (send (send goal :target_pose) :header) :stamp (send (ros::time-now) :sec-nsec))**
(send (send (send goal :target_pose) :header) :frame_id 'map')
(send (send (send (send goal :target_pose) :pose) :position) :x 34)
(send (send (send (send goal :target_pose) :pose) :position) :y 32)
(send (send (send (send goal :target_pose) :pose) :orientation) :w 1)      

(ros::ros-info "send_goal")
(send *c* :send-goal goal)
(ros::ros-info "wait for goal ...")
(send *c* :wait-for-result)

(send (send *c* :get-result) :sequence)
(ros::ros-info "done")
```

k-okada, 2019-11-25:
`(send (send (send goal :target_pose) :header) :stamp (ros::time-now)))`
で動くでしょうか．
--
◉ Kei Okada