jsk-enshu/robot-programming #319: メソッドが見つからないというエラーについて2

knt-fukushima, 2019-11-24:
今週提出の宿題3番について質問させていただきます。
(「 2.3 ロボットアームの角度制御」節を参考に /fullbody_controller/command をパブリッシュするプログラムを作成せよ.)

資料2.3でいうところのpythonコード下1/3付近
`joint_trajectory.points.append(point)`を以下のようにlispに書き換えようと試みました。
`  (send (send joint_trajectory :points) :append point))`
結果、
`/opt/ros/melodic/share/euslisp/jskeus/eus/Linux64/bin/irteusgl: ERROR th=0 cannot find method :append:append in (send (send joint_trajectory :points) :append point)E: `
というエラーが返ってきます。

pythonでの構造を見る限り、
joint_trajectoryインスタンス内のpoints変数にはappendというメソッドがあると考えています。
ゆえに、(send インスタンス method名 method 引数)の形で実行可能かと考えたのですが、うまくいかないようです。
lispでの書き方を把握できていないゆえの問題かとは思いますが、
どのようにすればインスタンス内にあるであろうappendメソッドにpointを渡すことができますでしょうか。

k-okada, 2019-11-24:
`.append` はListに要素を追加する、というPythonの組み込みのメンバ関数です。
Lispで要素を追加するのは`(append (list 1 2 3) (list 4 5 6))`などとします。
ただし、Lispでは関数型言語で、`(setq a (list 1 2 3)) (append a (list 4 5 6))`としても
`a` の値が変わるわけではないので、
```
(setq point_msg (instance trajectory_msgs::JointTrajectoryPoint :init))
(setq traj_msg (instance trajectory_msgs::JointTrajectory :init))
(send traj_msg :points (list point_msg))
```
などとするのが良いかと思います。

knt-fukushima, 2019-11-24:
ご返答ありがとうございます。
.appendがpythonの組み込みメンバ関数である点と、lispではappendを用いても、もとのlist自体が変更されない旨、理解しました。

ご返信のコードについてですが、
`(send traj_msg :points (list point_msg))`
の(list point_msg)は具体的にはdurationによる開始時間とpositionによる関節角度リスト、が複数セット入っているという理解であっていますか。
（point_msg をlist関数にかけると何がどのようになっているはずなのでしょうか。）

ちなみに、現状のコードは以下のようです。試行錯誤中なので、エラーをはきます。

```
#!/usr/bin/env roseus

(ros::roseus-add-msgs "trajectory_msgs")
(ros::roseus-add-msgs "control_msgs")

(ros::roseus "send_joint_position")
(ros::advertise "/fullbody_controller/command" trajectory_msgs::JointTrajectory 1)
(ros::sleep)
(setq joint_trajectory (instance trajectory_msgs::JointTrajectory :init))
(send (send joint_trajectory :header) :stamp (send (ros::time-now) :sec-nsec))
(send joint_trajectory :joint_names (list 'arm_joint1 'arm_joint2 'arm_joint3 'arm_joint4 'arm_joint5 'arm_joint6))
;;ここで複数のpointが入る変数をセット
(setq point_list (instance trajectory_msgs::JointTrajectoryPoint :init))
(dotimes (i 5)
  (setq point (instance trajectory_msgs::JointTrajectoryPoint :init))
  (send point :positions (list (/ pi 2) 0 (* (/ pi 4) (mod i 2)) 0 (/ pi 2) (/ pi 2)))
  (setq duration (instance ros::duration :init (+ 1.0 i)))
  (send point :time_from_start duration)
;;直前で作成したpointをpoint_listに追加、それをtmpに一旦入れておく
  (setq tmp (append point_list point))
;;global変数のpoint_listにtmpを入れる。ここまでで、(point_list にpointを追加し変化させる、を実現したかった。）
  (setq point_list tmp)
  (send joint_trajectory :points tmp)))
(ros::publish "/fullbody_controller/command" joint_trajectory)
(ros::sleep)
```

k-okada, 2019-11-25:
(send joint_trajectory :points (list point))
とかでよいかとおもいます．
--
◉ Kei Okada