jsk-enshu/robot-programming #320: rosdep install と catkin build でのエラー

kotarouetake, 2020-04-10:


k-okada, 2020-04-12:
以下の様にしてみてください．
```
cd catkin_ws/src
catkin clean -y
rm -fr linux_* libfree*
wstool mserge https://raw.githubusercontent.com/jsk-enshu/robot-programming/bbf6d8af5e90ffb09b40b6526b6a951887964085/.rosinstall.melodic
wstool update
catkin merge
```
ちなみにこのエラーの前にlibfreenctが，，というエラーが出そうなもんなんだけど，そういうのはなかったのかな．．．

kotarouetake, 2020-04-12:
返信ありがとうございます。

上記を試してみたのですが、相変わらずビルド時にいくつかビルド失敗してしまいました(上記ではcatkin mergeとなっていますが、catkin buildで良かったでしょうか？catkin mergeだと”コマンドが見つかりませんでした”と出ました)

また、講義資料に従いゼロからワークスペース作成を行いlibfreenectに関するエラーが出ていないか確認しましたが、rosdep install の際に出たもの以外エラーとして出ているものが見つかりませんでした。

k-okada, 2020-04-12:
いくつかtypoを直したのでもう一回試してください．`rosdep install`は`-r`オプションがついていますが，これは`rosdep install --help`とすると`  -r                    Continue installing despite errors.`と出てくるので，そこでのエラーは無視してよいです．
むしろ`catkin build`のところで`freenct`関連でエラーが出ると思ったんですが．．．

PR #321 (2020-06-24): update .rosinstall.meldoic to use Oct/2019 version of source
- please always  add version to rosinstall !!
 - do not install library via apt, use rosdep!

Closes #320
closed 2020-05-31 by ktro4989