jsk-enshu/robot-programming #322: 角度変更ができない

samejima00, 2020-06-29:
robot-programmingディレクトリにおいてroscd dxl_armed_turtbot/launchを実行するとroscd: No such package/stack 'dxl_armed_turtbot/launch'と言われます。
Gazeboを開いているのとは別の端末でrobot-programmingディレクトリに行って実行しているのですが、それは正しいのでしょうか？

また、catkin buildを実行した際以下のような出力が出ます。授業で見た出力と異なるように感じているのですが、原因は何でしょうか。
`[build] Summary: 7 of 42 packages succeeded.                                   
[build]   Ignored:   7 packages were skipped or are blacklisted.               
[build]   Warnings:  None.                                                     
[build]   Abandoned: 34 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 4.4 seconds total.  `

色々情報が足りないとは思いますが、よろしくお願いします。

k-okada, 2020-06-29:
```
catkin clean -y
catkin build -vi --make-args VERBOSE=1 2>&1 | tee build.log
```
としてでてきたbuild.log を共有してください．個々に直接貼っても良いですし，https://gist.github.com/ にファイルをアップロードしてリンクを貼ってくれるのでも良いです．

gawa-K, 2020-07-01:
私も同様の症状で以上のコードを実行しました。
大量に出力されたメッセージの最後の部分が以下のようなメッセージでした。
```
Finished <<< turtlebot_rapps                              [ 2.2 seconds ]      
[build] Summary: All 43 packages succeeded!                                    
[build] Ignored: 7 packages were skipped or are blacklisted.                   
[build] Warnings: 5 packages succeeded with warnings.                          
[build] Abandoned: No packages were abandoned.                                 
[build] Failed: No packages failed.                                            
[build] Runtime: 1 minute and 33.0 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

k-okada, 2020-07-02:
@gawa-K は`catkin build` は成功していて，その後の`source devel/setup.bash` してから，‘roscd dxl_armed_turtlebot/launch` してみてください．
（ちなみに今気が付きましたが @samejima00 は`dxl_armed_turtbot`　でタイポでしょうか？）
また，@gawa-K はログは全部アップロードしてましょう．

gawa-K, 2020-07-02:
ファイルにまとめてアップロードするべきでした。

`roscd dxl_armed_turtlebot/launch`
で動きました。
資料が
`roscd dxl_armed_turtbot/launch`
となっていて動かなかったのが原因のようです。

k-okada, 2020-07-02:
あ，それは失礼しました．

gawa-K, 2020-07-02:
親切なご対応ありがとうございました。