jsk-enshu/robot-programming #323: fetchのレポジトリ名が勝手に変更される

gawa-K, 2020-06-29:
`~/catkin_ws/src/robot-programming`ディレクトリ内で
`git remote add gawa-K http://github.com/gawa-K/robot-programming`を実行し、
`git fetch --all`を実行すると
```
remote: Repository not found.
fatal: repository 'https://github.com/gawa-K/robot-programmig/' not found
error: Could not fetch gawa-K
```
とエラーが出る。
`git remote add ~`でのURLは確かに存在している。
しかし `git fetch --all` でURL末尾が`programmig`である存在しないURLを指し示す。

k-okada, 2020-06-29:
```
git remote add gawa-K http://github.com/gawa-K/robot-programmingを実行し、
```
は`programing` ですが，
```
fatal: repository 'https://github.com/gawa-K/robot-programmig/' not found
```
は`programmig` です．最初の
addしたときにタイポがありそうです．

```
git remote remove gawa-K
git remote add gawa-K http://github.com/gawa-K/robot-programming
```
として作り直してみてください．

gawa-K, 2020-06-29:
```
git remote remove gawa-K
git remote add gawa-K http://github.com/gawa-K/robot-programming
git fetch --all
```
で改善しました。

closed 2020-07-01 by gawa-K