jsk-enshu/robot-programming #325: catkinbuildとroslaunchができない

tenrobo18, 2020-09-29:
学科PCで~/catkin_wsでcatkinbuildをしたところ添付のcatkinbuild_failのようなエラーが出たのですが、どのような原因が考えられますでしょうか。
また自分のgazeboのバージョン9.14になっており、学科の正常にシミュレーションが起動した友人に聞いてみたところ9.0となっているそうなのでこれが一因となっていることは考えられますでしょうか。

さらに
roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch
をしたところ添付のroslaunch_failのようなエラーが出てしまったのですがどのような原因が考えられますでしょうか。

よろしくお願いいたします
https://gist.github.com/tenrobo18/ac5a934f73f3966930271809f3a06abe

k-okada, 2020-09-29:
```
$ cd ~/catkin_ws
$ rosdep install --from-paths src --ignore-src -y -r
```
はしてみたかな？

tenrobo18, 2020-09-29:
$ rosdep install --from-paths src --ignore-src -y -r
をしたらcatkin buildもroslauchも通り、シミュレーターが正常に表示されるようになりました。

ありがとうございます