jsk-enshu/robot-programming #328: rqtのcontrollerプルダウンメニューが出てこない

ketaro-m, 2020-10-04:
2020/10/12の演習資料の3.1にあるように環境構築をし、
`roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch`
として動力学シミュレータは立ち上げた後
`rqt --perspective-file enshu.perspective`
として制御ツールを立ち上げたのですが、controllerのプルダウンタブが出てきません。

画像にあるように、gazeboを立ち上げた際にcontrollerに関するエラーが出ているので、おそらくはこの辺りが問題だとは思うのですが、原因がわかりますでしょうか。
[image]

~/.ros/log/latest/　以下のログファイルや、gazeboを立ち上げたときのターミナル出力は以下に置いてあります。
https://gist.github.com/ketaro-m/52cfeeeffa0d5ecd4bffaeb4a50184db

k-okada, 2020-10-04:
良い質問ですね．
重要そうなエラーメッセージ
https://gist.github.com/ketaro-m/52cfeeeffa0d5ecd4bffaeb4a50184db#file-controller_spawner-11-log-L53
で調べてみると以下があたりました．
https://answers.ros.org/question/360537/unknown-error-handler-name-rosmsg/
にかいてある以下を試すとどうでしょうか？
ちなみに現状のgenpyのバージョンを知りたければrosversion genpyとします．

sudo apt-get update && sudo apt-get upgrade

ketaro-m, 2020-10-04:
そのようにしたら直りました。
ありがとうございます。

closed 2020-10-09 by ketaro-m