jsk-enshu/robot-programming #329: ロボットシステム１０月５日課題１Pythonファイルで

kohtaro246, 2020-10-04:


ishiguroJSK, 2020-10-09:
```
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
```
というあたりからファイルに実行権限がついてないんじゃないかと思います．
`sudo chmod +x aaa.py`みたいな手順が必要です．
ROSのコマンドラインはTab補完が効く場合が多いので，補完されなかったら何かを間違えてると疑うとよいです．
```
~/catkin_ws/enshu/src/robot-programming/dxl_armed_turtlebot/scripts (2020) $ rosrun dxl_armed_turtlebot tuck_arm_non_exe.py
[rosrun] Couldn't find executable named tuck_arm_non_exe.py below /home/leus/catkin_ws/enshu/src/robot-programming/dxl_armed_turtlebot
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /home/leus/catkin_ws/enshu/src/robot-programming/dxl_armed_turtlebot/scripts/tuck_arm_non_exe.py
~/catkin_ws/enshu/src/robot-programming/dxl_armed_turtlebot/scripts (2020) $ ls -al
合計 28
drwxr-xr-x  2 leus leus 4096 10月  9 20:43 ./
drwxr-xr-x 12 leus leus 4096 10月  2 14:06 ../
-rwxr-xr-x  1 leus leus  190 10月  2 14:06 install_ps3joy_daemon.sh*
-rwxr-xr-x  1 leus leus 3489 10月  2 14:06 install_turtlebot_simulator.sh*
-rwxr-xr-x  1 leus leus 1844 10月  2 14:06 ps3joy*
-rwxr-xr-x  1 leus leus 3663 10月  2 14:06 tuck_arm.py*
-rw-r--r--  1 leus leus 3663 10月  9 20:43 tuck_arm_non_exe.py
```

kohtaro246, 2020-10-11:
ありがとうございます。実行権限を付け忘れていました。助かりました

closed 2020-10-28 by kohtaro246