jsk-enshu/robot-programming #330: １０月５日課題２：client.getResult()によって得られる計算結果の型がわからない

kohtaro246, 2020-10-04:
課題２(ActionLibの利用)でserverで行った足し算の結果をclient側で表示したいのですが、client.getResult()によって得られる計算結果の型がわからず、C++でprintfすることができません。表示方法をご教示いただけると幸いです。以下にclient側のプログラムを添付致します。
[image]

k-okada, 2020-10-04:
ソースコード（C++ファイル，actionファイル，CMakeLists.txt)ファイル．
```
catkin build 2>&1 | tee build.log
```
で出てくる`build.log`ファイルを教えてくれると助かります．

kohtaro246, 2020-10-04:
<script src="https://gist.github.com/kohtaro246/f7113df0f0e5d9e91edcf10b3ae7959a.js"></script>

kohtaro246, 2020-10-04:
https://gist.github.com/kohtaro246/f7113df0f0e5d9e91edcf10b3ae7959a

kohtaro246, 2020-10-04:
ちなみに、計算結果をprintする部分をコメントアウトすると、正しく計算できていることはわかります。
[image]

k-okada, 2020-10-04:
```
diff --git a/simple_addition_client.cpp b/simple_addition_client.cpp
index 677c704..a49b9e8 100644
--- a/simple_addition_client.cpp
+++ b/simple_addition_client.cpp
@@ -16,7 +16,14 @@ int main(int argc, char** argv)
     printf("Calculation Completed\n");
   printf("State: %s\n",client.getState().toString().c_str());
   
-  /*printf("Solution: %s\n",client.getResult().toString().c_str());  */
+  printf("Solution: %f\n",client.getResult()->solution);
+  //ROS_INFO_STREAM("Solution: " << *(client.getResult()));
+  beginner_tutorials::DoAdditionResultConstPtr result = client.getResult();
+  ROS_INFO("Solution: %f\n", result->solution);
+  ROS_INFO_STREAM("Solution: " << result->solution);
+  ROS_INFO_STREAM("Solution: " << client.getResult()->solution);
+  ROS_INFO("Solution: %f\n", client.getResult()->solution);
```
と変更してみてください．

`getResult()` はactionのresultへのポインタを返しますが，これは，`DoAddition.action`ファイルの
```
# Define the result
float32 solution
```
に相当します．`solution`は`float32`なので，そのまま`client.getResult()->solution`でよいです．

`getState()`は
http://docs.ros.org/jade/api/actionlib/html/classactionlib_1_1SimpleActionClient.html#a16a367c503a355b3872f046cdf914c65
http://docs.ros.org/jade/api/actionlib/html/classactionlib_1_1SimpleClientGoalState.html
を返しますが，ここでは`toString()`の返り値が`std:string`になっているので，`c_str()`が必要です．

ちなみに，ROSノード内で表示するときは
http://wiki.ros.org/roscpp/Overview/Logging
にある
`ROS_INFO`を使うと良いです．また，`ROS_INFO_STREAM`を使うと，C++のように型を意識せずに書けるので，
```
  ROS_INFO_STREAM("State: " << client.getState().toString());
```
や
```
  std::cout << "State: " << client.getState().toString() << std::endl;
```
と書くよいです．

kohtaro246, 2020-10-05:
できました。ありがとうございます。

closed 2020-10-05 by kohtaro246