jsk-enshu/robot-programming #333: listener.cppのコンパイルが通らない

softyanija, 2020-10-06:
1005の資料でlistener.cppをコンパイルするときに、以下のようなエラーを吐かれて失敗してしまいます。このエラーメッセージの示すものがわからないのでどうすればいいか教えてほしいです。
[image]
[image]
[image]

k-okada, 2020-10-07:
https://github.com/jsk-enshu/robot-programming/issues/89#issuecomment-256596336
と同じで`catkin create pkg` のところで間違えている可能性が高いです．
CMakeLists.txtを確認してみてください．
https://github.com/jsk-enshu/robot-programming/issues/95#issuecomment-256838850
みたいに
```
find_package(catkin REQUIRED COMPONENTS
（中略）
)
```
の中略のところに何が書いてあるか確認してみてください．

softyanija, 2020-10-07:
find_package(catkin REQUIRED COMPONENTS
  std_msgs
)
となっています。上げてもらったものを参考にしてcatkin create pkgを見直してみます

softyanija, 2020-10-07:
資料の
> catkin create pkg beginner_tutorials --catkin-deps std_msgs
　 geometry_msgs rospy roscpp message_generation message_runtime
の部分の解釈を間違えて上半分しか書いていませんでした.

softyanija, 2020-10-07:
`find_package(catkin REQUIRED COMPONENTS
  geometry_msgs
  message_generation
  message_runtime
  roscpp
  rospy
  std_msgs
)`
のようになり無事コンパイルが通りました。ありがとうございます。