jsk-enshu/robot-programming #334: Actionlibについて

ummaaa, 2020-10-07:
actionlibのプログラムを資料そのまま写したのですが、実行しても何も起こりません。自分で何か追加しないといけないのでしょうか。

k-okada, 2020-10-07:
どのようなプログラムを書いたか，ファイル一式を gist.github.com に書いてください．

ummaaa, 2020-10-07:
https://gist.github.com/ummaaa/38af5c7a5688f153d97878258e95ddc2
よろしくお願いします。

k-okada, 2020-10-08:
ごめんなさい，CMakeFile.txt と package.xml も追加してください．
また，コンパイルしたときのログ，＋実行したときのソロぞれのターミナルの画面キャプチャを教えてください．

参考：https://github.com/jsk-enshu/robot-programming/issues/330#issuecomment-703249861

ummaaa, 2020-10-08:
https://gist.github.com/ummaaa/38af5c7a5688f153d97878258e95ddc2

今実行してみたらclient側に2枚目の写真のようなものが表示されたのですがこれでできているでしょうか。上のリンクに足りなかったコードは追加しておきました。よろしくお願いします。

[image]
[image]

k-okada, 2020-10-08:
いい感じですね．
Pythonのコードのほうが親切な感じですが，
C++も以下のように変更すると，何が怒っているか見やすいかとおもいます．

```
diff --git a/simple_action_client.cpp b/simple_action_client.cpp
index 5457e90..2d98617 100644
--- a/simple_action_client.cpp
+++ b/simple_action_client.cpp
@@ -9,11 +9,13 @@ int main (int argc, char** argv) {
   client.waitForServer();
   beginner_tutorials::DoDishesGoal goal;
   //fill in the goal here
+  goal.dishwasher_id = 99;
   client.sendGoal(goal);
   client.waitForResult(ros::Duration(5.0));
   if (client.getState() == actionlib::SimpleClientGoalState::SUCCEEDED) {
-    printf("Yay! The dishes are now clean");
+    printf("Yay! The dishes are now clean\n");
   }
   printf("Current State: %s\n", client.getState().toString().c_str());
+  printf("Current Result: %d\n", client.getResult()->total_dishes_cleaned);
   return 0;
 }
diff --git a/simple_action_client.py b/simple_action_client.py
old mode 100644
new mode 100755
diff --git a/simple_action_server.cpp b/simple_action_server.cpp
index c6b6787..577e769 100644
--- a/simple_action_server.cpp
+++ b/simple_action_server.cpp
@@ -5,7 +5,10 @@ typedef actionlib::SimpleActionServer<beginner_tutorials::DoDishesAction> Server
 
 void execute (const beginner_tutorials::DoDishesGoalConstPtr& goal, Server* as) {
   //do lots of awesome groundbreaking robot stuff here
-  as -> setSucceeded();
+  ROS_INFO("do lots of awesome groundbreaking robot stuff here");
+  beginner_tutorials::DoDishesResult result;
+  result.total_dishes_cleaned = 999;
+  as -> setSucceeded(result);
 }
 
 int main (int argc, char** argv) {
```

ummaaa, 2020-10-08:
修正してみます。ありがとうございました。

closed 2020-10-12 by ummaaa