jsk-enshu/robot-programming #335: catkin build とcatkin_make ni

ketaro-m, 2020-10-09:
演習でホームディレクトリ直下に ~/catkin_ws というディレクトリを作って catkin build でパッケージをビルドしたのですが、他の要件で新たなパッケージをインストールする必要があり、そのセットアップには catkin_make を使うように書かれています。
色々と調べてみると、catkin buildとcatkin_makeは別のビルド方法であって共存できないとあり、またcatkin_makeを使う場合は ~/catkin_ws のみでしか使えないとあります。
[http://wiki.ros.org/catkin/commands/catkin_make](http://wiki.ros.org/catkin/commands/catkin_make)

この場合は、どのように解決するのが良いのでしょうか。
また、catkin_make は~/catkin_ws のみに作れるということは、複数のROSのプロジェクトを同じPCで行う場合も、それぞれ別のディレクトリで独立に作るのではなく、全てのパッケージを~/catkin_ws に作るのが普通の作り方なのでしょうか。
それとも、独立のフォルダを作ってそれぞれcatkin buildでビルドする方が望ましいのでしょうか。

ishiguroJSK, 2020-10-09:
catkin_makeできるROSパッケージならcatkin buildできるはずです．
ので，catkin buildでワークスペースを作っている人は，
新しく追加したいパッケージをワークスペースに置いて(必要ならrosdepなどして)catkin bするだけで良いはずです．

また，全く別のROSプロジェクト(授業A，授業B，ロボコンetc...)でいろいろ混ざるのを避けたい場合，
catkinワークスペースを分けるのが良いとされています．
分け方のディレクトリ構成は色々ありますが，
（
```
~/catkin_ws/my_workspace1/src
~/catkin_ws/my_workspace2/src
~/catkin_ws/my_workspace3/src
```
とか
```
~/catkin_ws1/src
~/catkin_ws2/src
~/catkin_ws3/src
```
）
いずれにしても現在扱いたいcatkinワークスペースを`source ~~/~~/setup.bash`するので完全に切り替えられます．

> またcatkin_makeを使う場合は ~/catkin_ws のみでしか使えないとあります。

これもsourceでcatkinワークスペースを切り替える方法で~/catkin_ws以外も出来るはずです

ketaro-m, 2020-10-09:
お返事ありがとうございます。catkin buildでも作ることができました。
稚拙な質問かもしれませんが、catkinワークスペースは例に挙げていただいたようにホームディレクトリ直下に置くのが普通なのでしょうか？

ishiguroJSK, 2020-10-09:
普通はどうかと言われると難しいですが，上に挙げた例のようにホームディレクトリ直下(直下と言うより以下？)に置きます．
Linux全般の話になりますが，基本，ユーザーが自由にしていいのはホームディレクトリ以下だと思ったほうがいいです．
なので/opt/ros〜とか/usr/local~とかにワークスペースを勝手に作るとややこしいことになります

ketaro-m, 2020-10-09:
ホームディレクトリ直下だけでなくさらに下の階層から作っても構わないかということでしたので、上の回答で解決しました。
ご丁寧な回答ありがとうございました。

closed 2020-10-09 by ketaro-m