jsk-enshu/robot-programming #337: ActionLibのpythonファイル実行時のASCIIに関するエラー

gawa-K, 2020-10-10:
ActionLibのサーバー側pythonファイルを実行（`rosrun beginner_tutorials simple_action_server.py`）すると

```
  File "/home/mech-user/catkin_ws/src/beginner_tutorials/scripts/simple_action_server.py", line 9
SyntaxError: Non-ASCII character '\xe2' in file /home/mech-user/catkin_ws/src/beginner_tutorials/scripts/simple_action_server.py on line 9, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
```
とコード上`self.server = actionlib.SimpleActionServer(’do_dishes’,DoDishesAction,self.execute, False)`部分でのエラーが発生します。

またクライアント側pythonファイルも同様に
```rosrun beginner_tutorials simple_action_client.py```
を実行すると
```
  File "/home/mech-user/catkin_ws/src/beginner_tutorials/scripts/simple_action_client.py", line 8
SyntaxError: Non-ASCII character '\xe2' in file /home/mech-user/catkin_ws/src/beginner_tutorials/scripts/simple_action_client.py on line 8, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details
```
とコード上`if __name__ == ’__main__’:`部分でのエラーが発生します。

なお資料のコードと同様に最初`#! /usr/bin/env python`を書いています。

ishiguroJSK, 2020-10-10:
エラー文的には意図せず全角文字が混ざっているような感じだけど，
line 9をよく見直して異変はないかな？
それでも解決しなかったら自分のコードをまるごとここにコピペしてみてください

k-okada, 2020-10-10:
これ，たぶん`’`が`'`と違うんだよね．で資料をコピペするとこうなるようです．
一回手で打ち込んでみてください．

gawa-K, 2020-10-10:
打ち直したら解決しました。
ありがとうございます。

closed 2020-10-10 by gawa-K