jsk-enshu/robot-programming #340: 10/12の資料、６ページ、rosdep installのエラー

Masayoshi-Dohi, 2020-10-12:
実行すると写真のようなエラーが出ます。
調べても解決方法が分からなかったので教えてください。

[image]

ishiguroJSK, 2020-10-12:
「#All required rosdeps installed successfully」とあるので，先に進んで大丈夫です．
（現在使われてないとか，そのUbuntuのバージョンに対してまだリリースされてないとかだと思われますが，
授業で使う部分の動作確認はされているはずなので，無視していいです）

Masayoshi-Dohi, 2020-10-12:
このまま進むとp,8のroslaunchのところでopencv_appsが無かったのでrosdepのエラーが原因だと思っていたのですが違うんですね。
opencv_appsもどこからかインストールしなければならないのでしょうか？
ただ今後の課題に支障がなければこのままでも大丈夫です。

ishiguroJSK, 2020-10-12:
あとなんとなくだけど，OpenCVの別のバージョンをどっかで入れてないかも気にしてみて欲しい
```
 $ pkg-config --modversion opencv
3.2.0
```

Masayoshi-Dohi, 2020-10-12:
opencv_appsディレクトリに移動したらできました。ありがとうございます。
もう一つ質問があって、今回の資料で新しく作ったenshu_wsにはCMakeList.txtが見当たらないのですが、これからの作業はどこで行えば良いのですか？

Masayoshi-Dohi, 2020-10-12:
新しくワークスペースを作る方法は分かるのですがそうすると今回セットアップした内容が使えなくなりそうで心配です。

ishiguroJSK, 2020-10-12:
> opencv_appsディレクトリに移動したらできました。

これはちょっとおかしなことを言ってる気がするけど，詳しく書くとどういうことだろう？
落ち着いてやってみたら，roscd opencv_appsが出来て，roslaunchでopencv_appsのエラーが出ることもなくなったということだろうか

> enshu_wsにはCMakeList.txtが見当たらないのですが

CMakeListsはワークスペースには無くて，各ROSパッケージ以下にあります．
roscd [パッケージ名] で移動するところです
```
~/catkin_ws/enshu $ ls
build  devel  logs  src
~/catkin_ws/enshu/src $ ls
dynamixel_motor  kobuki          multisense_ros     rtmros_gazebo  rtmros_tutorials  turtlebot       turtlebot_interactions  turtlebot_simulator
dynamixel_urdf   kobuki_desktop  robot-programming  rtmros_hrp2    trans_system      turtlebot_apps  turtlebot_msgs
~/catkin_ws/enshu/src $ roscd dxl_armed_turtlebot/
~/catkin_ws/enshu/src/robot-programming/dxl_armed_turtlebot (2020) $ ls
CHANGELOG.rst  CMakeLists.txt  README.md  config  doc  euslisp  images  kadai  launch  mainpage.dox  package.xml  scripts  test  urdf  worlds
```

> 新しくワークスペースを作る方法は分かるのですがそうすると今回セットアップした内容が使えなくなりそうで心配です

授業で一からワークスペースを作って全てのパッケージはその中に配置して進めてると思うけど，なにか状況が混乱してないかな？

Masayoshi-Dohi, 2020-10-12:
すみません自分rosというものがいまいち分かっていなくて、まずroscd opencv_appsをやっていませんでした。

それなら今回はdxl_armed_turtlebot内で作業を行うのですね。ただなぜか僕の場合はroscdでdxl_armed_turtlebotに移動できません。

以前まではcatkin_ws/src/beginners_tutorialの中にactionフォルダー、scriptsフォルダー、launchフォルダーなどを作って作業をしていたのですが、10/12の資料では新しくenshu_wsを作ってその中にrobot-programmingをクローンしたりしていたのでどこにpythonファイルを保存するのかなどが分からずに混乱していました。

Masayoshi-Dohi, 2020-10-12:
今まで通りbeginners_tutorials内で作業できるならそれが一番なのですが問題ありませんか？

ishiguroJSK, 2020-10-12:
> 以前まではcatkin_ws/src/beginners_tutorialの中にactionフォルダー、scriptsフォルダー、launchフォルダーなどを作って作業をしていたのですが、10/12の資料では新しくenshu_wsを作ってその中にrobot-programmingをクローンしたりしていたので

これは失礼．新しくenshu_wsを作る流れでしたね．

> 僕の場合はroscdでdxl_armed_turtlebotに移動できません。

授業資料によるとenshu_wsに対しては
```
$ git clone https://github.com/jsk-enshu/robot-programming
$ wstool init .
$ wstool merge robot-programming/.rosinstall.melodic
```
をしているはずなので，enshu_wsをビルド後は，robot-programming以下の
`daisya_euslisp_tutorials  dxl_armed_turtlebot  dynamixel_7dof_arm  turtleboteus`
にはroscdできる状況にないとおかしいのですが，
ビルドが正しく出来ているか，enshu_wsにパスは通っているかを確認すべきですね
（反映されてないだけ，ということもあるので端末を上げ直すなども確認すると良いです）

> どこにpythonファイルを保存するのかなどが分からずに混乱していました。

なるほど，確かに資料を上から進めると，新しくenshu_wsを作ってbeginner_tutorialsは無くなっている状態で，どこにpythonスクリプトを作成して進め，どのパッケージのCMakeLists.txtをITC-LMSで提出したらいいのか．という疑問が出ますね.

@k-okada これはenshu_ws 以下にもbeginner_tutorialsを作成して進めるということで良いのでしょうか？

Masayoshi-Dohi, 2020-10-13:
パソコンを再起動しても変化はありませんでした。
enshu_wsにパスが通っているかの確認はどのようにすればできますか？

一応ですがwstool init .で作成されたenshu_ws/src/robot-programming/.rosinstall.melodicは以下のようになっていました。

[image]

ishiguroJSK, 2020-10-13:
パス(環境変数)はコマンドライン端末(terminal)ごとに存在していて，
今使っている端末内でsource~~~とかexport~~~とかすると一時的にその端末内でのパスが変わります（つまり他で起動中の端末のパスは変更されません）．

授業では端末を上げるたびにsource AAA/devel/setup.bashを実行するように教えられているかと思います．

現在の端末でパスがどうなっているかを調べるときはecho $PATHとかでも一応わかりますが，
ROSの場合はroscdとだけ単体で打つと現在のパスに応じたcatkinワークスペースのdevel以下に飛ぶはずです

Masayoshi-Dohi, 2020-10-13:
まさにそれが原因でした。
ワークスペースが増えるとややこしくなるので、今まで通りcatkin_wsで作業を進めようと思います。
ありがとうございました。

ishiguroJSK, 2020-10-13:
> 今まで通りcatkin_ws/src/beginner_tutorialsで作業を進めようと思います。

本当はちょっと危ないんだけど，それで進めて見るのもいいと思う．
僕の理解ではcatkin_ws以下にはbeginner_tutorialsしか無い状態だと思うけど，
10/12の課題をcatkin_ws以下で進めて，もし「〇〇がない」というエラーが出て，その〇〇がenshu_ws以下にある場合は，
それに依存しているということだから，自分が進めてるpythonファイルをenshu_ws以下に移動する必要がある．

ワークスペースの違いをちゃんと理解していれば，玄人的に，
端末Aではsource ~/enshu_ws/devel/setup.bash; roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch
端末Bではsource ~/catkin_ws/devel/setup.bash; rosrun beginner_tutorials my_script.py
みたいな共存ができると思うけど，
いろいろ苦しくなってきたら，全部enshu_ws以下に配置して進めることになる．

岡田先生と少し話した結果，**10/12の課題はenshu_ws以下にbeginner_tutorialsみたいな新しくパッケージを作るか，面倒ならdxl_armed_turtlebot以下に置いてもいい**，とのことでした．

Masayoshi-Dohi, 2020-10-13:
了解しました。
ありがとうございます。

closed 2020-10-19 by Masayoshi-Dohi