jsk-enshu/robot-programming #346: eigen3を再インストールした後、RVizが起動しなくなった

RyotaUshio, 2020-10-18:
この講義とは別のあるプロジェクトに取り組んでいたところ、Eigenに関するトラブルが起き、
`sudo apt remove libeigen3-dev`してから
`sudo apt-get install libeigen3-dev`として再インストールしました。これによって依存関係が壊れてしまったのか、その後、ロボットシステムの10/12の資料P.7の`roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch`を実行したところ、本来ならばGazeboとRVizの両方が立ち上がるはずが、Gazeboしか起動しませんでした。どのように対処すればよいでしょうか。
(rqtによるロボットの台車の速度指令、アームの角度指令は今まで通り動作しました。)

`roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch`の実行結果は以下の通りでした。

```
ryotaushio@ryotaushio-VirtualBox:~/catkin_ws/src/robot-programming$ roslaunch dxl_armed_turtlebot dxl_armed_turtlebot_gazebo.launch
... logging to /home/ryotaushio/.ros/log/15ed823a-110b-11eb-b691-080027893130/roslaunch-ryotaushio-VirtualBox-3437.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Deprecated: xacro tag 'kobuki' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/ryotaushio/catkin_ws/src/robot-programming/dxl_armed_turtlebot/urdf/robot.urdf.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

xacro.py is deprecated; please use xacro instead
started roslaunch server http://ryotaushio-VirtualBox:36977/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /home/ryotaushio/...
 * /depthimage_to_laserscan/output_frame_id: /camera_depth_frame
 * /depthimage_to_laserscan/range_min: 0.45
 * /depthimage_to_laserscan/scan_height: 10
 * /fullbody_controller/action_monitor_rate: 100
 * /fullbody_controller/constraints/arm_joint1/goal: 0.02
 * /fullbody_controller/constraints/arm_joint2/goal: 0.02
 * /fullbody_controller/constraints/arm_joint3/goal: 0.02
 * /fullbody_controller/constraints/arm_joint4/goal: 0.02
 * /fullbody_controller/constraints/arm_joint5/goal: 0.02
 * /fullbody_controller/constraints/arm_joint6/goal: 0.02
 * /fullbody_controller/constraints/goal_time: 0.5
 * /fullbody_controller/constraints/stopped_velocity_tolerance: 0.02
 * /fullbody_controller/gains/arm_joint1/d: 0.0
 * /fullbody_controller/gains/arm_joint1/i: 0.0
 * /fullbody_controller/gains/arm_joint1/p: 25.0
 * /fullbody_controller/gains/arm_joint2/d: 0.0
 * /fullbody_controller/gains/arm_joint2/i: 0.0
 * /fullbody_controller/gains/arm_joint2/p: 25.0
 * /fullbody_controller/gains/arm_joint3/d: 0.0
 * /fullbody_controller/gains/arm_joint3/i: 0.0
 * /fullbody_controller/gains/arm_joint3/p: 25.0
 * /fullbody_controller/gains/arm_joint4/d: 0.0
 * /fullbody_controller/gains/arm_joint4/i: 0.0
 * /fullbody_controller/gains/arm_joint4/p: 25.0
 * /fullbody_controller/gains/arm_joint5/d: 0.0
 * /fullbody_controller/gains/arm_joint5/i: 0.0
 * /fullbody_controller/gains/arm_joint5/p: 25.0
 * /fullbody_controller/gains/arm_joint6/d: 0.0
 * /fullbody_controller/gains/arm_joint6/i: 0.0
 * /fullbody_controller/gains/arm_joint6/p: 25.0
 * /fullbody_controller/joints: ['arm_joint1', 'a...
 * /fullbody_controller/state_publish_rate: 100
 * /fullbody_controller/stop_trajectory_duration: 0
 * /fullbody_controller/type: velocity_controll...
 * /gazebo/enable_ros_network: True
 * /gripper_controller/action_monitor_rate: 100
 * /gripper_controller/constraints/goal_time: 0.5
 * /gripper_controller/constraints/gripper_joint/goal: 0.02
 * /gripper_controller/constraints/stopped_velocity_tolerance: 0.02
 * /gripper_controller/gains/gripper_joint/d: 0.0
 * /gripper_controller/gains/gripper_joint/i: 0.0
 * /gripper_controller/gains/gripper_joint/p: 10.0
 * /gripper_controller/joints: ['gripper_joint']
[... 174 lines cut ...]
[ INFO] [1603002492.214040534, 0.220000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1603002492.214174776, 0.220000000]: GazeboRosKobuki plugin ready to go! [dxl_armed_turtlebot]
[ INFO] [1603002492.237848491, 0.220000000]: Loading gazebo_ros_control plugin
[ INFO] [1603002492.238253095, 0.220000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1603002492.240005160, 0.220000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ERROR] [1603002492.376680483, 0.220000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint1
[ERROR] [1603002492.377738925, 0.220000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint2
[ERROR] [1603002492.378777206, 0.220000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint3
[ERROR] [1603002492.380721979, 0.220000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint4
[ERROR] [1603002492.381795415, 0.220000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint5
[ERROR] [1603002492.382841665, 0.220000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/arm_joint6
[ERROR] [1603002492.383977530, 0.220000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/gripper_joint
[urdf_spawner-7] process has finished cleanly
log file: /home/ryotaushio/.ros/log/15ed823a-110b-11eb-b691-080027893130/urdf_spawner-7*.log
[ INFO] [1603002492.396594416, 0.220000000]: Loaded gazebo_ros_control.
[INFO] [1603002492.506424, 0.310000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1603002492.515392, 0.310000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1603002492.519475, 0.310000]: Loading controller: joint_state_controller
[INFO] [1603002492.565360, 0.370000]: Loading controller: fullbody_controller
[INFO] [1603002492.889160, 0.690000]: Loading controller: gripper_controller
[INFO] [1603002493.017169, 0.810000]: Controller Spawner: Loaded controllers: joint_state_controller, fullbody_controller, gripper_controller
[INFO] [1603002493.037289, 0.820000]: Started controllers: joint_state_controller, fullbody_controller, gripper_controller
[tuck_arm-12] process has finished cleanly
log file: /home/ryotaushio/.ros/log/15ed823a-110b-11eb-b691-080027893130/tuck_arm-12*.log
```

ishiguroJSK, 2020-10-18:
`sudo apt remove libeigen3-dev`した時にeigenに依存している多くのrosパッケージも消されたんだと思います.
「以下のパッケージが削除されますが良いですか？」的な警告が出ていたんじゃないかと思います．
おそらく普通にros-melodic-desktop-fullを入れ直して、ワークスペース内でもう一度rosdep installとかのくだりをやり直せば，必要なものを再度入れられるはずです

RyotaUshio, 2020-10-18:
やはりそこからやり直すことになるのですね。ありがとうございます。

closed 2020-11-02 by RyotaUshio